wlcrs/huawei_solar #491: [Feature Request]: Wallbox Integration

SvenAbels, 2023-09-30:
### Describe your feature request

Huawei also also offers a smart charger (FusionSolar with 7kw or 22kw) and it wouold be fantastic to integrte it as well

### Proper usage

- [X] I confirm that this is not a bug report or support request
- [X] I confirm that this feature request is within the stated scope of the integration

SvenAbels, 2023-09-30:
URL https://solar.huawei.com/download?p=%2F-%2Fmedia%2FSolarV4%2Fsolar-version2%2Feurope%2Fde%2Fprofessionals%2Fall-products%2Fresidential%2Fsmart-charger%2Fdatasheet%2FHUAWEI_Datenblatt_Residential_Smart_Charger_20230808.pdf

WiebKastanje, 2023-09-30:
Please share product integration details like modbus definitions of this carcharger.

wlcrs, 2023-09-30:
I will not develop things if I cannot test it myself against the actual car charger. I'm also not sure if this needs to be part of this integration: if it needs its own Modbus connection to communicate with, a separate integration is a better idea.

I'm leaving this open for now, hoping that another developer shows interest to implement it.

Roving-Ronin, 2023-10-03:
@SvenAbels 
What type of information would you be looking to get out of any such integration ? Or are you thinking you could control the EV charger ?

If you refer to the user manual for the chargers, they are installed either with an ethernet connection to a DDSU / DTSU-666-FE 
https://support.huawei.com/enterprise/en/doc/EDOC1100260967

Note: Usually this is a 666H with RS485 communication, but it appears there is now an 'FE' model that has Fast Ethernet / RJ45 instead. This model however appears to be aimed at people without a Huawei inverter and allows the 666-FE to connect to your home network (and then to FusionSolar on the internet) and for the EV charger to also connect to the LAN, and then allow the EV charger to directly query the 666-FE for meter data using modbus-tcp.
https://support.huawei.com/enterprise/en/doc/EDOC1100096889/103cc39c/device-commissioning-charger

For sites with a Huawei inverter (and optional LUNA ESS) the EV charger is installed using "virtual meter mode", that sees the EV charger connected to the home network using the RJ-45 Ethernet port, and it then connects to the sDongle via modbus-tcp. Given the inverter (that the sDongle is connected to) has an RS485 connection to the DDSU/DTSU-666H meter, this would mean the EV charger is querying (client) the inverter for the power meter data (via modbus-tcp).

Given above, I don't think that you'll be able to get any information **FROM** the EV charge itself. 

It might be worth having a look instead at the Energy Management Assistant Huawei (EMMA-A01/A02) that are coming out shortly, and look like being positioned to replace the use of DDSU / DTSU-666H meters (that are actually made by CHINT, not Huawei). These WILL combine the functions of these meters, plus the SmartLogger and it appears also the sDongle (with the EMMA having an RS485 port to communicate with the inverter(s) and via that, the LUNA ESS and then also have 2 x Fast Ethernet ports (RJ-45) with 1 x Internal (this looks like it is used to direct patch/connect to an EV charger though, and is not for "LAN" connectivity) and 1 x WAN port (connecting it to your home LAN and providing it access to FusionSolar portal, whilst also potentially providing the modbus-tcp access to the inverter [ if it replaces the sDongle]). 

Quick Guide – [https://support.huawei.com/enterprise/en/doc/EDOC1100321602](https://forums.whirlpool.net.au/go?https%3A%2F%2Fsupport.huawei.com%2Fenterprise%2Fen%2Fdoc%2FEDOC1100321602)

User Manual – [https://support.huawei.com/enterprise/en/doc/EDOC1100325377](https://forums.whirlpool.net.au/go?https%3A%2F%2Fsupport.huawei.com%2Fenterprise%2Fen%2Fdoc%2FEDOC1100325377)

FusionSolar App Quick Guide (EMMA) – [https://support.huawei.com/enterprise/en/doc/EDOC1100312274?idPath=258788303|258788491|258789989|23205712|21608721](https://forums.whirlpool.net.au/go?https%3A%2F%2Fsupport.huawei.com%2Fenterprise%2Fen%2Fdoc%2FEDOC1100312274%3FidPath%3D258788303%7C258788491%7C258789989%7C23205712%7C21608721)

From the EMMA user guide:

[image]

[image]

Note: The EV charger will be required to have firmware 'FusionCharge V100R023C10' installed to communicate to the EMMA. However, current version released on the 28th April 2023 is 'FusionCharge V100R022C10SPC070' (see: https://support.huawei.com/enterprise/en/fusioncharge/fusioncharge-pid-254563737 ), so it would appear an EV charger firmware update will be released AFTER the EMMA is released.

In summary...

a) It appears there's no 'smarts' built into the EV charger itself, if you want to remotely control it at this time it might be best to look at the FusionSolar integration ( https://github.com/tijsverkoyen/HomeAssistant-FusionSolar ) and see if there's any management/reporting that is available via that integration / API.

b) Watch and wait and see what happens with the EMMA-A02 model and see if it adds support for managing/monitoring the EV chargers, or if its just being able to support the chargers querying the EMMA over modbus-tcp for "power meter" type data.

c) Watch and wait and see what happens with the EMMA-A02 model and if it replaces the DDSU/DTSU-666H meters, and IF any additional updates are made to the Huawei inverters modbus registers to support managing/monitoring the EV chargers (same as how the inverters can manage/monitor a LUNA battery connected to them via RS485 cable....).

Side Note: There is a discussion thread created re the EMMA device, to allow discussion about this and the affect on deployments and subsequent flow through affect to WLCRS's great integration (and do we need to sponsor him to fund him getting one, to allow for testing / adding support for EMMA) ;-) 

See Discussion Thread: https://github.com/wlcrs/huawei_solar/discussions/457

Roving-Ronin, 2023-10-03:
Just found this that covers scenarios for the EV chargers install and usage, including options for 

https://photomate.zendesk.com/hc/en-gb/article_attachments/10442149835805

Includes this, that says Fast Ethernet on the charger and the sDongle (No info on if an EMMA is used instead of a DDSU/DSTU-666H, given its still to hit the market) are to be used, in preferance to WiFi.

[image]

Also it advises if you have a charger and inverter talking together (to get the DTSU-666H meter data) then this will stop the sDongle supporting any other modbus-tcp access (i.e. WLCRS integration !). This may be worked around by needing to install the Modbus Proxy via HACS, so the sDongle only 'sees' one modbus-tcp connection, yet WLCRS + Charger talk to it via the Proxy. This would need to be tested and stability checked, as the proxy does take some time to start up and could see the integration and charger timeout waiting for it.

Charging options and 'where' to get the power from are covered in Section 2.2.2 - https://photomate.zendesk.com/hc/en-gb/article_attachments/10442149835805#%5B%7B%22num%22%3A1120%2C%22gen%22%3A0%7D%2C%7B%22name%22%3A%22XYZ%22%7D%2Cnull%2C422.16483%2Cnull%5D

Reading this and re-evaluating it, and their options to install just a charger (with DSTU-666FE meter) or in certain situations where you can't connect the charger and inverter together (so the charger can access the DSTU-666 data via the sDongle and modbus-tcp) to instead install DSTU-666FE (in addition to the existing DTSU-666H that is providing data to the inverter via RS485), as per this diagram:

[image]

It looks more like the EV chargers are really a seperate device from the whole other inverter / battery environment. This (and especially being able to be installed without an inverter setup) would mean the 'smarts' / communication is all contained within the EV charger, with it appearing to be being controlled via FusionSolar / App on phone.

If this is so, this would likely mean this integration would be unlikely to support it, unless the ev chargers end up supporting modbus-tcp access themselves. However this seems unlikely, as these chargers have been on the market for a while and there is no documentation or mention of modbus support from Huawei, and just the Communication Matrix (for SPC070) doesn't appear to support modbus connections to the ev chargers, instead it looks like the charger just uses modbus to query the DTSU-666FE meter or ' Virtual Meter'  (i.e. sDongle on inverter, that would pass through the DTSU-666H wired to it via RS485)

Traffic Flow - Charger as Destination
[image]

Traffic Flow - Charger as Source
[image]

Anyhow... apologies for long and rambling, as sourcing info and trying to work out how this charger works... :-(

But it really looks like it is a modbus-tcp client (not server) and its controlled via mobile app and/or FusionSolar, which (if you have a PV plant setup) is where it gets the info as to how much PV capacity you have. This data then being used, in conjunction with the Power Meter (ie. Active Power) to allow the charger to determine if your PV is exporting (so a positive number in HA for Active Power) and when it reaches the minimum level of surplus power, to then turn the charger on (and off etc). This is assuming setting it up to preference PV power charging over grid.

If this is correct it still goes back to there is limited chance the WLCRS integration will be able to work with the ev Charger, as it doesn't have modbus access (server) and it doesn't tie into the inverter's RS485 cascade /modbus setup, so there's no way to talk to it. As mentioned above the best option looks to be to see if the FusionSolar integration adds support for it (that would require Huawei API supporting / publishing it). Doing it this way (if/when available) would still allow you to "have a switch" in HA to allow toggling charge from PV / Grid, as the FusionSolar would provide you a sensor entity to do this, you'd just need to add it into your dashboard.

Re the tracking Active Power (W) and total consumption of Energy (kWh) you would be easiest to do with with a Shelly EM3 or IoTaWatt appliance with their own CT clamps.

I can't find any mention and it looks like Hauwei also haven't included support for Open Charge Point Protocol (OCPP), that allows 3rd party software to pull data from inverters (or their consumption meter data) and then use AI to work out when to charge / Grid vs PV / when cheapest. An example being https://chargehq.net/ but alas Huawei inverters aren't supported by them (yet?)
If OCPP support was added and ChargeHQ added Shelly EM's as a data source for meter data (likely given their global market) then it would work (but no holding ones breath for OCPP support).

wlcrs, 2023-10-03:
Hi @Roving-Ronin , thank you for your ramblings! They were quite enlightening. 

I agree with your conclusions:
- it's a separate product which is not a part of the SUN2000/LUNA2000 ecosystem;
- the charger requiring Modbus-TCP access will probably collide with the use of this integration;
- a workaround with a Modbus Proxy can be tried, but it remains to be seen how stable that will be in practice.

I'm keeping this issue open for now to aid discoverability due to all the noteworthy info in it, but I'll mark it as 'wontfix'.

SvenAbels, 2023-10-04:
Thanks  alot for the detailed information. This actually sounds worse than I expected because in reality it means that I either need to rely on a proxy which might be stable or not or that I have to choose between making my charger smart or using this integration. The fact that the modbus-tcp connection of the wallbox is limited to one device was a surprise to me.  I could buy this additional DTSU666-FE but it's hard to eveb buy it here in my region.

Therefore, I will switch to another charger with a seperate smart meter (go-e) so that it does not impact the modbus tcp connection.

However, the general problem will of course still exist for anyone who buys a Huawei charger in connection with the Huawei inverter and if you order all at the same time then it's not unlikely that you want to stay at the same manufacturer. I wonder why Huawei implemted the charger this way...

Roving-Ronin, 2023-10-04:
I think the one to watch will be the option of the EMMA. As per the first picture / page pasted above, the logical diagram Huawei has provided shows the EV charger directed patched via Cat5/6 cable, between it and the EMMA. Conversely the EMMA is only shown connecting to the inverter(s) using RS485 serial cable.
From this we can deduce that the ev charger would therefore not be able to talk to the inverter over the LAN and so would not 'tie up' the Modbus-tcp port.
Given EMMA replaces the DDSU/DTSU-666H and the sdongle (and has a WAN port & WiFi option, to allow it to connect to the LAN and on to the internet / FusionSolar portal) and hence has its own 'power / consumption meter' capabilities, it's probably a good theory that it will have a Modbus tcp server built in (& a decent one able to handle many clients simultaneously, like the SmartLogger appliances it also replaces).
- [ ] This means if the ev charger queries of its directly patched LAN port to the EMMA, it will be serviced directly by EMMA with power meter data, without affecting Modbus access to the inverters (think about it, as it's directly patched between the two, this means those two LAN ports will have to configure themselves as an RFC1812 (non routable address space, ie 10.0.0.0/8, 172.16.0.0/16, 192.168.1.0 etc). With this happening it then means the WAN port (on your home LAN and its address range) will be totally seperate and running another (or dual homed at least) Modbus server instance, that then services queries through to the inverter / batteries etc.

As above, we'll still have to wait and grab the release notes and communication matrix info from when EMMA is released soon.

PS. I think that looking to support EMMA would appear to be critical for the integration, giving it is being pushed as the default install and connectivity method on Huawei installs in future. Still waiting to get hands on the Modbus register definitions, but hopeful that as it's the 'next generation' of the DDSU / DTSU (& a heap more features / functions) that it will see the existing Modbus registers for the meter be used for it (& extra registers added if needed), which would mean it would be compatible with wlcrs for the existing meter data.

Definitely worth every sponsoring $10 for wlcrs to upgrade his setup to a EMMA in future ;-)

trollsoft7, 2023-10-11:
Good evening, I was following your very sophisicated discussion and would like to bring in 2 other facts. I have an Inverter (10kw) combined with a Luna 10kWh and a DTSU-666 smart meter installed. Your integration is connected via RPI-WIFI-Inverter -> from my point of view very stable.
And now also I had the Wallbox22kW installed. Installation in the Fusionportal was quite a FW-updating-everything - but then it worked smoothly. Yet I found, that I can't "control" or "monitor" the wallbox.
Therefore I searched the net and found this: 
[image]
https://forum.huawei.com/enterprise/en/huawei-smart-charger-scharger-7ks-s0-modbus-definitions/thread/708706730801774592-667213868771979264

Trying it out at home - could connect to the wallbox via Modbus-tool:
[image]

Some values arrive - but continuously - no clue what that is, but maybe you experts have an idea.
If I have misunderstood your discussion and this is not helpful, please apologize. :-)

SvenAbels, 2023-10-11:
@trollsoft7 So you connected the charger but you can still use the home assistant wlcrs integration?

trollsoft7, 2023-10-11:
Yes - charger is connected and I see it in the fusion portal. wlcrs-integration continues to work flawlessly. As written my HA-RPi4 is connected to the Inverters WIFI (which is not recommended in WLCRS-documentation, but it works very stable for me). Though I haven't charged a car yet, as I don't have any :-(.

chrissi5120, 2023-11-22:
Hi, i got pretty much the same setup as @trollsoft7 but I use a wifi-router to connect via client-mode to the inverter-Wifi, all very stable. I would also really like to get this topic forward, and I also got a car, so I could report back on actual charging.

Roving-Ronin, 2023-11-23:
@chrissi5120 so your RPi4 has 2 x WiFi dongles and they connect to the inbuilt (commissioning) WiF AP on each inverter, whilst you have the WallCharger connected to the LAN. 
Also looking at the register addresses, given they are numbered 0 to 12 (decimal) this would appear to be the registers if your locally/directly connecting to the WallBox (i.e. HA to Wallbox IP).

This is a software architecture question that falls to how WLCRS wants to add support for the wallbox. Using that method would mean requiring a method of indicating and/or detecting that your connecting directly to the WallBox whereas with the apparent move from sDongles and DDSU/DTSU-666 meters to instead using EMMA appliances (that have the meter/CT functionality, and also a direct connection LAN port for plugging in the WallBox, RS485 to talk with inverters/batteries and a WAN port to connect it to the home network (and FusionSolar / modbus-tcp support etc) that would require supporting that (default) method.

EMMA Issue/Discussion - https://github.com/wlcrs/huawei_solar/issues/498

NiFu90, 2023-12-05:
Hey guys,

I am also very interested in having the Wallbox integrated into home assistant mainly for tracking of power usage for the car, so read-only information gathering.
Browsing through the internet and after creating an Huawei-account and registering my wallbox I was able to download the modbus tcp interface definitions of the device. Maybe this gives one of you some more information to work with.
[Northbound Modbus-TCP Interconnection Protocol for AC Chargers.pdf](https://github.com/wlcrs/huawei_solar/files/13556228/Northbound.Modbus-TCP.Interconnection.Protocol.for.AC.Chargers.pdf)
I also found a Communication Matrix which might help.
[FusionCharge V100R022C10SPC070 Communication Matrix.xlsx](https://github.com/wlcrs/huawei_solar/files/13556245/FusionCharge.V100R022C10SPC070.Communication.Matrix.xlsx)

I would be grateful for any kind of implementation and am willing to give feedback if needed.
Thanks

moppsgti, 2024-02-03:
Hello, I would also be very interested in a solution ;)

Huawei Sun 10kw
Huawei SDoungel SDongle V200R022C10SPC108
Huawei Wallbox Smartcharger 22KT-S0 V100R023C10SPC020

SDoungel and Wallbox Connected via Wi-Fi

Roving-Ronin, 2024-05-05:
@chrissi5120 @trollsoft7 

Just wondering if anyone with a SmartCharger has managed to connect to it, using modbus and the register list above?
The above hexadecimal list though seems 'off' as those register addresses literally translate to 0 and 1 - 12, that seems very strange verses the normal 4-5 digital decimal number of a register.

How are your SmartChargers also connected into the Huawei Inverters / PV setup? Would you have an diagrams of this, as the Huawei documentation previously has mentioned that to connect it into the Inverters etc your connecting via the 

Looking at this post (above - https://github.com/wlcrs/huawei_solar/issues/491#issuecomment-1744626189 ) it looks like you can either connect the SmartCharger to the meter using a 'virtual meter', but that uses up the single modbus-tcp connection the sDongle WiFi+FE provides, and therefore stops WCLRS being able to connect. Alternatively it looks like you have to install a second DTSU-666FE (Fast Ethernet) meter (Datasheet - https://cdn.shopify.com/s/files/1/0781/9748/9944/files/Huawei_TDS_DDSU666-FE_EV-Charger__EN.pdf?v=1708256060 ) that the SmartCharger connects to over the home FastEthernet (wired) network. Meaning you'd have a DDSU/DTSU-666H meter that is wired into the Inverter(s) via RS485 and operating seperate, just to provide Grid Power/Energy data to the SmartCharger would be this new/secondary DTSU-666-FE meter.

If your using something like https://qmodbus.sourceforge.net/ to query the SmartCharger and see if the modbus responds, I've just found these document that has some more 'realistic' appearing modbus numbers, but its confusing if these are for the SmartChargers or are for FusionCharge DC (i.e. commercial EV charge setup) devices. If anyone could test it would be appreciated.

[image]

From: **Northbound Modbus-TCP Interconnection Protocol for AC Chargers**
At: https://support.huawei.com/enterprise/en/doc/EDOC1100330890?idPath=258788305%7C254827211%7C254827535%7C258790439%7C254563737

trollsoft7, 2024-05-05:
@Roving-Ronin 
Thanks for your reply - my Charger is connected via WIFI and uses that "virtual meter", which is provided by the SDONGLE (connected via LAN). Meanwhile I also browsed over to https://github.com/evcc-io/evcc which claimed to support the wallbox, but obviously isn't yet: https://github.com/evcc-io/evcc/discussions/10262  - I added there a link to your information.

I tried to poll the wallbox with your "new" addresses -  as soon it connects (directly to the wallbox port 502) it starts receiving (one message per second):
[image]

Sending a request for example for charging control at 8198 register does not change anything. It seems to be ignored.
If you find I have something setup wrongly, please let me know.

Maybe following information is also interesting: in the setup menu of the Wallbox I can choose a 3rd-party-application with a IP-address. I used the IP of Home Assistant - but of course nothing happens, as home assistant should run a modbus-server, which it isn't.

Thanks for all your efforts!

Roving-Ronin, 2024-05-05:
@trollsoft7 

OK but re your setup, what you have 1/2/3 x L1 or M1 inverters and a DDSU (1p) or DTSU (3p) 666H meter connected via RS485 to the inverter?

IMHO I would be seeing about getting a secondary DDSU/DTSU-666-FE meter installed to provide data to the SmartCharger, whilst the inverter gets its data from the DDSU/DTSU-666H meter. 
Given that Huawei documentation, apart from the alternative of getting an EMMA installed (that I'm hearing is ~€900 - €1,000! and you can do more with HA anyhow) if it can support multiple Modbus-tcp connections, you will NEED to have this FE meter. Without it you will be reliant upon Modbus-proxy (if/when it works) or forced to choose between HA+WLCRS but no access/control of the Charger, or Charger able to get grid data and on/off itself but no HA connection.

smoki3, 2024-05-05:
The wallbox Modus ist actually not working. You can check out evcc here on GitHub. There is a long discussion about it.

lucabacchipg, 2024-05-23:
Hi, i read all article below, but my question is very easy:  Since Fusionsolar display all Fusioncharge date, how is possible to take same data in HA ? I need to have same data to use for automation purpose.
[image]

Roving-Ronin, 2024-05-23:
@lucabacchipg until someone can get their hands upon the modbus register definitions, provide Thijs (i.e. WLCRS) access remotely to their Huawei EV Charger to test it again, there will be no chance of this. Step 1 all hinges on getting those modbus definitions.

lucabacchipg, 2024-05-23:
I can give access to a PC remotely to access the entire infrastructure, just organize

NiFu90, 2024-05-23:
@Roving-Ronin I already added the Modbus definitions, see below.

> Hey guys,
>
> I am also very interested in having the Wallbox integrated into home assistant mainly for tracking of power usage for the car, so read-only information gathering. Browsing through the [...]

lucabacchipg, 2024-05-23:
I have received full documentation from Huawey Support below
[Northbound Modbus TCP Interconnection Protocol for AC Chargers.pdf](https://github.com/wlcrs/huawei_solar/files/15415593/Northbound.Modbus.TCP.Interconnection.Protocol.for.AC.Chargers.pdf)
[Solar.Inverter.Modbus.Interface.Definitions.v05.pdf](https://github.com/wlcrs/huawei_solar/files/15415594/Solar.Inverter.Modbus.Interface.Definitions.v05.pdf)
[SmartPVMS 24.2.0 Northbound API Reference.pdf](https://github.com/wlcrs/huawei_solar/files/15415595/SmartPVMS.24.2.0.Northbound.API.Reference.pdf)

Roving-Ronin, 2024-06-04:
Previously mentioned "I can't find any mention and it looks like Hauwei also haven't included support for Open Charge Point Protocol (OCPP), that allows 3rd party software to pull data from inverters (or their consumption meter data) and then use AI to work out when to charge / Grid vs PV / when cheapest. An example being https://chargehq.net/ but alas Huawei inverters aren't supported by them (yet?)"

Just looking at this again and noticed OCPP 1.6 support is "planned to be available via software upgrade in 2023.Q4" and current manual refers to it:

[image]

kanzie, 2024-06-11:
Joining the conversation as I just installed one of these boxes today and needed it integrated to home assistant. I have infra and comfortable with a keyboard but have little knowledge about modbus-signaling. Happy to help if anyone needs hands on the setup here to test or debug.

ad-ha, 2024-08-01:
Hi, everyone,

Any further developments on this end? It seems really odd that with all data available on FusionSolar and having the Scharger connected to the inverter and Sdongle, we may not be able to get some data.

I am waiting for a contact in Spain to help with this and OCPP (it seems to be supported but there's no option on the menu to get it working). I believe the guy is on vacation, but as soon as I hear from him I will let you know.

Cheers

Roving-Ronin, 2024-08-03:
Please read up the thread (& also EMMA thread). ATM we are still seeking more/clarification of the Modbus registers for the SCharger.

Given we've been lucky enough to get what appears to be valid register definitions for the EMMA effort (as people have time) is focused there. There are some SCharger Modbus registers in the EMMA btw, but they have to be validated, let alone they appear to o my be a subset of the overall list of registers (ie iirc doesn't have registers for setting max kW limit, or on/off?).

Btw re your comment about not being able to work, but you have it connected and also an sDongle. The issue there is that the SCharger uses the sDongle to create a 'virtual meter' to provide the SCharger with power/current/energy data. However in doo g this it uses the SINGLE connection for Modbus the SDongle allows for, thereby stopping any others (such as from HomeAssistant).
Unless you also install an EMMA, that I believe support multiple connections, the other option is you need to I install a 2nd meter just for the SCharger, this being the new DDSU/DTSU-666FE meter that has a Fast Ethernet connection, that connects it to your home LAN and/direct patch to the SCharger.

ad-ha, 2024-08-05:
Hi @Roving-Ronin,

Thanks for the reply. I've been reading the thread and was just not fully understanding why it was not possible to get it integrated. Anyhow, with you explanation on the Modbus single connection, I got it.

Also, as promised, I heard back from Huawei on the OCCP protocol and enabling it on the SCharger.

Seems like there is the need to install a "special" version (060), which I attach, along with the OCCP Commissioning manual. Hope this may help.

[FusionCharge_V100R023C10SPC060_FusionCharge-EU-CHARGE.zip](https://github.com/user-attachments/files/16494657/FusionCharge_V100R023C10SPC060_FusionCharge-EU-CHARGE.zip)
[FusionCharge AC OCPP1.6 Protocol Messages and Commissioning Guide.pdf](https://github.com/user-attachments/files/16494659/FusionCharge.AC.OCPP1.6.Protocol.Messages.and.Commissioning.Guide.pdf)

Cheers

trollsoft7, 2024-09-02:
Obviously @DNAngelX did the job for ioBroker - https://github.com/DNAngelX/ioBroker.huawei-charger/tree/main
I am not an expert, but it seems he decodes Modbus-messages. FYI

cailloup007, 2024-09-07:
Hello all
Interesting topic, though a bit disappointing that Huawei did not fully integrate Scharger to fusionsolar...
I have sun2000L1 + luna + powermeter + luna
I haven't need a smart dongle when I did the sun + powermeter installation, inverter is connected to router through wifi, powermeter is connected to inveter by wire
If I understand correctly the below screenshot, I won't be able to use virtual powermeter, I need either a smart dongle, or another meter called "FE", is this correct ?
[image]

ad-ha, 2024-09-07:
> I won't be able to use virtual powermeter, I need either a smart dongle (which can be connected to the router through wifi?), or another meter called "FE", is this correct ?

Hi,

Yes, that's correct, That's exactly what happened in my case as well. Once I got the WLAN FE for the inverter, I was able to get the SCharger to get the virtual meter and have a balanced installation (means that it will not overload the power available from the network).

I also hoped that Huawei managed this way better and in a different way, having a meter directly embebed on the Charger for example as many other brands. Also the OCPP connection is something that I have not been able to setup yet.

Hope this helps

cailloup007, 2024-09-07:
Thanks for quick help! let's order the fu**ing dongle then...

ad-ha, 2024-09-07:
> There's something I'm not sure about the behaviour
> Scharger is only controlled by the meter, there's no communication with inverter and ess sadly
> Let's say battery is full
> And we have >1400W of excess power from sun measured by the meter
> Car charge start, charger tries to get excess power ~=0
> Then there's clouds, power is taken from the network, who will react first ? ess will give power ? or car charge will decrease power ?
> I'm guessing ess will react first and car will keep charging at >1400W until ess is empty

Hi again,

I do not have a battery, so I am not sure on how to correctly reply to that. However, from my experience, the charger connects to the FusionSolar and the Inverter through the WLAN FE dongle. With that, it is able to know the power available and produced from the PV and balance it with the grid power available. 

In my case, i have a limit of 5.75kW power supply. When charging, if for example I connect home's AC, the Scharger decreases the power to charge the car to the maximum possible to balance between PV and the grid supply. With battery I understand that you may do the same.

You also have an option on SCharger to be ECO mode, so you will only charge when PV is enough or when you have batteries to supply that energy, without getting power from the grid. At least, that's my understanding.

Cheers

ad-ha, 2024-09-07:
> Obviously @DNAngelX did the job for ioBroker - https://github.com/DNAngelX/ioBroker.huawei-charger/tree/main I am not an expert, but it seems he decodes Modbus-messages. FYI

Is this what was missing to get the SCharger included on this integration?

kanzie, 2024-09-08:
Just to add my experience. I have this exact setup with the wlan dongle and
have not been able to ever charge the car using PV preferred despite full
battery and instead feeding 3kW back to grid.

I’m really disappointed with this product and only wish people would stop
buying this crap so they get the message.

PRiekenberg, 2024-09-29:
@wlcrs Hi, is it possible for you @wlcrs to adopt the iobroker adapter for the huawei wallbox ([](https://github.com/DNAngelX/ioBroker.huawei-charger/tree/main)) from @DNAngelX into this home assistant integration? This would be really nice and helpful for all the people using this wallbox!

passibe15, 2024-09-29:
That one is not fully working yet (see [here](https://github.com/evcc-io/evcc/discussions/10262#discussioncomment-10498514)), and, unfortunately, I doubt it ever will (see my comment in that thread). IMO modbus is not the way to go, we will have to wait for OCPP to be implemented.

Kugelfang666, 2024-09-29:
>  we will have to wait for OCPP to be implemented.

I somehow have lost my faith for Huawei ever doing this. Its was announced for Q4'23 so were almost a year overdue...

ad-ha, 2024-09-29:
Hi, @passibe15 and @Kugelfang666 

> we will have to wait for OCPP to be implemented

Do you guys mean this OCCP implementation: https://github.com/wlcrs/huawei_solar/issues/491#issuecomment-2268774105

Huawei already released this firmware, still available only for users whom request it, but is done: https://github.com/user-attachments/files/16494657/FusionCharge_V100R023C10SPC060_FusionCharge-EU-CHARGE.zip

lucabacchipg, 2024-09-29:
Hello, i can give access via vpn to my installation of  fusioncharge.

Kugelfang666, 2024-09-29:
> Hi, @passibe15 and @Kugelfang666
> 
> > we will have to wait for OCPP to be implemented
> 
> Do you guys mean the OCPP implementation that I referred here: [#491 (comment)](https://github.com/wlcrs/huawei_solar/issues/491#issuecomment-2268774105) ??
> 
> Huawei already released this firmware, still available only for users whom request it, but is done: (although I was not able to get it working with OCPP addon from HACS on HA). https://github.com/user-attachments/files/16494657/FusionCharge_V100R023C10SPC060_FusionCharge-EU-CHARGE.zip

check this report is that the implementation of OCPP is not working in this version. https://github.com/evcc-io/evcc/discussions/3806#discussioncomment-9224416

ad-ha, 2024-09-29:
> check this report is that the implementation of OCPP is not working in this version. [evcc-io/evcc#3806 (reply in thread)](https://github.com/evcc-io/evcc/discussions/3806#discussioncomment-9224416)

Thanks for pointing it out, although German is not ,y strongest thing, I got the idea.

I have a contact in Huawei Europe and will enquire about that.

Cheers

PRiekenberg, 2024-10-24:
@DNAngelX found a solution how to add the Huawei Wallbox via Huawei API to IOBroker.

Would it be possible for you to integrate this too into your HomeAssistant Integration?

You can see his documentation here:

https://github.com/evcc-io/evcc/discussions/10262#discussioncomment-11028418

smoki3, 2024-10-24:
I think if someone create a HA integragtion than it would be easy to use this in combination with evcc

Criaga, 2024-11-14:
Good evening everyone, I was looking at the Huawei website to find new updates. I came across version V100R024C10. I think it's a pre-release but it looks promising.

[main-content.pdf](https://github.com/user-attachments/files/17753397/main-content.pdf)

PRiekenberg, 2024-11-14:
> Good evening everyone, I was looking at the Huawei website to find new updates. I came across version V100R024C10. I think it's a pre-release but it looks promising.
> 
> [main-content.pdf](https://github.com/user-attachments/files/17753397/main-content.pdf)

Looks promising. Can you share the firmware file?

Criaga, 2024-11-14:
> Looks promising. Can you share the firmware file?

I can't find it anywhere -.-

Criaga, 2024-11-14:
> Don't get your hopes up.
> 
> > V100R024C10
> 
> That's a firmware for DC chargers (just like V100R023C10SPC070 was). See the changelog where it talks about liquid cooling, charging with 300 A and POS-terminals.

I hate you 🤣

HolgerMiara, 2024-11-14:
I usually look at one of their distributors for recent firmware, have a look yourself: https://photomate.zendesk.com/hc/en-gb/categories/4406398674589-Firmware

lucabacchipg, 2024-11-14:
Hello, this is last Modbus documentaion (Oct 2024) and lat Realease not for next day FW upgrade of FusionCharge 22KW  
[FusionCharge V100R023C10SPC160T_Release Notes.pdf](https://github.com/user-attachments/files/17754053/FusionCharge.V100R023C10SPC160T_Release.Notes.pdf)

If someone know how to handle...

[Northbound Modbus-TCP Interconnection Protocol for AC Chargers.pdf](https://github.com/user-attachments/files/17754001/Northbound.Modbus-TCP.Interconnection.Protocol.for.AC.Chargers.pdf)

Hollerith98T, 2024-11-25:
I just received information via email from Huawei today:

**Upgrade SCharger firmware version manually**:
[Charger_2023_011 Upgrade SCharger firmware version manually.pdf](https://github.com/user-attachments/files/17909545/Charger_2023_011.Upgrade.SCharger.firmware.version.manually.pdf)

**OCPP1.6 Protocol Messages and Commissioning Guide**:
[FusionCharge AC OCPP1.6 Protocol Messages and Commissioning Guide (1).pdf](https://github.com/user-attachments/files/17909546/FusionCharge.AC.OCPP1.6.Protocol.Messages.and.Commissioning.Guide.1.pdf)

**Last firmware for my SCharger-7KS-S0**:
[FusionCharge_V100R023C10SPC120_FusionCharge-EU-CHARGE.zip](https://github.com/user-attachments/files/17909547/FusionCharge_V100R023C10SPC120_FusionCharge-EU-CHARGE.zip)

**Northbound Modbus-TCP Interconnection Protocol for AC Chargers Guide**:
[Northbound Modbus-TCP Interconnection Protocol for AC Chargers.pdf](https://github.com/user-attachments/files/17909548/Northbound.Modbus-TCP.Interconnection.Protocol.for.AC.Chargers.pdf)

I'll try to start working on it but I need some support, if anyone wants to help let me know!

Kugelfang666, 2024-11-25:
> I just received information via email from Huawei today:
>
> **Upgrade SCharger firmware version manually**: [Charger_2023_011 Upgrade SCharger firmware version [...]

I might have to disappoint you. From what I understand the 120 FW does not include OCPP.... maybe others can also confirm just to save you the hassle of trying

Geroldin, 2024-11-28:
I just had a look at my Charger: SCharger-22KT-S0 (Software: V100R023C10SPC120). Nothing available anywhere regarding OCapP (not under third party mhmt). Checked Installer and User Gui. However I'm still hope the Wallbox will be integrated through my Emma. It is recognised by this integration but ignored for now. Looks like the Emma ist acting as a modus proxy here (similar to the inverter via emma). So if we get the modus parameters integrated into this integration then it could work
 Already took a look at it on Huawei didd but sadly doesn't say anything about "PV preferred charging). Really looking for a way to integrate that function locally so I can cutoff my cloud connection (especially after the latest development with the shutdown of Deye Inverters remotely...).

Kugelfang666, 2024-11-28:
OCPP is present but just in one FW Version weich is xxxx-060

The more recent Version xxxx-120 does not feature it. 

Apparently the implementation of ocpp is not according to the specs as the creatures on the EVCC repo found out. 

There is someone who implemented an cloud API based script in nodered/iobroker more infos can be found here

https://github.com/evcc-io/evcc/discussions/10262#discussioncomment-9133946

If you're right and the EMMA controls the Wallbox via modbus then we probably need someone who can sniff this communication and reverse engineer... unfortunately my setup is based on the sDongle but I would absolutely love if we finally get the chance to remote control the charger

ad-ha, 2024-11-28:
Hi

> OCPP is present but just in one FW Version weich is xxxx-060
> 
> The more recent Version xxxx-120 does not feature it.

I have version 120 and it has OCPP menu and options. The issue is that there's no way to get it working.

I've been in touch with Huawei Europe and seems like they are now testing version 160, which in theory, according to what I've been told, will solve all those OCPP issues and improve the actual authentication methods (which are quite unclear). 

I was told that this 160 version will be release at some point during 2025 (which month, no idea... 🙄 )

Kugelfang666, 2024-12-02:
> Hi
> 
> 
> 
> > OCPP is present but just in one FW Version weich is xxxx-060
> 
> > 
> 
> > The more recent Version xxxx-120 does not feature it.
> 
> 
> 
> I've been in touch with Huawei Europe and seems like they are now testing version 160, which in theory, according to what I've been told, will solve all those OCPP issues and improves the actual authentication methods (which are quite unclear). 
> 
> 
> 
> Huawei seems to be doing a beta test with some users... so if someone is able to get this FW **V100R023C10SPC160T** let us know.
> 
> 
> 
> I was told that this 160 version will be release at some point during 2025 (which month, no idea... 🙄 )

The release notes of this version don't mention OCPP... not very encouraging 

https://www.photovoltaikforum.com/core/attachment/511563-fusioncharge-v100r023c10spc160t-release-notes-pdf/

Hollerith98T, 2024-12-10:
> Hi
>
>
> I've been in touch with Huawei Europe and seems like they are now testing version 160, which in theory, according to what I've been told, will solve all those OCPP issues and improves the [...]

 Nothing to be done, this is the response from support:
" The firmware version is not on the support platform yet, therefore, it cannot be downloaded and provided to the customers.
Kindly wait until it will be officially out "

kanzie, 2024-12-10:
Why wouldn’t that ask any of us to beta test since this group probably are
the best audience. Frustrating! I wish so that we never touched huawei
products

XalaTheShepard, 2025-02-06:
Hello everyone, I really hope the wallbox could also be integrated in this integration. The most useful part would be to start and stop the charging and to assign the energy consumption. Kind regards, Robin

[150 more replies]