wlcrs/huawei_solar #493: end of charge soc[Feature Request]:

kurtdefeyter, 2023-10-02:
### Describe your feature request

Dear,

I appears to me that the end of charge soc has a range between 90-100
Could this be changed in the future to make a wider range , like 80-100
Many thanks

### Proper usage

- [X] I confirm that this is not a bug report or support request
- [X] I confirm that this feature request is within the stated scope of the integration

closed 2023-10-02 by wlcrs

kurtdefeyter, 2023-10-03:
Hi thijs,

Unfortunately not working trying to change to any % under 90 result in 
Logger: homeassistant.components.websocket_api.http.connection
Source: custom_components/huawei_solar/number.py:284
Integration: Home Assistant WebSocket API (documentation, issues)
First occurred: 07:31:59 (13 occurrences)
Last logged: 07:32:15

[281473184251072] Failed to write value [890] to register 47081: SlaveFailure
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 227, in handle_call_service
    await hass.services.async_call(
  File "/usr/src/homeassistant/homeassistant/core.py", line 1969, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2006, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 235, in handle_service
    return await service.entity_service_call(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 870, in entity_service_call
    response_data = await _handle_entity_call(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 942, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/number/__init__.py", line 105, in async_set_value
    await entity.async_set_native_value(native_value)
  File "/config/custom_components/huawei_solar/number.py", line 284, in async_set_native_value
    if await self.bridge.set(self.entity_description.key, int(value)):
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/bridge.py", line 365, in set
    return await self.client.set(name, value, slave=self.slave_id)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 523, in set
    return await _do_set()
           ^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/backoff/_async.py", line 151, in retry
    ret = await target(*args, **kwargs)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 519, in _do_set
    return await self._write_registers(reg.register, builder.to_registers(), slave)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 551, in _write_registers
    raise WriteException(
huawei_solar.exceptions.WriteException: Failed to write value [890] to register 47081: SlaveFailure

I did check the value in the .py

wlcrs, 2023-10-03:
The error shows that the battery firmware is rejecting the value

kurtdefeyter, 2023-10-03:
You are correct Sir .
Did not notice that in the luna docu .

Parameter

Description

Value Range

Working mode

For details, see the description on the app screen.

  *   Maximum self-consumption
  *   TOU (time-of-use)
  *   Fully fed to grid

Maximum charge power (kW)

Retain this parameter to the maximum charge power. Additional configuration is not required.

Charge: [0, Maximum charge power]

Maximum discharge power (kW)

Retain this parameter to the maximum discharge power. Additional configuration is not required.

Discharge: [0, Maximum discharge power]

End-of-charge SOC (%)

Set the charge cutoff capacity.

90%–100%

End-of-discharge SOC (%)

Set the discharge cutoff capacity.

0%–20%

Charge from grid

If Charge from grid function is disabled by default, comply with the grid charge requirements stipulated in local laws and regulations when this function is enabled.

  *   Disable
  *   Enable

Grid charge cutoff SOC

Set the grid charge cutoff SOC.

[20%, 100%]

Many thanks

kurtdefeyter, 2023-10-03:
Update , i did open a request at huawei to change this behavior, again many thanks

commit d312c0a0 (2023-10-02): Make cutoff charge range bigger

Fixes #493