wlcrs/huawei_solar #494: [Bug]: Loosing the battery during the night

woopstar, 2023-10-03:
### Describe the issue

Lately, I've been losing the battery controls during the night. Reloading the integration or restarting HA makes everything work.

Battery sensors, Inverter, and SmartMeter are not lost and work perfectly. It is only the controls that become unavilable.

[image]

### Bescribe your Huawei Solar Setup

Inverter Type: SUN2000-10KTL-M1SUN2000-12
Inverter Firmware version: V100R001C00SPC153 
SDongle present: yes
Power meter present: three phase
Battery: LUNA2000 10kWh
Battery Firmware version: V100R002C00SPC119

### How do you connect to the inverter?

Via the SDongle, wired connection

### Upload your Diagnostics File

[config_entry-huawei_solar-b6969ff22160fb972ea75303bbc34452.json.txt](https://github.com/wlcrs/huawei_solar/files/12789321/config_entry-huawei_solar-b6969ff22160fb972ea75303bbc34452.json.txt)

### Upload your relevant debug logs

```text
I dont have any debug logs yet.
```

### Please confirm the following:

- [X] The problem is still present in the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

wlcrs, 2023-10-03:
Without debug logs showing what is going wrong I will not be able to
diagnose and solve the problem.

woopstar, 2023-10-03:
> Without debug logs showing what is going wrong I will not be able to diagnose and solve the problem.

Yes, ofc. I have enabled the debug logger. Not waiting for it to happen again. I suspect it happens during the night when the battery has been used (and is down to 5%) that the battery might go offline or so. But it's odd that it's only the controls/configuration/diagnostics that become unavailable. The sensors continue to work.

HolgerMiara, 2023-10-03:
Maybe too stupid to consider, but do you have any wifi access points which put themselves to sleep mode at night after a certain duration of inactivity?  That's at least what happened to me...

woopstar, 2023-10-03:
> Maybe too stupid to consider, but do you have any wifi access points which put themselves to sleep mode at night after a certain duration of inactivity? That's at least what happened to me...

Not at all. It's also ONLY battery configuration that stops working. Everything else continues to work perfect.

JohnMcLear, 2023-10-08:
I appear to also lose battery control but I think it's related to batteries going to < 2% SoC and they do appear to come back.

woopstar, 2023-10-08:
> Without debug logs showing what is going wrong I will not be able to diagnose and solve the problem.

It happend this night. Lost the configuration to the battery again. Sensors are working perfect. The battery configuratoin does not come back even though the battery is charged to 100% again.

The HA log is 65mb so I'm unable to upload it in a comment. I added it to WeTransfer:

https://wetransfer.com/downloads/07c02170c65b216d4e457c98e12e020320231008195732/446eb517e507b0dc61db3b606af6b7ae20231008195742/594af7

wlcrs, 2023-10-09:
Hi @woopstar , do you have a timestamp when the sensors became unavailable? You should be able to see that happening in the logbook of your HA quite easily if you filter on one of the config entities:

[image]

woopstar, 2023-10-09:
Fair point. This should show:

[image]

wlcrs, 2023-10-09:
I'm confused: you sent me logs starting on 2023-10-07 15:12:48.955, and the failure occurred on 2023-10-07 08:51:14 if I read your screenshot correctly?

Did it also occur in the night from 7 on 8 October?

woopstar, 2023-10-09:
Sorry. Updated my view to match the logs:

[image]

wlcrs, 2023-10-09:
last successful update before failure: 
```
2023-10-08 09:03:33.631 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 47075 with length 14 from slave 1
2023-10-08 09:03:33.631 DEBUG (MainThread) [pymodbus.logging] send: 0x4b 0x6e 0x0 0x0 0x0 0x6 0x1 0x3 0xb7 0xe3 0x0 0xe
2023-10-08 09:03:33.631 DEBUG (MainThread) [pymodbus.logging] Adding transaction 19310
2023-10-08 09:03:34.078 DEBUG (MainThread) [pymodbus.logging] recv: 0x4b 0x6e 0x0 0x0 0x0 0x1f 0x1 0x3 0x1c 0x0 0x0 0x13 0x88 0x0 0x0 0x13 0x88 0x7f 0xff 0xff 0xff 0x3 0xe8 0x0 0x32 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x2 0x0 0x0 0x3 0xe8 old_data:  addr=None
2023-10-08 09:03:34.078 DEBUG (MainThread) [pymodbus.logging] Processing: 0x4b 0x6e 0x0 0x0 0x0 0x1f 0x1 0x3 0x1c 0x0 0x0 0x13 0x88 0x0 0x0 0x13 0x88 0x7f 0xff 0xff 0xff 0x3 0xe8 0x0 0x32 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x2 0x0 0x0 0x3 0xe8
2023-10-08 09:03:34.078 DEBUG (MainThread) [pymodbus.logging] Factory Response[ReadHoldingRegistersResponse': 3]
2023-10-08 09:03:34.078 DEBUG (MainThread) [pymodbus.logging] Getting transaction 19310
2023-10-08 09:03:34.078 DEBUG (MainThread) [pymodbus.logging] [0, 5000, 0, 5000, 32767, 65535, 1000, 50, 0, 0, 0, 2, 0, 1000]
2023-10-08 09:03:34.078 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b'\x13\x88']
2023-10-08 09:03:34.078 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b'\x13\x88']
2023-10-08 09:03:34.128 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 47200 with length 46 from slave 1
2023-10-08 09:03:34.128 DEBUG (MainThread) [pymodbus.logging] send: 0x4b 0x6f 0x0 0x0 0x0 0x6 0x1 0x3 0xb8 0x60 0x0 0x2e
2023-10-08 09:03:34.129 DEBUG (MainThread) [pymodbus.logging] Adding transaction 19311
2023-10-08 09:03:34.783 DEBUG (MainThread) [pymodbus.logging] recv: 0x4b 0x6f 0x0 0x0 0x0 0x5f 0x1 0x3 0x5c 0x0 0x0 0xf 0x0 0xe8 0x3 0xf4 0x1 0x90 0x1 0x2c 0x1 0x58 0x2 0xc8 0x0 0x1 0x0 0x96 0x0 0x84 0x3 0x70 0x17 0x2 0x0 0xf7 0x0 0xa0 0x5 0xff 0xff 0x6 0x0 0x5 0x0 0x3f 0x0 0x3 0x0 0x0 0x0 0x10 0x27 0x0 0x0 0x0 0x0 0x0 0x0 0x80 0x43 0x6d 0x38 0x88 0x13 0x0 0x0 0xff 0xff 0xff 0xff 0x0 0xe1 0x0 0x3 0x2 0x0 0x0 0x0 0x24 0x73 0x12 0x0 0x0 0x0 0x0 0x0 0x1 0xe1 0x0 0x3 0x0 0x0 0x0 0x0 0x27 0x10 0x0 0x0 0x27 0x10 old_data:  addr=None
2023-10-08 09:03:34.783 DEBUG (MainThread) [pymodbus.logging] Processing: 0x4b 0x6f 0x0 0x0 0x0 0x5f 0x1 0x3 0x5c 0x0 0x0 0xf 0x0 0xe8 0x3 0xf4 0x1 0x90 0x1 0x2c 0x1 0x58 0x2 0xc8 0x0 0x1 0x0 0x96 0x0 0x84 0x3 0x70 0x17 0x2 0x0 0xf7 0x0 0xa0 0x5 0xff 0xff 0x6 0x0 0x5 0x0 0x3f 0x0 0x3 0x0 0x0 0x0 0x10 0x27 0x0 0x0 0x0 0x0 0x0 0x0 0x80 0x43 0x6d 0x38 0x88 0x13 0x0 0x0 0xff 0xff 0xff 0xff 0x0 0xe1 0x0 0x3 0x2 0x0 0x0 0x0 0x24 0x73 0x12 0x0 0x0 0x0 0x0 0x0 0x1 0xe1 0x0 0x3 0x0 0x0 0x0 0x0 0x27 0x10 0x0 0x0 0x27 0x10
2023-10-08 09:03:34.783 DEBUG (MainThread) [pymodbus.logging] Factory Response[ReadHoldingRegistersResponse': 3]
2023-10-08 09:03:34.783 DEBUG (MainThread) [pymodbus.logging] Getting transaction 19311
2023-10-08 09:03:34.783 DEBUG (MainThread) [pymodbus.logging] [0, 3840, 59395, 62465, 36865, 11265, 22530, 51200, 256, 38400, 33795, 28695, 512, 63232, 40965, 65535, 1536, 1280, 16128, 768, 0, 4135, 0, 0, 0, 32835, 27960, 34835, 0, 65535, 65535, 225, 3, 512, 0, 9331, 4608, 0, 0, 481, 3, 0, 0, 10000, 0, 10000]
2023-10-08 09:03:34.783 DEBUG (MainThread) [pymodbus.logging] handle: [b'\xf4\x01', b'\x90\x01']
2023-10-08 09:03:34.783 DEBUG (MainThread) [pymodbus.logging] handle: [b'\xc8\x00', b'\x01\x00']
2023-10-08 09:03:34.783 DEBUG (MainThread) [pymodbus.logging] handle: [b'p\x17', b'\x02\x00']
2023-10-08 09:03:34.783 DEBUG (MainThread) [pymodbus.logging] handle: [b'\xff\xff', b'\x06\x00']
2023-10-08 09:03:34.783 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x03\x00', b'\x00\x00']
2023-10-08 09:03:34.783 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b'\x00\x00']
2023-10-08 09:03:34.783 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x88\x13', b'\x00\x00']
2023-10-08 09:03:34.783 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\xe1', b'\x00\x03']
2023-10-08 09:03:34.783 DEBUG (MainThread) [pymodbus.logging] handle: [b'$s', b'\x12\x00']
2023-10-08 09:03:34.783 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x01\xe1', b'\x00\x03']
2023-10-08 09:03:34.783 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b"'\x10"]
2023-10-08 09:03:34.783 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b"'\x10"]
2023-10-08 09:03:34.834 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 47255 with length 45 from slave 1
2023-10-08 09:03:34.834 DEBUG (MainThread) [pymodbus.logging] send: 0x4b 0x70 0x0 0x0 0x0 0x6 0x1 0x3 0xb8 0x97 0x0 0x2d
2023-10-08 09:03:34.834 DEBUG (MainThread) [pymodbus.logging] Adding transaction 19312
2023-10-08 09:03:35.796 DEBUG (MainThread) [pymodbus.logging] recv: 0x4b 0x70 0x0 0x0 0x0 0x5d 0x1 0x3 0x5a 0x0 0x1 0x0 0x0 0x5 0x9f 0x0 0x7f 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x1 old_data:  addr=None
2023-10-08 09:03:35.796 DEBUG (MainThread) [pymodbus.logging] Processing: 0x4b 0x70 0x0 0x0 0x0 0x5d 0x1 0x3 0x5a 0x0 0x1 0x0 0x0 0x5 0x9f 0x0 0x7f 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x [... cut ...]
Sadly no obvious messages or stacktraces that explain why the [`HuaweiSolarConfigurationUpdateCoordinator`](https://github.com/wlcrs/huawei_solar/blob/main/__init__.py#L393) stops with updating after that failure.

The timeout also happens for the normal [`HuaweiSolarUpdateCoordinator`](https://github.com/wlcrs/huawei_solar/blob/main/__init__.py#L280) of which the code is 99% identical, but which does continue to refresh even after that failure which happened at the same time.

I honestly don't really know how to proceed on debugging this issue.

woopstar, 2023-10-13:
I tried to update both the inverter, battery, and dongle to the latest version from FusionSolar. The issue still persists. I'm wondering if it happens when the battery is empty (5%) and goes into hibernation.

This is my lame mitigation so far, simply to restart HA so it works again.

```
alias: Huawei Solar - Restart HA if the battery is unavailable
description: ""
trigger:
  - platform: time_pattern
    minutes: /5
condition:
  - condition: template
    value_template: "{{ states.switch.battery_charge_from_grid.state == 'unavailable' }}"
  - condition: template
    value_template: >-
      {{ ((as_timestamp(now()) - as_timestamp(states('sensor.uptime'))) / 60) |
      round(0) > 30 }}
action:
  - service: homeassistant.restart
    data: {}
mode: single
```

JohnMcLear, 2023-10-29:
Note that this could be related to SDongle and connectivity to Fusion Solar.

@woopstar  Can you please confirm if your inverters are talking to the Huawei Fusion Solar Management system?  Thanks!

jires, 2024-01-14:
When your battery is empty and switch "charge from grid" is of, battery goes into standby mode. All configurations sensor will become unknown. Your battery status sensor from diagnostic section should say "standby" or something like that.
When "charge from grid" is on, goes battery never standby, because is waiting for charging from grid at any time, Status sensor is "running" all the time.
So, check status sensor.

JohnMcLear, 2024-01-14:
> When your battery is empty and switch "charge from grid" is of, battery goes into standby mode. All configurations sensor will become unknown. Your battery status sensor from diagnostic section [...]

Who is this comment aimed at @jires ?

jires, 2024-01-14:
> Who is this comment aimed at @jires ?

Mainly to @woopstar and anybody else who experiencing this, when charge from grid is off and battery goes on standby.

scorpio2k2, 2024-01-20:
for me, even if charge from grid is off, if I touch the "Battery Backup power SOC" the battery is charged to that level... I had to disable the automation because It was charging it from grid.

closed 2024-05-20 by github-actions[bot]