wlcrs/huawei_solar #497: [Feature Request]:

Krasto82, 2023-10-08:
### Describe your feature request

Hi, I wanted to suggest the possibility of receiving data from fusionsolar server instead of modbus. Although less updated, thank you

### Proper usage

- [X] I confirm that this is not a bug report or support request
- [X] I confirm that this feature request is within the stated scope of the integration

wlcrs, 2023-10-08:
Not in scope. You can use https://github.com/tijsverkoyen/HomeAssistant-FusionSolar instead

closed 2023-10-08 by wlcrs

Krasto82, 2023-10-08:
Thanks
___________________________________________

*Patrick Donchi*
Le informazioni qui trasmesse sono destinate soltanto alle persone o entità
a cui sono state inviate.
È possibile che il contenuto di questo materiale sia riservato, personale
e/o protetto dal segreto professionale.
Se il messaggio venisse inviato per errore a un'altra persona, la
riservatezza e segretezza delle informazioni
rimangono tutelate e non potranno essere violate. È vietata la revisione,
ritrasmissione, diffusione,
utilizzo in qualsiasi altra forma di queste informazioni, o azione sulla
base di queste informazioni,
da parte di entità non autorizzate. Se questo messaggio vi è stato
recapitato per errore, siete pregati
di darne avviso al mittente e di eliminare immediatamente il materiale.

The information transmitted is intended only for the person(s)or entity to
which it is addressed and may
contain confidential and/or legally privileged material. Delivery of this
message to any person other than
the intended recipient(s) is not intended in any way to waive privilege or
confidentiality.
Any review, retransmission, dissemination or other use of, or taking of any
action in reliance upon,
this information by entities other than the intended recipient is
prohibited. If you receive this in error,
please contact the sender and delete the material from any computer.