wlcrs/huawei_solar #498: [Feature Request]: Huawei Energy Management Assistant (EMMA-A01 / A02) Support

Roving-Ronin, 2023-10-09:
### Describe your feature request

As per the wall charger feature request, this 'issue' is intended as a location for people to be able to start gathering information and discussing how it works and how it may potentially be supported in the integration.

Given that the EMMA in future will be the preferred method of installation for Huawei Solar installations, with it :
- providing the energy/consumption meter functionality (superceded the standalone DDSU/DTSU-666H meter);
- connecting to the inverter(s) via RS485 to provide the aforementioned energy/consumption meter functionality;
- providing network/internet connectivity for the inverter(s) and LUNA ESS connected via RS485, with its inbuilt WiFi+WAN ethernet ports;
- connecting to the (forthcoming) SmartGuard-63A that is the replacement for the BackupBox device, that provides for a protected circuit in the event of a grid outage;
- connecting to the (forthcoming) Huawei Smart Charger 7/21kW for EV charging, via either the LAN port (bad naming, it is only for charger <--> EMMA direct patching, with a single charger). See Issue: https://github.com/wlcrs/huawei_solar/issues/491
- Also adds smart / cloud capabilities such as being able to access weather information and make decisions, such as if it will be raining / overcast tomorrow, then charge the LUNA ESS off the cheapest grid power available before tomorrow.

Hopefully Huawei will be utilising the same modbus registers used on the DDSU/DTSU-666H meters, to provide themselves with maximum compatibility for existing installations upgrading, that will allow for 'out of the box' compatibility with WLCRS. However this needs to be confirmed when documentation is published detailing the modbus register definitions for EMMA. 
As EMMA is adding additional functionality, it would be a good expectation that there may be additional modbus registers added to EMMA, that would then need to be reviewed and see if any are 'worthwhile' or of use to request being added to WLCRS.

To date documentation published on EMMA is:
[EMMA-(A01, A02) User Manual.pdf](https://github.com/wlcrs/huawei_solar/files/12842554/EMMA-.A01.A02.User.Manual.pdf)
[FusionSolar App Quick Guide (EMMA).pdf](https://github.com/wlcrs/huawei_solar/files/12842555/FusionSolar.App.Quick.Guide.EMMA.pdf)
[Huawei - EMMA Quick Installation Guide.pdf](https://github.com/wlcrs/huawei_solar/files/12842556/Huawei.-.EMMA.Quick.Installation.Guide.pdf)

### Proper usage

- [X] I confirm that this is not a bug report or support request
- [X] I confirm that this feature request is within the stated scope of the integration

Roving-Ronin, 2023-11-23:
FYI Huawei have released an installation videa for the EMMA devices, and a new User Guide that includes updating network information.

Note:  The EMMA is a device that manages energy in a home with a PV system. It can implement unified scheduling and management of home energy. The EMMA has two models:

**EMMA-A01:** _Only PV and ESS features are supported_. Plant-level power control functions, such as maximum self-consumption and grid-tied point control, are supported for unified scheduling of home energy and plant-level energy optimization, further improving the PV energy utilization

**EMMA-A02:** _Features of PV, ESSs, smart chargers, and smart loads are supported. In addition to unified scheduling of home energy, the EMMA-A02 can connect to smart loads such as chargers, SG Ready heat pumps, and smart switches_. Users can set the reservation time to charge vehicles and heat water in advance at the specified time. In addition, users can set the priority to use PV energy for devices as required to make the best use of PV energy.

_______________________________________

**EMMA Installation Video:** https://support.huawei.com/enterprise/en/doc/EDOC1100338746?idAbsPath=7639526%7C258788303%7C258788491%7C258789989%7C23205712%7C256539983

**[EMMA-(A01, A02) User Manual - 2023-10-31 - v02.pdf]**(https://github.com/wlcrs/huawei_solar/files/13445816/EMMA-.A01.A02.User.Manual.-.2023-10-31.-.v02.pdf) from https://support.huawei.com/enterprise/en/doc/EDOC1100325377/9a24721e/networking

**Comments:** 

Updated network diagrams appear to confirm that the EMMA will replace the sDongle as the (default?) method to provide connectivity for inverters/site to FusionSolar, whilst the DDSU/DTSU-666 meters will also be superceded with the EMMA providing this functionality also (option to use inbuilt CT's <63A or connect external CT's for >63A systems).

There is only reference to modbus-tcp in relation to certificates and authenticating with 'peer NMS' and FusionSolar.

[image]

[image]

Looking at the RS485 details, it has an RS485 A and B, with A going to the inverter/cascade and B for '3rd party devices'.
Previously on the SmartLogger devices this meant the DDSU/DTSU-666 meter, however this functionality is now inbuilt to EMMA, 
To date EMMA documentation has zero reference to connecting to a SmartLogger device (IMHO looks like EMMA will be pitched at home users and IF anyone needs a SL that is for more commercial installations where a DDSU/DTSU is not used/needed). 
Pending more information about EMMA and its modbus-tcp specifics, given metion of it and prior support in the sDongle it must have it, but as a fallback there is the question of could RS485-B be used as method to connect and Elfin modbus-tcp appliance if needed as a 'last resort'?

[image]

torwag, 2024-01-19:
Hi,
just to understand, I did nowhere see a configuration with cascaded inverters. Did Huawei mention anywhere how this EMMA would work in that case. To me it looks like a combination of a smart meter (DTSU666-H) with the smart logger (SmartLogger 3000 which seems to be intended for larger PV systems) or Smart Module (Smart Module 1000).
Honestly, Huawei is terrible in naming devices so that it doesn't confuse the heck out of people...
SmartMeter, SmartLogger, SmartModule, SmartDongle and now EMMA....

Roving-Ronin, 2024-01-22:
@torwag the EMMA appliance is a replacement option for the DDSU/DTSU-666H, USB sDongle etc so from an inverters persepctive is doesn't matter is its 1 or 3 inverters in a cascade. Instead of wiring in the DDSU/DTSU-666H in future you'll connect the EMMA via that RS485 port.

Also read the prior posts carefully and you'll notice at the bottom of 'Figure 2-2' it mentions:

_"Inverters connect to the EMMA over RS485. A maximum of three inverters can be connected in parallel"_

Worth noting also is that since that post, Huawei / Shelly have also announced that EMMA can integrate with some of their devices, see:

[image]

MarkusKAuatria, 2024-03-09:
Hello
EMMA is now available in Europe

When do you want to start the integration?
Thank you
Mark

Roving-Ronin, 2024-03-10:
@MarkusKAuatria 

Do you have an EMMA installed already and are offering to install the current WLCRS integration version and see if it 'mostly' works already (assuming it uses the same modbus registers as the DDSU/DTSU-666H for the power meter data)? Post that your able to provide WLCRS with remote access to your setup to allow him to develop and test, for any new modbus registers that are required?

MarkusKAuatria, 2024-03-16:
Hello
I have already installed EMMA-A02.
YES, I can give you remote access from my HASSIO test system.
I would like to send you the login details on a private channel
(Email, Telegram,...)
I'm looking forward to a good collaboration

Konfiguration
->EMMA-A02

--> SUN2000-10KTL-M1 (Modbus 1)
---> 1. LUNA2000-10-S0 (15kw)
---> 2. LUNA2000-10-S0 (15kw)

--> SUN2000-5KTL-M1 (Modbus 2)
--->  14x SUN2000-450W-P

Roving-Ronin, 2024-03-18:
@MarkusKAuatria 
Have you tried installing the integration yet, any luck ? If so, what information do you see, or are their any logs ?  

"_Emma is connected via pin 2/3 and not via PIN 5/7 like the DTSU"_ 

PIN 5 ias 'shielding ground' (shared with LUNA also) with PIN7 (485A2) and PIN9 (RS485B2) being used to connect to the DTSU-666H.  Are you sure that EMMA is correctly wired though with it using PIN 2 (RS485B1-1 'Differential -') and PIN 3  (RS485A1-2 'Differential +'), apart from these being from different pairs, the Huawei documentation shows it as PIN 1 (+) and PIN 3 (-) , that is also the A1/B1 pair previously used to cascade   (i.e. out from 1&3 from higher up modbus device, that then goes into 2&4 from lower device in the cascade)? See diagram posted earlier/above, from the EMMA manual.

**Note:** This would mean that EMMA now would become the Modbus Master, with the 'primary' inverter now becoming a modbus slave to it. 
This requires confirming the modbus ID# of EMMA and the inverters, as installation may now change to be using ID# of EMMA, then Primary inverter and lastly seconary inverter.
i.e. if originally inverters installed with Primary inverter is ID#1, Secondary is ID#2 and when EMMA was installed it just assumed ID#3. Then installing the integration would now mean needing to use ID 3,1,2  (whereas previously the sDongle didn't have an ID and you just entered it as 1,2).

Hoping that Huawei used the same modbus registers that the DDSU/DTSU-666H meter uses, meaning the Smart Meter part of EMMA will work with WLCRS. I haven't see any updated information as to Modbus Registers for EMMA, and IF it adds any new ones to 'share' whatever additional data has / provides access to.

wlcrs, 2024-03-18:
@MarkusKAuatria 
Do you have an account on https://community.home-assistant.io/ ? I can send you my contact details via private message over there.

Can you also download the latest (and current only) file from https://support.huawei.com/enterprise/en/fusionsolar/smarthems-pid-256539983/software/ and upload it here? For SUN2000 software downloads, the ZIP-files with the firmware updates also include PDF's with documentation on how the available modbus registers look like. As I do not own such a device, I do not have access to those downloads.

MarkusKAuatria, 2024-03-18:
@wlcrs 
My https://community.home-assistant.io/ account ist the same as the GITHUB account (I changed it now)

@Roving-Ronin 
Sorry it is 7 and 9
The Modbus Adress from EMMA is 0 so it is 0,1,2
[image]

torwag, 2024-03-19:
Just out of curiosity, what would be the benefit of the EMMA for home assistant integration? As I understood, the "smart" features would be why we connect it to home assistant in the first place.
If the connection would be easier, that would be a win, I'm just in the middle of setting up a dedicated Rpi-based home assistant server (as remote unit to the main server). Dealing with (in my case) three wifi dongles for three inverters, is a bit a hassle right now, as HaOS doesn't come with the wifi kernel modules yet. As usual I was stupid selecting the "wrong" wifi sticks 😉

wlcrs, 2024-03-19:
I've received access details from Markus. The EMMA device does not have any open TCP ports at the moment, so he will need to check if he can access a setting to fix this (which seems to be available according to a screenshot from the [manual](https://support.huawei.com/enterprise/en/doc/EDOC1100312274/a23f4487/setting-router-parameters-of-the-charger?idPath=258788303|258788491|258789989|23205712|21608721) ).

Alternatively, the EMMA-device also exposes a serial Modbus RTU interface on 2 pins (cfr. bottom screenshot of https://github.com/wlcrs/huawei_solar/issues/498#issuecomment-1823788453). So that is another scenario that we can explore if necessary.

wlcrs, 2024-06-05:
The EMMA device I had access to via Markus does not answer to any queries: not via TCP, not via RTU (serial). We haven't been able to obtain instructions on how we can locally connect to a solar installation with EMMA from Huawei.

I would urge everybody considering to order a Huawei Solar installation with EMMA device to first contact Huawei to verify if a local Modbus-connection is supported. If possible, please obtain a manual on how to achieve this and share it with me here.

fkhofmann, 2024-06-05:
Oh, that's too bad, that is certainly not an optimal situation. I have also tried to get information on the topic as a potential future Huawei customer. Huawei's technical support didn't provide any information because I am not a customer (couldn't provide a device serial no) and referred me to a distributor. The distributor also didn't want to tell me anything because I am an end user.

I guess a Huawei device owner would have to open a service request.

tmenguy, 2024-06-05:
> Hallo
> These were the last two feedbacks from HUAWEI Support
>
> • > Unfortunately, we do not have any guideline about how access works exactly and how the individual services can be accessed (Modbus TCP, MQTT, web,...) especially for EMMA-A02.
> • > Please be advised that we created related SR NO. xxxxxxx and we will come back with a resolution as soon as possible.
> Kindly ask you to choose the option “Reply All” when communicating with our engineers in order to track your response easily.

Hi,

And did you get a chance to try to connect through the inverter Wifi? It should be Independant from the EMMA no?

BR
Thomas

Roving-Ronin, 2024-06-06:
@tmenguy the issue is adding support for EMMA, especially given in future installations it replaces the sDongle, DDSU/DTSU-666H consumption meter and becomes the primary means for communication for the plant (that also has 2 x M1 3p inverters).

tmenguy, 2024-06-06:
Yes, and I was looking at adding an EMMA to my existing installation (with the new backup box ... but it is not 3phases for now it seems). But correct me if I'm wrong: it will replace the meter and Sdongle all at once, but the inverter wifi will stay no? so perhaps there is a way to still connect through it? 
(Or perhaps I'm completely dumb here and it is the Sdongle that was creating all the modbus connection and the inverter wifi was only a way to access the Sdongle capabilities, and the inverter is only a pass through. But if not it means at least the inverter inner meter and commands could still be available through the wifi no? )

Roving-Ronin, 2024-06-06:
@tmenguy The EMMA-A02 is available to support 3p, with AC voltage options of:
1P2W: 100 ~ 240 V, 50 / 60 Hz 
3P3W: 346 ~ 415 V, 50 / 60 Hz 
3P4W: 346 ~ 415 V, 50 / 60 Hz 

The SmartGuard documentation has started mentioning a 'SmartGuard-63A-T0'  with T0 = Three-phase Whole Home Backup system.  See:

User Manual: https://support.huawei.com/enterprise/en/doc/EDOC1100366781?idPath=258788303%7C258788489%7C258789981%7C21102413%7C255740899

Networking Quick Guide: https://support.huawei.com/enterprise/en/doc/EDOC1100366779?idPath=258788303|258788483|258789967|21102414|21571652

Be aware though, that whilst it mentions 'whole home backup' have a look at the wiring diagram:
[image]

You'll note that it refers to 'Backup Load' and 'Non Backup Load', then have a look at the new 5k-12k MAP0 inverter series manual, with some handy networking diagrams on page 17-18:
https://support.huawei.com/enterprise/en/doc/EDOC1100366802?idPath=258788305|254827209|258792409|22755755

[image]

From this it appears the SmartGuard-63A-T0 will allow connecting the grid feed into it and then attaching the (bulk of the) house to its protected output up to 63A, so its now inline between your previous grid feed and the switchboard. You could then attach your 3p HVAC/Ducted Aircon unit to the unprotected circuit. Idea appears to be that you are protected this way and then can add an Automatic Transfer Switch (ATS) in to allow cutting over from a grid/mains feed to having a generator kick in, so your LUNA ESS + SmartGuard 'protects' the house for the 20ms until the generator kicks in automatically.

Note that is with the MAP0 inverters and typically it costs a few extra $ for a 3p with a automatic start. If you have the $ for a 3p generator and already have a M1 inverter(s), you could just have an ATS / Manual Transfer Switch (that don't cost much) installed and rely upon the LUNA covering you for a few minutes and walking outside and manually starting the generator.

ismiregal17, 2024-06-07:
Hello, the new backup box Smartguard can also only supply 1 phase. The 3-phase designation refers to the inverter. There will be a real 3-phase backup box from Huawei in the future, but it will not be able to work together with the inverters currently on the market. There will be a new inverter for this backup box. The reason is that the current inverters are not unbalanced load capable.

tmenguy, 2024-06-08:
Ouch ...

Roving-Ronin, 2024-06-08:
@ismiregal17 
"The reason is that the current inverters are not unbalanced load capable."

Good point, overlooked that....

dragosmd, 2024-07-29:
Hi all,
I've just installed Huawei SmartGuard-63A-S0 with EMMA-A02 and updated it the latest version that allowed me to open 502 port.
I've tried everything connecting to it via Modbus but no luck.
Anything that I could do to help this move on?

I'm still pulling data though the Inverter AP

dernicc, 2024-08-01:
Ich schließe mich hier auch mal an.
Seit dem letzten Update kann ich in FusionSolar in den EMMA Parametereinstellungen ein Verwaltungsgerät hinzufügen.

[image]

Ich habe aber noch keine Info gefunden, wie dieses Verwaltungssystem mit EMMA kommuniziert.

Hat da jemand weitere Details zu?

wlcrs, 2024-08-03:
I've implemented initial support for EMMA devices, but as I do not own this device, I was not able to properly test this. I don't want to do a release before I have confirmation that it works.

**To test:**

0/ make sure that the firmware version of your EMMA is V100R024C00SPC100 or higher.
1/ turn off TLS encryption for Modbus TCP:

[image]

2/ download the repository ZIP file: https://github.com/wlcrs/huawei_solar/archive/refs/heads/main.zip
3/ extract the ZIP archive and rename the folder to huawei-solar
4/ put it in the `config/custom_components` folder of your Home Assistant.
5/ follow instructions from README: https://github.com/wlcrs/huawei_solar?tab=readme-ov-file#installation

If the documentation is correct, the EMMA will always be device 0. You need to lookup and add the slave ids of your inverters yourself, as they were chosen by your installer.

If I didn't make any error (which is highly unlikely), the integration should properly initialize.

If it fails, then please post the relevant parts of the full logs with the errors (buttons on the bottom of the Logs page).

If you don't see anything in the logs, then add the following to your `configuration.yaml` and try again:
```
logger:
  logs:
    pymodbus: debug 
    huawei_solar: debug
    homeassistant.components.huawei_solar: debug
```

ismiregal17, 2024-08-03:
@wlcrs I have tried it but unfortunately it does not work.

This error was caused by a custom integration

Logger: custom_components.huawei-solar.config_flow
Quelle: custom_components/huawei-solar/config_flow.py:410
Integration: huawei-solar
Erstmals aufgetreten: 19:03:43 (Incidents 5): Last logged 19:04:30

Unexpected exception while connecting via ModbusTCP
Traceback (most recent call last):
  File "/config/custom_components/huawei-solar/config_flow.py", line 410, in async_step_setup_network
    info = await validate_network_setup(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/huawei-solar/config_flow.py", line 104, in validate_network_setup
    bridge = await HuaweiSolarBridge.create(
                   ^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: HuaweiSolarBridge.create() got an unexpected keyword argument 'host'

dragosmd, 2024-08-03:
TypeError: HuaweiSolarBridge.create() got an unexpected keyword argument 'host'
2024-08-03 21:05:24.740 ERROR (MainThread) [custom_components.huawei_solar.config_flow] Unexpected exception while connecting via ModbusTCP
Traceback (most recent call last):
  File "/config/custom_components/huawei_solar/config_flow.py", line 410, in async_step_setup_network
    info = await validate_network_setup(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/huawei_solar/config_flow.py", line 104, in validate_network_setup
    bridge = await HuaweiSolarBridge.create(
                   ^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: HuaweiSolarBridge.create() got an unexpected keyword argument 'host'
2024-08-03 21:05:24.934 ERROR (MainThread) [custom_components.huawei_solar.config_flow] Unexpected exception while connecting via ModbusTCP
Traceback (most recent call last):
  File "/config/custom_components/huawei_solar/config_flow.py", line 410, in async_step_setup_network
    info = await validate_network_setup(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/huawei_solar/config_flow.py", line 104, in validate_network_setup
    bridge = await HuaweiSolarBridge.create(
                   ^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: HuaweiSolarBridge.create() got an unexpected keyword argument 'host'

Going to verify the tls.
Firmware is ok
config # nc -vz 192.168.1.11 502
192.168.1.11 (192.168.1.11:502) open

wlcrs, 2024-08-03:
@ismiregal17  @dragosmd Thank for your the feedback. I fixed that bug in the config flow. Hopefully you can add the integration successfully now.

dragosmd, 2024-08-03:
@wlcrs 
So far no: 

2024-08-03 22:48:14.478 DEBUG (MainThread) [pymodbus.logging] Connecting to 192.168.1.11:502.
2024-08-03 22:48:14.479 DEBUG (MainThread) [pymodbus.logging] Connecting comm
2024-08-03 22:48:14.480 DEBUG (MainThread) [pymodbus.logging] Connected to comm
2024-08-03 22:48:14.480 DEBUG (MainThread) [pymodbus.logging] Resetting frame - Current Frame in buffer -
2024-08-03 22:48:14.490 DEBUG (MainThread) [pymodbus.logging] -> transport: received eof
2024-08-03 22:48:14.490 DEBUG (MainThread) [pymodbus.logging] Connection lost comm due to None
2024-08-03 22:48:14.491 DEBUG (MainThread) [pymodbus.logging] callback_disconnected called: None
2024-08-03 22:48:14.491 DEBUG (MainThread) [pymodbus.logging] Wait comm 1000000 ms before reconnecting.
2024-08-03 22:48:15.766 ERROR (MainThread) [huawei_solar.huawei_solar] Modbus client is not connected to the inverter.
NoneType: None
2024-08-03 22:48:15.767 ERROR (MainThread) [backoff] Giving up _do_read(...) after 6 tries (huawei_solar.exceptions.ConnectionInterruptedException: Modbus client is not connected to the inverter.)
2024-08-03 22:48:15.767 ERROR (MainThread) [huawei_solar.huawei_solar] Aborting client creation due to error.
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/backoff/_async.py", line 151, in retry
    ret = await target(*args, **kwargs)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/backoff/_async.py", line 151, in retry
    ret = await target(*args, **kwargs)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 365, in _do_read
    raise ConnectionInterruptedException(message)
huawei_solar.exceptions.ConnectionInterruptedException: Modbus client is not connected to the inverter.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 167, in create
    await huawei_solar._initialize()
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 116, in _initialize
    self.time_zone = (await self.get(rn.TIME_ZONE)).value
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 239, in get
    return (await self.get_multiple([name], slave))[0]
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 276, in get_multiple
    response = await self._read_registers(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 419, in _read_registers
    return await _do_read()
           ^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/backoff/_async.py", line 159, in retry
    await _call_handlers(on_giveup, **details, exception=e)
  File "/usr/local/lib/python3.12/site-packages/backoff/_async.py", line 37, in _call_handlers
    await handler(details)
  File "/usr/local/lib/python3.12/site-packages/backoff/_async.py", line 16, in f
    return coro_or_func(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 341, in backoff_giveup
    raise ReadException(
huawei_solar.exceptions.ReadException: Failed to read register 43006 after 6 tries
2024-08-03 22:48:15.981 ERROR (MainThread) [huawei_solar.huawei_solar] Modbus client is not connected to the inverter.
NoneType: None
2024-08-03 22:48:15.982 INFO (MainThread) [backoff] Backing off _do_read(...) for 1.0s (huawei_solar.exceptions.ConnectionInterruptedException: Modbus client is not connected to the inverter.)
2024-08-03 22:48:16.983 ERROR (MainThread) [huawei_solar.huawei_solar] Modbus client is not connected to the inverter.
NoneType: None
2024-08-03 22:48:16.984 INFO (MainThread) [backoff] Backing off _do_read(...) for 2.0s (huawei_solar.exceptions.ConnectionInterruptedException: Modbus client is not connected to the inverter.)
2024-08-03 22:48:18.985 ERROR (MainThread) [huawei_solar.huawei_solar] Modbus client is not connected to the inverter.
NoneType: None
2024-08-03 22:48:18.985 INFO (MainThread) [backoff] Backing off _do_read(...) for 4.0s (huawei_solar.exceptions.ConnectionInterruptedException: Modbus client is not connected to the inverter.)
2024-08-03 22:48:18.985 DEBUG (MainThread) [pymodbus.logging] Connecting to 192.168.1.11:502.
2024-08-03 22:48:18.985 DEBUG (MainThread) [pymodbus.logging] Connecting comm
2024-08-03 22:48:18.987 DEBUG (MainThread) [pymodbus.logging] Connected to comm
2024-08-03 22:48:18.987 DEBUG (MainThread) [pymodbus.logging] Resetting frame - Current Frame in buffer -
2024-08-03 22:48:18.999 DEBUG (MainThread) [pymodbus.logging] -> transport: received eof
2024-08-03 22:48:18.999 DEBUG (MainThread) [pymodbus.logging] Connection lost comm due to None
2024-08-03 22:48:18.999 DEBUG (MainThread) [pymodbus.logging] callback_disconnected called: None
2024-08-03 22:48:19.000 DEBUG (MainThread) [pymodbus.logging] Wait comm 1000000 ms before reconnecting.
[... 23 lines cut ...]
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 167, in create
    await huawei_solar._initialize()
[... cut ...]
  File "/usr/local/lib/python3.12/site-packages/backoff/_async.py", line 37, in _call_handlers
    await handler(details)
  File "/usr/local/lib/python3.12/site-packages/backoff/_async.py", line 16, in f
    return coro_or_func(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 341, in backoff_giveup
    raise ReadException(
huawei_solar.exceptions.ReadException: Failed to read register 43006 after 6 tries

ismiregal17, 2024-08-03:
Unfortunately not yet

This error was caused by a custom integration

Logger: custom_components.huawei-solar.config_flow
Quelle: custom_components/huawei-solar/config_flow.py:410
Integration: huawei-solar
Appeared for the first time: 19:03:43 (7 occurrences)
Last logged: 23:10:20

Unexpected exception while connecting via ModbusTCP
Traceback (most recent call last):
  File "/config/custom_components/huawei-solar/config_flow.py", line 410, in async_step_setup_network
    info = await validate_network_setup(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/huawei-solar/config_flow.py", line 104, in validate_network_setup
    bridge = await HuaweiSolarBridge.create(
                   ^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: HuaweiSolarBridge.create() got an unexpected keyword argument 'host'

dragosmd, 2024-08-04:
@wlcrs not sure if the Emma is ready for it :)
I have a RS485 to ETH mounted RS485A1 B1 for some time on the same port as EMMA is connected to the inverter.
[image]

If I test connecting to RS485 with Slave 0 I get the following output:

2024-08-04 21:09:08.492 DEBUG (MainThread) [pymodbus.logging] Connecting to 192.168.1.6:502.
2024-08-04 21:09:08.492 DEBUG (MainThread) [pymodbus.logging] Connecting comm
2024-08-04 21:09:08.494 DEBUG (MainThread) [pymodbus.logging] Connected to comm
2024-08-04 21:09:08.494 DEBUG (MainThread) [pymodbus.logging] Resetting frame - Current Frame in buffer -
2024-08-04 21:09:09.995 DEBUG (MainThread) [pymodbus.logging] Adding transaction 1
2024-08-04 21:09:09.995 DEBUG (MainThread) [pymodbus.logging] Resetting frame - Current Frame in buffer -
2024-08-04 21:09:09.995 DEBUG (MainThread) [pymodbus.logging] send: 0x0 0x1 0x0 0x0 0x0 0x6 0x0 0x3 0xa7 0xfe 0x0 0x1
2024-08-04 21:09:19.996 DEBUG (MainThread) [pymodbus.logging] Adding transaction 1
2024-08-04 21:09:19.996 DEBUG (MainThread) [pymodbus.logging] Resetting frame - Current Frame in buffer -
2024-08-04 21:09:19.996 DEBUG (MainThread) [pymodbus.logging] send: 0x0 0x1 0x0 0x0 0x0 0x6 0x0 0x3 0xa7 0xfe 0x0 0x1
2024-08-04 21:09:29.996 DEBUG (MainThread) [pymodbus.logging] Adding transaction 1
2024-08-04 21:09:29.997 DEBUG (MainThread) [pymodbus.logging] Resetting frame - Current Frame                                                                                                 in buffer -
2024-08-04 21:09:29.997 DEBUG (MainThread) [pymodbus.logging] send: 0x0 0x1 0x0 0x0 0x0 0x6 0                                                                                                x0 0x3 0xa7 0xfe 0x0 0x1
2024-08-04 21:09:40.000 DEBUG (MainThread) [pymodbus.logging] Adding transaction 1
2024-08-04 21:09:40.000 DEBUG (MainThread) [pymodbus.logging] Resetting frame - Current Frame in buffer -
2024-08-04 21:09:40.000 DEBUG (MainThread) [pymodbus.logging] send: 0x0 0x1 0x0 0x0 0x0 0x6 0x0 0x3 0xa7 0xfe 0x0 0x1
2024-08-04 21:09:50.001 DEBUG (MainThread) [pymodbus.logging] Connection lost comm due to Server not responding
2024-08-04 21:09:50.001 DEBUG (MainThread) [pymodbus.logging] callback_disconnected called: Server not responding
2024-08-04 21:09:50.001 ERROR (MainThread) [huawei_solar.huawei_solar] Aborting client creation due to error.
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 167, in create
    await huawei_solar._initialize()
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 116, in _initialize
    self.time_zone = (await self.get(rn.TIME_ZONE)).value
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 239, in get
    return (await self.get_multiple([name], slave))[0]
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 276, in get_multiple
    response = await self._read_registers(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 419, in _read_registers
    return await _do_read()
           ^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/backoff/_async.py", line 151, in retry
    ret = await target(*args, **kwargs)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/backoff/_async.py", line 151, in retry
    ret = await target(*args, **kwargs)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 367, in _do_read
    response = await self._client.read_holding_registers(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pymodbus/client/base.py", line 183, in async_execute
    raise ModbusIOException(
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] ERROR: No response received after 3 retries

If I test with Slave 0,1
2024-08-04 21:14:30.976 DEBUG (MainThread) [pymodbus.logging] Connecting to 192.168.1.6:502.
2024-08-04 21:14:30.976 DEBUG (MainThread) [pymodbus.logging] Connecting comm
2024-08-04 21:14:30.978 DEBUG (MainThread) [pymodbus.logging] Connected to comm
2024-08-04 21:14:30.978 DEBUG (MainThread) [pymodbus.logging] Resetting frame - Current Frame in buffer -
2024-08-04 21:14:32.479 DEBUG (MainThread) [pymodbus.logging] Adding transaction 1
2024-08-04 21:14:32.479 DEBUG (MainThread) [pymodbus.logging] Resetting frame - Current Frame in buffer -
2024-08-04 21:14:32.479 DEBUG (MainThread) [pymodbus.logging] send: 0x0 0x1 0x0 0x0 0x0 0x6 0x0 0x3 0xa7 0xfe 0x0 0x1
2024-08-04 21:14:42.481 DEBUG (MainThread) [pymodbus.logging] Adding transaction 1
2024-08-04 21:14:42.481 DEBUG (MainThread) [pymodbus.logging] Resetting frame - Current Frame in buffer -
2024-08-04 21:14:42.481 DEBUG (MainThread) [pymodbus.logging] send: 0x0 0x1 0x0 0x0 0x0 0x6 0x0 0x3 0xa7 0xfe 0x0 0x1
2024-08-04 21:14:52.481 DEBUG (MainThread) [pymodbus.logging] Adding transaction 1
2024-08-04 21:14:52.482 DEBUG (MainThread) [pymodbus.logging] Resetting frame - Current Frame in buffer -
2024-08-04 21:14:52.482 DEBUG (MainThread) [pymodbus.logging] send: 0x0 0x1 0x0 0x0 0x0 0x6 0x0 0x3 0xa7 0xfe 0x0 0x1
2024-08-04 21:15:02.483 DEBUG (MainThread) [pymodbus.logging] Adding transaction 1
[... cut ...]
2024-08-04 21:15:58.602 DEBUG (MainThread) [pymodbus.logging] recv: 0x0 0x3 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0x2 old_data:  addr=None
2024-08-04 21:15:58.603 DEBUG (MainThread) [pymodbus.logging] Processing: 0x0 0x3 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0x2
2024-08-04 21:15:58.603 DEBUG (MainThread) [pymodbus.logging] Factory Response[ReadHoldingRegistersResponse': 3]
2024-08-04 21:15:58.603 DEBUG (MainThread) [pymodbus.logging] Getting transaction 3
2024-08-04 21:15:58.603 DEBUG (MainThread) [pymodbus.logging] Very short frame (NO MBAP):  wait for more data
2024-08-04 21:15:58.603 DEBUG (MainThread) [pymodbus.logging] [2]

StefanOltmann, 2024-08-04:
You need to set the client adress to a valid value.
The default `0.0.0.0` did not work for me.

ismiregal17, 2024-08-05:
Hello and what value is that?

dragosmd, 2024-08-05:
@StefanOltmann can you paste your settings , firmware and a working output? I have my doubts since I can't query it directly from .

dragosmd, 2024-08-05:
@StefanOltmann  I've set all kinds of IPs in MODBUS Page
[image]
Emma has 192.168.1.11. Tested with 192.168.1.2 that is HA, 192.168.1.0/ 255.255.255.255 or 0.0.0.0 non of them are working.

So in my case its not related to that TCP IP. Cause as I mentioned I have a rs485 to eth connected directly to emma and that doesn't work also.
I don't say its not working at all. Its not working with the libraries provided by @wlcrs

StefanOltmann, 2024-08-05:
@dragosmd

If HA is running on 192.168.1.2 this is exactly what you should enter as client IP.

Did you try to run my script to see if that is working for you?
Just to come closer to the main issue.

dragosmd, 2024-08-05:
If HA is running on 192.168.1.2 this is exactly what you should enter as client IP. Used that no luck.
If you send me the details on how to run it and what packages I need to install, gladly

StefanOltmann, 2024-08-05:
You can run my script from IDE using IntelliJ IDEA.

andheller123, 2024-08-06:
For me it's working as expected:
Set Client IP to your HA-IP, disable TLS and set the right Slave-ID - in my case which is a default setup it is 2.

dragosmd, 2024-08-06:
@andheller123  where did you extract the slave-id ? none work for me.

dragosmd, 2024-08-06:
If I set slave 0,123 it doesn't work:

2024-08-06 18:44:52.936 DEBUG (MainThread) [pymodbus.logging] Connecting to 192.168.1.11:502.
2024-08-06 18:44:52.936 DEBUG (MainThread) [pymodbus.logging] Connecting comm
2024-08-06 18:44:52.937 DEBUG (MainThread) [pymodbus.logging] Connected to comm
2024-08-06 18:44:52.937 DEBUG (MainThread) [pymodbus.logging] Resetting frame - Current Frame in buffer -
2024-08-06 18:44:54.439 DEBUG (MainThread) [pymodbus.logging] Adding transaction 1
2024-08-06 18:44:54.439 DEBUG (MainThread) [pymodbus.logging] Resetting frame - Current Frame in buffer -
2024-08-06 18:44:54.439 DEBUG (MainThread) [pymodbus.logging] send: 0x0 0x1 0x0 0x0 0x0 0x6 0x0 0x3 0xa7 0xfe 0x0 0x1
2024-08-06 18:44:54.443 DEBUG (MainThread) [pymodbus.logging] recv: 0x0 0x1 0x0 0x0 0x0 0x3 0x0 0x83 0x3 old_data:  addr=None
2024-08-06 18:44:54.443 DEBUG (MainThread) [pymodbus.logging] Processing: 0x0 0x1 0x0 0x0 0x0 0x3 0x0 0x83 0x3
2024-08-06 18:44:54.443 DEBUG (MainThread) [pymodbus.logging] Factory Response[131]
2024-08-06 18:44:54.443 DEBUG (MainThread) [pymodbus.logging] Getting transaction 1
2024-08-06 18:44:54.443 DEBUG (MainThread) [pymodbus.logging] Very short frame (NO MBAP):  wait for more data
2024-08-06 18:44:54.443 ERROR (MainThread) [huawei_solar.huawei_solar] Aborting client creation due to error.
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 167, in create
    await huawei_solar._initialize()
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 116, in _initialize
    self.time_zone = (await self.get(rn.TIME_ZONE)).value
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 239, in get
    return (await self.get_multiple([name], slave))[0]
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 276, in get_multiple
    response = await self._read_registers(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 419, in _read_registers
    return await _do_read()
           ^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/backoff/_async.py", line 151, in retry
    ret = await target(*args, **kwargs)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/backoff/_async.py", line 151, in retry
    ret = await target(*args, **kwargs)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 394, in _do_read
    raise ReadException(
huawei_solar.exceptions.ReadException: Got error while reading from register 43006 with length 1: Exception Response(131, 3, IllegalValue)
2024-08-06 18:45:05.652 DEBUG (MainThread) [pymodbus.logging] Connecting to 192.168.1.11:502.
2024-08-06 18:45:05.652 DEBUG (MainThread) [pymodbus.logging] Connecting comm
2024-08-06 18:45:05.653 DEBUG (MainThread) [pymodbus.logging] Connected to comm
2024-08-06 18:45:05.653 DEBUG (MainThread) [pymodbus.logging] Resetting frame - Current Frame in buffer -
2024-08-06 18:45:07.155 DEBUG (MainThread) [pymodbus.logging] Adding transaction 1
2024-08-06 18:45:07.156 DEBUG (MainThread) [pymodbus.logging] Resetting frame - Current Frame in buffer -
2024-08-06 18:45:07.156 DEBUG (MainThread) [pymodbus.logging] send: 0x0 0x1 0x0 0x0 0x0 0x6 0x0 0x3 0xa7 0xfe 0x0 0x1
2024-08-06 18:45:07.158 DEBUG (MainThread) [pymodbus.logging] recv: 0x0 0x1 0x0 0x0 0x0 0x3 0x0 0x83 0x3 old_data:  addr=None
2024-08-06 18:45:07.158 DEBUG (MainThread) [pymodbus.logging] Processing: 0x0 0x1 0x0 0x0 0x0 0x3 0x0 0x83 0x3
2024-08-06 18:45:07.159 DEBUG (MainThread) [pymodbus.logging] Factory Response[131]
2024-08-06 18:45:07.159 DEBUG (MainThread) [pymodbus.logging] Getting transaction 1
2024-08-06 18:45:07.159 DEBUG (MainThread) [pymodbus.logging] Very short frame (NO MBAP):  wait for more data
2024-08-06 18:45:07.159 ERROR (MainThread) [huawei_solar.huawei_solar] Aborting client creation due to error.
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 167, in create
    await huawei_solar._initialize()
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 116, in _initialize
    self.time_zone = (await self.get(rn.TIME_ZONE)).value
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 239, in get
    return (await self.get_multiple([name], slave))[0]
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 276, in get_multiple
    response = await self._read_registers(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 419, in _read_registers
    return await _do_read()
           ^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/backoff/_async.py", line 151, in retry
    ret = await target(*args, **kwargs)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
[... 43 lines cut ...]
2024-08-06 18:45:22.750 ERROR (MainThread) [homeassistant.components.sensor] Platform huawei_solar does not generate unique IDs. ID HV1234567890_state_2#2 already exists - ignoring sensor.inverter_dsp_data_coll [... cut ...]
2024-08-06 18:45:22.753 ERROR (MainThread) [homeassistant.components.sensor] Platform huawei_solar does not generate unique IDs. ID HV1234567890_storage_fixed_charging_and_discharging_periods already exists - ignoring sensor.batteries_fixed_charging_periods
2024-08-06 18:45:22.753 ERROR (MainThread) [homeassistant.components.sensor] Platform huawei_solar does not generate unique IDs. ID HV1234567890_forcible_charge_discharge_write already exists - ignoring sensor.batteries_forcible_charge
2024-08-06 18:45:22.774 DEBUG (MainThread) [pymodbus.logging] Adding transaction 20

[image]
[image]
[image]

wlcrs, 2024-08-06:
I've just published a new version which works around the different behavior towards datetime registers by EMMA. Can you test again with the lastest commit on the main branch?

dernicc, 2024-08-06:
Could anyone please help me find the location where to enable Modbus TCP?
I guess it is only available in the installer account? My installer did not find the settings.

Is it just in the mobile app or on the web as well?

Screenshots would be perfect.

And thank you in advance.

dragosmd, 2024-08-06:
@wlcrs 
Redownloaded restarted:
2024-08-06 21:13:37.060 DEBUG (MainThread) [pymodbus.logging] Factory Response[ReadHoldingRegistersResponse': 3]
2024-08-06 21:13:37.060 DEBUG (MainThread) [pymodbus.logging] Getting transaction 81
2024-08-06 21:13:37.060 DEBUG (MainThread) [pymodbus.logging] Very short frame (NO MBAP):  wait for more data
2024-08-06 21:13:37.060 DEBUG (MainThread) [pymodbus.logging] [408]
2024-08-06 21:13:37.060 DEBUG (MainThread) [huawei_solar.bridge] Batch update of the following registers: storage_state_of_capacity, storage_running_status,
storage_bus_voltage, storage_bus_current, storage_charge_discharge_power, storage_total_charge, storage_total_discharge, storage_current_day_charge_capacity,
 storage_current_day_discharge_capacity
2024-08-06 21:13:37.060 DEBUG (MainThread) [huawei_solar.huawei_solar] Sleeping for 0.049498 seconds before making next call.
2024-08-06 21:13:37.111 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37760 with length 28 from slave 123
2024-08-06 21:13:37.111 DEBUG (MainThread) [pymodbus.logging] Adding transaction 82
2024-08-06 21:13:37.111 DEBUG (MainThread) [pymodbus.logging] Resetting frame - Current Frame in buffer -

grep for ERROR:
ady exists - ignoring sensor.inverter_inverter_state
2024-08-06 21:15:16.696 ERROR (MainThread) [homeassistant.components.sensor] Platform huawei_solar does not generate unique IDs. ID HV2290299408_state_2#0 already exists - ignoring sensor.inverter_locking_status
2024-08-06 21:15:16.696 ERROR (MainThread) [homeassistant.components.sensor] Platform huawei_solar does not generate unique IDs. ID HV2290299408_state_2#1 already exists - ignoring sensor.inverter_pv_connection_status
2024-08-06 21:15:16.696 ERROR (MainThread) [homeassistant.components.sensor] Platform huawei_solar does not generate unique IDs. ID HV2290299408_state_2#2 already exists - ignoring sensor.inverter_dsp_data_collection
2024-08-06 21:15:16.696 ERROR (MainThread) [homeassistant.components.sensor] Platform huawei_solar does not generate unique IDs. ID HV2290299408_state_3#0 already exists - ignoring sensor.inverter_off_grid_status
2024-08-06 21:15:16.696 ERROR (MainThread) [homeassistant.components.sensor] Platform huawei_solar does not generate unique IDs. ID HV2290299408_state_3#1 already exists - ignoring sensor.inverter_off_grid_switch
2024-08-06 21:15:16.696 ERROR (MainThread) [homeassistant.components.sensor] Platform huawei_solar does not generate unique IDs. ID HV2290299408_ALARMS already exists - ignoring sensor.inverter_alarms
2024-08-06 21:15:16.696 ERROR (MainThread) [homeassistant.components.sensor] Platform huawei_solar does not generate unique IDs. ID HV2290299408_pv_01_voltage already exists - ignoring sensor.inverter_pv_1_voltage
2024-08-06 21:15:16.697 ERROR (MainThread) [homeassistant.components.sensor] Platform huawei_solar does not generate unique IDs. ID HV2290299408_pv_01_current already exists - ignoring sensor.inverter_pv_1_current
2024-08-06 21:15:16.697 ERROR (MainThread) [homeassistant.components.sensor] Platform huawei_solar does not generate unique IDs. ID HV2290299408_pv_02_voltage already exists - ignoring sensor.inverter_pv_2_voltage
2024-08-06 21:15:16.697 ERROR (MainThread) [homeassistant.components.sensor] Platform huawei_solar does not generate unique IDs. ID HV2290299408_pv_02_current already exists - ignoring sensor.inverter_pv_2_current
2024-08-06 21:15:16.698 ERROR (MainThread) [homeassistant.components.sensor] Platform huawei_solar does not generate unique IDs. ID HV2290299408_storage_state_of_capacity already exists - ignoring sensor.batteries_state_of_capacity
2024-08-06 21:15:16.698 ERROR (MainThread) [homeassistant.components.sensor] Platform huawei_solar does not generate unique IDs. ID HV2290299408_storage_running_status already exists - ignoring sensor.batteries_status
2024-08-06 21:15:16.698 ERROR (MainThread) [homeassistant.components.sensor] Platform huawei_solar does not generate unique IDs. ID HV2290299408_storage_bus_voltage already exists - ignoring sensor.batteries_bus_voltage
2024-08-06 21:15:16.698 ERROR (MainThread) [homeassistant.components.sensor] Platform huawei_solar does not generate unique IDs. ID HV2290299408_storage_bus_current already exists - ignoring sensor.batteries_bus_current
2024-08-06 21:15:16.698 ERROR (MainThread) [homeassistant.components.sensor] Platform huawei_solar does not generate unique IDs. ID HV2290299408_storage_charge_discharge_power already exists - ignoring sensor.batteries_charge_discharge_power
2024-08-06 21:15:16.699 ERROR (MainThread) [homeassistant.components.sensor] Platform huawei_solar does not generate unique IDs. ID HV2290299408_storage_total_charge already exists - ignoring sensor.batteries_total_charge
2024-08-06 21:15:16.699 ERROR (MainThread) [homeassistant.components.sensor] Platform huawei_solar does not generate unique IDs. ID HV2290299408_storage_total_discharge already exists - ignoring sensor.batteries_total_discharge
2024-08-06 21:15:16.699 ERROR (MainThread) [homeassistant.components.sensor] Platform huawei_solar does not generate unique IDs. ID HV2290299408_storage_current_day_charge_capacity already exists - ignoring sensor.batteries_day_charge
2024-08-06 21:15:16.699 ERROR (MainThread) [homeassistant.components.sensor] Platform huawei_solar does not generate unique IDs. ID HV2290299408_storage_current_day_discharge_capacity already exists - ignoring sensor.batteries_day_discharge
[... cut ...]
    state = self._stringify_state(available)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1013, in _stringify_state
    if (state := self.state) is None:
                 ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 586, in state
    raise ValueError(
ValueError: Invalid datetime: sensor.inverter_startup_time provides state '2024-08-06 02:11:16', which is missing timezone information
[image]

[image]

dragosmd, 2024-08-06:
@wlcrs 2nd Update. Deleted the folder, unzip again. added Slave: 0,123
Emma appeared:
[image]

Entities still not being populated:
[image]

[image]
[image]

2024-08-06 21:27:39.002 DEBUG (MainThread) [huawei_solar.huawei_solar] Sleeping for 0.048491 seconds before making next call.
2024-08-06 21:27:39.003 ERROR (MainThread) [custom_components.huawei_solar] Unexpected exception from <bound method DataUpdateCoordinator.async_refresh of <custom_components.huawei_solar.update_coordinator.HuaweiSolarUpdateCoordinator object at 0x7f13c9fcdfa0>>
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/debounce.py", line 137, in _handle_timer_finish
    await task
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 288, in async_refresh
    await self._async_refresh(log_failures=True)
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 411, in _async_refresh
    self.async_update_listeners()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 165, in async_update_listeners
    update_callback()
  File "/config/custom_components/huawei_solar/sensor.py", line 1378, in _handle_coordinator_update
    self._attr_extra_state_attributes.clear()
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'HuaweiSolarFixedChargingPeriodsSensorEntity' object has no attribute '__attr_extra_state_attributes'. Did you mean: '_attr_extra_state_attributes'?
2024-08-06 21:27:39.051 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37760 with length 28 from slave 123
2024-08-06 21:27:39.051 DEBUG (MainThread) [pymodbus.logging] Adding transaction 23

2024-08-06 21:37:16.341 DEBUG (MainThread) [custom_components.huawei_solar] Finished fetching HV2290299408_inverter_data_update_coordinator data in 0.087 seconds (success: True)
2024-08-06 21:37:16.344 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved (None)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 255, in _handle_refresh_interval
    await self._async_refresh(log_failures=True, scheduled=True)
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 411, in _async_refresh
    self.async_update_listeners()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 165, in async_update_listeners
    update_callback()
  File "/config/custom_components/huawei_solar/sensor.py", line 1105, in _handle_coordinator_update
    self.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1007, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1133, in _async_write_ha_state
    state, attr, capabilities, shadowed_attr = self.__async_calculate_state()
                                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1068, in __async_calculate_state
    state = self._stringify_state(available)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1013, in _stringify_state
    if (state := self.state) is None:
                 ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 586, in state
    raise ValueError(
ValueError: Invalid datetime: sensor.inverter_startup_time provides state '2024-08-06 02:11:16', which is missing timezone information

wlcrs, 2024-08-06:
Can you please test again with the latest changes?

(yes, this is annoying. This is why I always said that I didn't want to develop against devices that I cannot directly access).

dragosmd, 2024-08-06:
annoying no :) cause I'm searching for emma features so its my interest to help. I have another specialty in this domain but  I can help how I can.
Redownloaded. Logs and pictures attached. Battery entities especially numbers still empty.
Inverter PV1, On grid fixed.
as for later Emma registry can I help add some of them? with some guidance as well? I would like so info from it and the on-grid / off-grid switch for example

2024-08-06 22:13:00.452 DEBUG (MainThread) [pymodbus.logging] Connecting to 192.168.1.11:502.
2024-08-06 22:13:00.452 DEBUG (MainThread) [pymodbus.logging] Connecting comm
2024-08-06 22:13:00.453 DEBUG (MainThread) [pymodbus.logging] Connecting to 192.168.200.1:6607.
2024-08-06 22:13:00.453 DEBUG (MainThread) [pymodbus.logging] Connecting comm
2024-08-06 22:13:00.474 DEBUG (MainThread) [pymodbus.logging] Connected to comm
2024-08-06 22:13:00.474 DEBUG (MainThread) [pymodbus.logging] Resetting frame - Current Frame in buffer -
2024-08-06 22:13:00.475 DEBUG (MainThread) [pymodbus.logging] Connected to comm
2024-08-06 22:13:00.475 DEBUG (MainThread) [pymodbus.logging] Resetting frame - Current Frame in buffer -

2024-08-06 22:13:01.975 DEBUG (MainThread) [pymodbus.logging] Adding transaction 1

2024-08-06 22:13:02.714 DEBUG (MainThread) [pymodbus.logging] Very short frame (NO MBAP):  wait for more data
2024-08-06 22:13:02.714 DEBUG (MainThread) [pymodbus.logging] [0]
2024-08-06 22:13:02.740 DEBUG (MainThread) [pymodbus.logging] Adding transaction 15
2024-08-06 22:13:02.740 DEBUG (MainThread) [pymodbus.logging] Resetting frame - Current Frame in buffer -
2024-08-06 22:13:02.740 DEBUG (MainThread) [pymodbus.logging] send: 0x0 0xf 0x0 0x0 0x0 0x6 0x7b 0x3 0xb8 0xc3 0x0 0x1
2024-08-06 22:13:02.742 DEBUG (MainThread) [pymodbus.logging] recv: 0x0 0xf 0x0 0x0 0x0 0x5 0x7b 0x3 0x2 0x0 0x1 old_data:  addr=None
2024-08-06 22:13:02.742 DEBUG (MainThread) [pymodbus.logging] Processing: 0x0 0xf 0x0 0x0 0x0 0x5 0x7b 0x3 0x2 0x0 0x1
2024-08-06 22:13:02.742 DEBUG (MainThread) [pymodbus.logging] Factory Response[ReadHoldingRegistersResponse': 3]
2024-08-06 22:13:02.743 DEBUG (MainThread) [pymodbus.logging] Getting transaction 15
2024-08-06 22:13:02.743 DEBUG (MainThread) [pymodbus.logging] Very short frame (NO MBAP):  wait for more data
2024-08-06 22:13:02.743 DEBUG (MainThread) [pymodbus.logging] [1]
2024-08-06 22:13:02.751 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.battery_1_day_discharge (<class 'custom_components.huawei_solar.sensor.HuaweiSolarSensorEntity'>) is using state class 'total_increasing' which is impossible considering device class ('energy_storage') it is using; expected None or one of 'measurement'; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/wlcrs/huawei_solar/issues
2024-08-06 22:13:02.765 DEBUG (MainThread) [pymodbus.logging] Adding transaction 9

2024-08-06 22:13:03.114 DEBUG (MainThread) [pymodbus.logging] Very short frame (NO MBAP):  wait for more data
2024-08-06 22:13:03.165 DEBUG (MainThread) [pymodbus.logging] Adding transaction 12
2024-08-06 22:13:03.165 DEBUG (MainThread) [pymodbus.logging] Resetting frame - Current Frame in buffer -
2024-08-06 22:13:03.165 DEBUG (MainThread) [pymodbus.logging] send: 0x0 0xc 0x0 0x0 0x0 0x3f 0x0 0x41 0x25 0x3b 0x14 0x9d 0x7f 0xab 0x1e 0x4e 0x5e 0x4b 0x38 0x49 0xa5 0x86 0x93 0xbc 0x85 0xf7 0x9 0x69 0x6e 0x73 0x74 0x61 0x6c 0x6c 0x65 0x72 0x20 0xf3 0xe4 0x10 0xdb 0x7d 0x66 0x83 0xed 0xd1 0x26 0xb1 0x4d 0xe6 0x5a 0x8e 0x63 0xdf 0x6e 0xf6 0xb4 0x29 0x8d 0xc1 0x8c 0xcf 0x74 0xe6 0x46 0x14 0xb3 0x2a 0x6e
2024-08-06 22:13:05.197 DEBUG (MainThread) [pymodbus.logging] recv: 0x0 0xc 0x0 0x0 0x0 0x58 0x0 0x41 0x25 0x54 0x0 0x20 0xd9 0xce 0x98 0xed 0x2 0x85 0x67 0x67 0x5 0xcd 0x16 0xad 0x31 0x11 0x9f 0xea 0x2a 0xd7 0x93 0xcf 0x6b 0xbd 0xa8 0x3c 0xe6 0x6b 0xd4 0xf8 0x13 0x56 0xb8 0xd1 0x1 0x0 0xc9 0x47 0xb4 0x4b 0xc7 0xeb 0x6f 0x55 0xef 0x88 0x5b 0xf 0x35 0x9 0xa6 0x64 0x7a 0x94 0xdc 0x2 0x73 0x6d 0x24 0x33 0xd 0x69 0x37 0x9a 0x25 0x8c 0x2 0xe 0xff 0x56 0x64 0x70 0xc1 0xa6 0x30 0xe1 0x27 0x6 0xa0 0xe5 0x35 0xea 0xb0 0xd6 old_data:  addr=None
2024-08-06 22:13:05.197 DEBUG (MainThread) [pymodbus.logging] Processing: 0x0 0xc 0x0 0x0 0x0 0x58 0x0 0x41 0x25 0x54 0x0 0x20 0xd9 0xce 0x98 0xed 0x2 0x85 0x67 0x67 0x5 0xcd 0x16 0xad 0x31 0x11 0x9f 0xea 0x2a 0xd7 0x93 0xcf 0x6b 0xbd 0xa8 0x3c 0xe6 0x6b 0xd4 0xf8 0x13 0x56 0xb8 0xd1 0x1 0x0 0xc9 0x47 0xb4 0x4b 0xc7 0xeb 0x6f 0x55 0xef 0x88 0x5b 0xf 0x35 0x9 0xa6 0x64 0x7a 0x94 0xdc 0x2 0x73 0x6d 0x24 0x33 0xd 0x69 0x37 0x9a 0x25 0x8c 0x2 0xe 0xff 0x56 0x64 0x70 0xc1 0xa6 0x30 0xe1 0x27 0x6 0xa0 0xe5 0x35 0xea 0xb0 0xd6
2024-08-06 22:13:05.198 DEBUG (MainThread) [pymodbus.logging] Factory Response[PrivateHuaweiModbusResponse': 65]
2024-08-06 22:13:05.198 DEBUG (MainThread) [pymodbus.logging] Getting transaction 12
2024-08-06 22:13:05.198 DEBUG (MainThread) [pymodbus.logging] Very short frame (NO MBAP):  wait for more data
2024-08-06 22:13:05.199 ERROR (MainThread) [homeassistant.components.select] Platform huawei_solar does not generate unique IDs. ID HV2290299408_storage_excess_pv_energy_use_in_tou already exists - ignoring select.batteries_excess_pv_energy_use_in_tou
2024-08-06 22:13:05.199 ERROR (MainThread) [homeassistant.components.select] Platform huawei_solar does not generate unique IDs. ID HV2290299408_storage_working_mode_settings already exists - ignoring select.batteries_working_mode
[... cut ...]
2024-08-06 22:13:05.791 ERROR (MainThread) [homeassistant.components.number] Platform huawei_solar does not generate unique IDs. ID HV2290299408_storage_power_of_charge_from_grid already exists - ignoring number.batteries_grid_charge_maximum_power

[image]

2024-08-06 22:13:05.790 ERROR (MainThread) [homeassistant.components.number] Platform huawei_solar does not generate unique IDs. ID HV2290299408_active_power_fixed_value_derating already exists - ignoring number.inverter_power_derating

[image]

tmycgn, 2024-08-06:
@wlcrs I'm still getting my Emma installed, when it's connected I'll offer my remote access if technically possible. 🤙

wlcrs, 2024-08-06:
Again a new version to test. I improved how I test for write-capabilities. Still no clue on how EMMA handles this: do I need to do the same "magic" modbus commands to perform the login sequence as I need to do for the inverters, or will it just be enabled without needing to enter the installer credentials?

Other notable change: I've added slave auto-discovery as documented in the EMMA modbus definitions pdf. Luckily my SUN2000 inverter does also support those special messages.

Can you remove and re-install the integration? Please leave the slave id field set to 'AUTO' to test this feature.

Please also add the following to your configurtion.yaml to make some things a bit more verbose:

```yaml
logger:
  default: info
  logs:
    custom_components.huawei_solar: debug
    huawei_solar: debug
```

wlcrs, 2024-08-06:
@dragosmd Also, as your logs indicate that your HA is in an inconsistent state because of half-populated sensors, can you please:
- uninstall the integration
- go to Settings > Devices&services > Entities
- Remove all the entities associated with Huawei Solar from there
- only then install the integration again?

wlcrs, 2024-08-06:
> as for later Emma registry can I help add some of them? with some guidance as well? I would like the info from some registry and the on-grid / off-grid switch for example

Sensors needs to be defined on 2 locations: in the underlying library, and in the HA integration.

In the underlying library @Roving-Ronin has already provided some helpful input: https://github.com/Roving-Ronin/huawei-solar-lib/blob/dfd6f5dce7d20052c3a35e01994bfa2c1a866950/src/huawei_solar/registers.py#L1302

For the moment, only a part of his work [has already been integrated](https://gitlab.com/Emilv2/huawei-solar/-/blob/feature/emma/src/huawei_solar/registers.py?ref_type=heads#L1357), as I wanted to verify that we are able to connect to EMMA to begin with. As you have now validated that connecting via Modbus TCP actually works, we can continue implementing these registers.

The second part is defining the sensors in the integration. The first few sensors were defined here: https://github.com/wlcrs/huawei_solar/blob/74bdd52c9be23a677e01827b1b5a909b6e7c1d33/sensor.py#L961

dragosmd, 2024-08-06:
AUTO slave:

2024-08-06 23:00:38.664 DEBUG (MainThread) [pymodbus.logging] Connecting to 192.168.1.11:502.
2024-08-06 23:00:38.664 DEBUG (MainThread) [pymodbus.logging] Connecting comm
2024-08-06 23:00:38.665 DEBUG (MainThread) [pymodbus.logging] Connected to comm
2024-08-06 23:00:38.665 DEBUG (MainThread) [pymodbus.logging] Resetting frame - Current Frame in buffer -
2024-08-06 23:00:40.166 DEBUG (MainThread) [pymodbus.logging] Adding transaction 1
2024-08-06 23:00:40.167 DEBUG (MainThread) [pymodbus.logging] Resetting frame - Current Frame in buffer -
2024-08-06 23:00:40.167 DEBUG (MainThread) [pymodbus.logging] send: 0x0 0x1 0x0 0x0 0x0 0x6 0x0 0x3 0x75 0x30 0x0 0x41
2024-08-06 23:00:40.170 DEBUG (MainThread) [pymodbus.logging] recv: 0x0 0x1 0x0 0x0 0x0 0x85 0x0 0x3 0x82 0x53 0x6d 0x61 0x72 0x74 0x48 0x45 0x4d 0x53 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x4e 0x53 0x32 0x34 0x34 0x31 0x33 0x33 0x31 0x30 0x38 0x37 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x53 0x6d 0x61 0x72 0x74 0x48 0x45 0x4d 0x53 0x20 0x56 0x31 0x30 0x30 0x52 0x30 0x32 0x34 0x43 0x30 0x30 0x53 0x50 0x43 0x31 0x30 0x30 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 old_data:  addr=None
2024-08-06 23:00:40.170 DEBUG (MainThread) [pymodbus.logging] Processing: 0x0 0x1 0x0 0x0 0x0 0x85 0x0 0x3 0x82 0x53 0x6d 0x61 0x72 0x74 0x48 0x45 0x4d 0x53 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x4e 0x53 0x32 0x34 0x34 0x31 0x33 0x33 0x31 0x30 0x38 0x37 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x53 0x6d 0x61 0x72 0x74 0x48 0x45 0x4d 0x53 0x20 0x56 0x31 0x30 0x30 0x52 0x30 0x32 0x34 0x43 0x30 0x30 0x53 0x50 0x43 0x31 0x30 0x30 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0
2024-08-06 23:00:40.170 DEBUG (MainThread) [pymodbus.logging] Factory Response[ReadHoldingRegistersResponse': 3]
2024-08-06 23:00:40.170 DEBUG (MainThread) [pymodbus.logging] Getting transaction 1
2024-08-06 23:00:40.170 DEBUG (MainThread) [pymodbus.logging] Very short frame (NO MBAP):  wait for more data
2024-08-06 23:00:40.170 DEBUG (MainThread) [pymodbus.logging] [21357, 24946, 29768, 17741, 21248, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 20051, 12852, 13361, 13107, 12592, 14391, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 21357, 24946, 29768, 17741, 21280, 22065, 12336, 21040, 12852, 17200, 12371, 20547, 12592, 12288, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
2024-08-06 23:00:40.171 WARNING (MainThread) [huawei_solar.bridge] TODO: EMMA product model is not known at the moment. Assuming that it is SmartHEMS.
2024-08-06 23:00:40.221 DEBUG (MainThread) [pymodbus.logging] Adding transaction 2
2024-08-06 23:00:40.221 DEBUG (MainThread) [pymodbus.logging] Resetting frame - Current Frame in buffer -
2024-08-06 23:00:40.221 DEBUG (MainThread) [pymodbus.logging] send: 0x0 0x2 0x0 0x0 0x0 0x6 0x0 0x3 0x76 0xe 0x0 0x14
2024-08-06 23:00:40.223 DEBUG (MainThread) [pymodbus.logging] recv: 0x0 0x2 0x0 0x0 0x0 0x2b 0x0 0x3 0x28 0x45 0x4d 0x4d 0x41 0x2d 0x41 0x30 0x32 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 old_data:  addr=None
2024-08-06 23:00:40.224 DEBUG (MainThread) [pymodbus.logging] Processing: 0x0 0x2 0x0 0x0 0x0 0x2b 0x0 0x3 0x28 0x45 0x4d 0x4d 0x41 0x2d 0x41 0x30 0x32 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0
2024-08-06 23:00:40.224 DEBUG (MainThread) [pymodbus.logging] Factory Response[ReadHoldingRegistersResponse': 3]
2024-08-06 23:00:40.224 DEBUG (MainThread) [pymodbus.logging] Getting transaction 2
2024-08-06 23:00:40.224 DEBUG (MainThread) [pymodbus.logging] Very short frame (NO MBAP):  wait for more data
2024-08-06 23:00:40.224 DEBUG (MainThread) [pymodbus.logging] [17741, 19777, 11585, 12338, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
2024-08-06 23:00:40.224 DEBUG (MainThread) [pymodbus.logging] Adding transaction 3
2024-08-06 23:00:40.224 DEBUG (MainThread) [pymodbus.logging] Resetting frame - Current Frame in buffer -
2024-08-06 23:00:40.224 DEBUG (MainThread) [pymodbus.logging] send: 0x0 0x3 0x0 0x0 0x0 0x5 0x0 0x2b 0xe 0x3 0x87
2024-08-06 23:00:40.227 DEBUG (MainThread) [pymodbus.logging] recv: 0x0 0x3 0x0 0x0 0x0 0xb 0x0 0x2b 0xe 0x3 0x3 0xff 0x88 0x6 0x87 0x1 0x5 old_data:  addr=None
2024-08-06 23:00:40.227 DEBUG (MainThread) [pymodbus.logging] Processing: 0x0 0x3 0x0 0x0 0x0 0xb 0x0 0x2b 0xe 0x3 0x3 0xff 0x88 0x6 0x87 0x1 0x5
2024-08-06 23:00:40.227 DEBUG (MainThread) [pymodbus.logging] Factory Response[ReadDeviceIdentifierResponse': 43]
2024-08-06 23:00:40.227 ERROR (MainThread) [pymodbus.logging] General exception: unpack_from requires a buffer of at least 11 bytes for unpacking 2 bytes at offset 9 (actual buffer size is 9)
2024-08-06 23:00:40.227 DEBUG (MainThread) [pymodbus.logging] Resetting frame - Current Frame in buffer - 0x0 0x3 0x0 0x0 0x0 0xb 0x0 0x2b 0xe 0x3 0x3 0xff 0x88 0x6 0x87 0x1 0x5
2024-08-06 23:00:40.227 ERROR (MainThread) [homeassistant] Error doing job: Fatal error: protocol.data_received() call failed. (None)
Traceback (most recent call last):
[... cut ...]
2024-08-06 23:13:05.623 DEBUG (MainThread) [pymodbus.logging] recv: 0x0 0x27 0x0 0x0 0x0 0x2b 0x0 0x3 0x28 0x0 0x6 0x0 0x2 0x0 0x5 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0xe 0x81 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 old_data:  addr=None
2024-08-06 23:13:05.623 DEBUG (MainThread) [pymodbus.logging] Processing: 0x0 0x27 0x0 0x0 0x0 0x2b 0x0 0x3 0x28 0x0 0x6 0x0 0x2 0x0 0x5 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0xe 0x81 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0

AUTO doesn't work. After adding EMMA with Slaves 0,123 the above errors are seen again but they work.

dragosmd, 2024-08-06:
@wlcrs didn't see the last debug comment. I'll redo everything from my pc tomorrow and paste the logs here.

andheller123, 2024-08-06:
@wlcrs Hi, great work so far, thx a lot. I can get you access to my Emma via VPN if needed, just lmk

dragosmd, 2024-08-07:
@wlcrs  Morning, here is the debug session
### **AUTO** **- No Elevated**

2024-08-07 09:18:46.082 INFO (MainThread) [huawei_solar.huawei_solar] Waiting for connection ...
2024-08-07 09:18:47.583 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 30000 with length 65 from slave 0
2024-08-07 09:18:47.586 WARNING (MainThread) [huawei_solar.bridge] TODO: EMMA product model is not known at the moment. Assuming that it is SmartHEMS.
2024-08-07 09:18:47.587 DEBUG (MainThread) [huawei_solar.huawei_solar] Sleeping for 0.049150 seconds before making next call.
2024-08-07 09:18:47.637 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 30222 with length 20 from slave 0
2024-08-07 09:18:47.642 INFO (MainThread) [custom_components.huawei_solar.config_flow] Successfully connected to inverter SmartHEMS with SN NS2441331087
2024-08-07 09:18:47.645 ERROR (MainThread) [pymodbus.logging] General exception: unpack_from requires a buffer of at least 11 bytes for unpacking 2 bytes at offset 9 (actual buffer size is 9)
2024-08-07 09:18:47.645 ERROR (MainThread) [homeassistant] Error doing job: Fatal error: protocol.data_received() call failed. (None)
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/asyncio/selector_events.py", line 1017, in _read_ready__data_received
    self._protocol.data_received(data)
  File "/usr/local/lib/python3.12/site-packages/pymodbus/transport/transport.py", line 302, in data_received
    self.datagram_received(data, None)
  File "/usr/local/lib/python3.12/site-packages/pymodbus/transport/transport.py", line 336, in datagram_received
    cut = self.callback_data(self.recv_buffer, addr=addr)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pymodbus/client/base.py", line 207, in callback_data
    self.framer.processIncomingPacket(data, self._handle_response, slave=0)
  File "/usr/local/lib/python3.12/site-packages/pymodbus/framer/base.py", line 139, in processIncomingPacket
    self.frameProcessIncomingPacket(single, callback, slave, **kwargs)
  File "/usr/local/lib/python3.12/site-packages/pymodbus/framer/socket_framer.py", line 91, in frameProcessIncomingPacket
    raise ModbusIOException("Unable to decode request")
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Unable to decode request
2024-08-07 09:18:57.644 ERROR (MainThread) [pymodbus.logging] Cancel send, because not connected!
2024-08-07 09:18:57.644 ERROR (MainThread) [custom_components.huawei_solar.config_flow] Unexpected exception while connecting via ModbusTCP
Traceback (most recent call last):
  File "/config/custom_components/huawei_solar/config_flow.py", line 499, in async_step_setup_network
    info = await validate_network_setup_auto_slave_discovery(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/huawei_solar/config_flow.py", line 122, in validate_network_setup_auto_slave_discovery
    device_infos = await bridge.client.get_device_infos()
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 464, in get_device_infos
    response = await self._client.execute(ReadDeviceIdentifierRequest(read_dev_id_code=0x03, object_id=0x87))
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pymodbus/client/base.py", line 175, in async_execute
    resp = await asyncio.wait_for(
           ^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/asyncio/tasks.py", line 520, in wait_for
    return await fut
           ^^^^^^^^^
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Client is not connected

### **AUTO ELEVATED**
2024-08-07 09:21:01.969 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 40490 with length 1 from slave 0
2024-08-07 09:21:01.972 ERROR (MainThread) [custom_components.huawei_solar.config_flow] Unexpected exception while connecting via ModbusTCP
Traceback (most recent call last):
  File "/config/custom_components/huawei_solar/config_flow.py", line 499, in async_step_setup_network
    info = await validate_network_setup_auto_slave_discovery(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/huawei_solar/config_flow.py", line 120, in validate_network_setup_auto_slave_discovery
    result["has_write_permission"] = await bridge.has_write_permission()
                                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/bridge.py", line 300, in has_write_permission
    await self.client.set(self._write_test_register, result.value, self.slave_id)
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 630, in set
    raise WriteException("Register is not writable")
huawei_solar.exceptions.WriteException: Register is not writable

### **Manual Slaves 0,123**

2024-08-07 09:21:55.203 DEBUG (MainThread) [huawei_solar.modbus] Waiting for 1500 milliseconds after connection before performing operations
2024-08-07 09:21:55.204 INFO (MainThread) [huawei_solar.huawei_solar] Waiting for connection ...
2024-08-07 09:21:56.704 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 30000 with length 65 from slave 0
[... cut ...]
    result["has_write_permission"] = await bridge.has_write_permission()
                                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/bridge.py", line 300, in has_write_permission
    await self.client.set(self._write_test_register, result.value, self.slave_id)
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 630, in set
    raise WriteException("Register is not writable")
huawei_solar.exceptions.WriteException: Register is not writable

ismiregal17, 2024-08-07:
Good morning, unfortunately nothing is connecting for me either. I have set all the parameters as described. 
If you like, I can give you remote access to my HA.

dragosmd, 2024-08-07:
@wlcrs 
I've setup a HA + EMMA besides it SSH+Web available.
How do I pass the credentials?

Tried to installed main and leave emma added:

024-08-07 13:45:38.841 WARNING (MainThread) [homeassistant.util.loop] Detected blocking call to import_module with args ('custom_components.huawei_solar.config_flow',) inside the event loop by integration 'config' at homeassistant/components/config/config_entries.py, line 179: return await super()._post_impl(request, data) (offender: /usr/src/homeassistant/homeassistant/loader.py, line 1296: return importlib.import_module(f"{self.pkg_path}.{platform_name}")), please create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+config%22
For developers, please see https://developers.home-assistant.io/docs/asyncio_blocking_operations/#import_module
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/usr/src/homeassistant/homeassistant/__main__.py", line 223, in <module>
    sys.exit(main())
  File "/usr/src/homeassistant/homeassistant/__main__.py", line 209, in main
    exit_code = runner.run(runtime_conf)
  File "/usr/src/homeassistant/homeassistant/runner.py", line 190, in run
    return loop.run_until_complete(setup_and_run_hass(runtime_config))
  File "/usr/local/lib/python3.12/asyncio/base_events.py", line 674, in run_until_complete
    self.run_forever()
  File "/usr/local/lib/python3.12/asyncio/base_events.py", line 641, in run_forever
    self._run_once()
  File "/usr/local/lib/python3.12/asyncio/base_events.py", line 1990, in _run_once
    handle._run()
  File "/usr/local/lib/python3.12/asyncio/events.py", line 88, in _run
    self._context.run(self._callback, *self._args)
  File "/usr/local/lib/python3.12/site-packages/aiohttp/web_protocol.py", line 452, in _handle_request
    resp = await request_handler(request)
  File "/usr/local/lib/python3.12/site-packages/aiohttp/web_app.py", line 543, in _handle
    resp = await handler(request)
  File "/usr/local/lib/python3.12/site-packages/aiohttp/web_middlewares.py", line 114, in impl
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/security_filter.py", line 92, in security_filter_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/forwarded.py", line 83, in forwarded_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/request_context.py", line 26, in request_context_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/ban.py", line 85, in ban_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/auth.py", line 242, in auth_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/headers.py", line 32, in headers_middleware
    response = await handler(request)
  File "/usr/src/homeassistant/homeassistant/helpers/http.py", line 73, in handle
    result = await handler(request, **request.match_info)
  File "/usr/src/homeassistant/homeassistant/components/http/decorators.py", line 81, in with_admin
    return await func(self, request, *args, **kwargs)
  File "/usr/src/homeassistant/homeassistant/components/http/data_validator.py", line 74, in wrapper
    return await method(view, request, data, *args, **kwargs)
  File "/usr/src/homeassistant/homeassistant/components/config/config_entries.py", line 172, in post
    return await self._post_impl(request, data)
  File "/usr/src/homeassistant/homeassistant/components/config/config_entries.py", line 179, in _post_impl
    return await super()._post_impl(request, data)

2024-08-07 13:45:38.852 ERROR (MainThread) [homeassistant.config_entries] Error occurred loading flow for integration huawei_solar: cannot import name 'HuaweiSolarBridge' from partially initialized module 'huawei_solar' (most likely due to a circular import) (/config/huawei_solar/__init__.py)

wlcrs, 2024-08-07:
> @wlcrs Hi, great work so far, thx a lot. I can get you access to my Emma via VPN if needed, just lmk
@andheller123 
That would be very useful! Can you send the authentication details + connection details to the EMMA to wlcrs@outlook.com ?

dragosmd, 2024-08-07:
@wlcrs mail sent

andheller123, 2024-08-07:
> > @wlcrs Hi, great work so far, thx a lot. I can get you access to my Emma via VPN if needed, just lmk
> 
> @andheller123 That would be very useful! Can you send the authentication details + connection details to the EMMA to my username at outlook.com ?

Linux, Mac, Windows ?

wlcrs, 2024-08-07:
I'm already set with the SSH access that Dragos gave me, thanks anyway. 

With the latest commit in this repo, auto discovery works and elevated permissions also work (no installer credentials needed anymore for EMMA devices :-) )
Not everything works yet: it seems that some queries are now failing, not sure yet what the cause is. I'll continue debugging in the next few days when I find a moment to spare.

ismiregal17, 2024-08-07:
THX, great work. Modbus tcp only work via Ethernet or also via wifi?

[104 more replies]

closed 2024-10-14 by wlcrs