wlcrs/huawei_solar #499: [Feature Request]: Huawei SmartGuard-63A- S0/AUS0

Roving-Ronin, 2023-10-09:
### Describe your feature request

As per the EMMA feature request, this 'issue' is intended as a placeholder to allow the gathering of information and discussion re the forthcoming SmartGuard-63A- S0/AUS0 (also referred to as 'BackupBoxA' on Huawei Support Site), that will supercede the current BackupBox device, that provides automatic grid outage failover for LUNA ESS users, with the provision of a single protected phase.

As per the other new products, this will require the release of the modbus register definitions for this product, at which stage we can see if they are the same as the current BackupBox, potentially meaning WLCRS will be 'out of the box'  compatible with the SmartGuard, or if additional effort would be required to support it. At which point review of the registers would need to be done and put forward to see if any provide any useful functionality, or not.

Documentation available currently:
[BackupBoxA V100R023C10 Communication Matrix.pdf](https://github.com/wlcrs/huawei_solar/files/12842596/BackupBoxA.V100R023C10.Communication.Matrix.pdf)
[BackupBoxA V100R023C10 Open Source Software Notice.docx](https://github.com/wlcrs/huawei_solar/files/12842597/BackupBoxA.V100R023C10.Open.Source.Software.Notice.docx)
[BackupBoxA V100R023C10 Release Notes.pdf](https://github.com/wlcrs/huawei_solar/files/12842598/BackupBoxA.V100R023C10.Release.Notes.pdf)
[BackupBoxA V100R023C10 Security Maintenance Manual.pdf](https://github.com/wlcrs/huawei_solar/files/12842593/BackupBoxA.V100R023C10.Security.Maintenance.Manual.pdf)
[BackupBoxA V100R023C10 Upgrade Guide.pdf](https://github.com/wlcrs/huawei_solar/files/12842594/BackupBoxA.V100R023C10.Upgrade.Guide.pdf)

Note: From the communication matrix above, it already confirms that the SmartGuard connects via RS485 to the EMMA device, that in turn connects via RS485 to the inverter(s) / LUNA. Additionally the SmartGuard enclosure includes a DIN rail for mounting the EMMA inside of it (though you can install EMMA without a SmartGuard, for non LUNA users). From this it appears that in future installations of LUNA that want to operate in an off-grid/outage situation, will be expected to deploy Inverter(s)+LUNA ESS+EMMA+SmartGuard by default.

### Proper usage

- [X] I confirm that this is not a bug report or support request
- [X] I confirm that this feature request is within the stated scope of the integration

Roving-Ronin, 2024-03-07:
Some new information released at: https://solar.huawei.com/en/professionals/all-products/HUAWEI-SmartGuard

Instead of previously having to deploy a Backup-Box and a DDSU/DTSU-666H meter (and optional SmartLogger), the EMMA product is now included as part of the SmartGuard, providing 'Smart' functions and also the ability to shed load (via Shelly PM/Pro devices). Looking at the User Manual and the wiring diagrams, this however appears to remove the option of using CT's to monitor the power (as the standalone EMMA allows for), with the SmartGuard having the inverter, backed up and non backed up loads and a connection for the grid to be connected to it, Huawei designed it so the EMMA is intetgrated into these connections.

[image]

[image]

menyhartda, 2024-12-27:
> ### Describe your feature request
>
> As per the EMMA feature request, this 'issue' is intended as a placeholder to allow the gathering of information and discussion re the forthcoming SmartGuard-63A- [...]

Hi there. 

I just downloaded the latest MODBUS interface definiton and communication matrix for the EMMA 2.0 with Smartguard, also known in the Fusionsolar app as SmartHEMS. Maybe theese files can help. If needed i can help with testing as i have a system installed at my house. I never coded anything for HA so i'm a bit lost on that side.

[SmartHEMS V100R024C00SPC100 MODBUS Interface Definitions.pdf](https://github.com/user-attachments/files/18259554/SmartHEMS.V100R024C00SPC100.MODBUS.Interface.Definitions.pdf)
[SmartHEMS V100R024C00SPC100 Communication Matrix.xlsx](https://github.com/user-attachments/files/18259555/SmartHEMS.V100R024C00SPC100.Communication.Matrix.xlsx)

Best Regards,

Roving-Ronin, 2024-12-27:
@menyhartda This refers to EMMA and firmware version V100R024C00SPC100, please have a read through the above thread, as I believe this is the version of firmware (and V100R024C00SPC102) that people have working already, based upon WLCRS Integration v1.5.1

vschild, 2025-02-10:
Hi! Anyone managed to get it working with the smartguard? When i try to install de integration i get this error:

Logger: custom_components.huawei_solar.config_flow
Source: custom_components/huawei_solar/config_flow.py:508
integration: Huawei Solar ([documentation](https://github.com/wlcrs/huawei_solar/wiki), [issues](https://github.com/wlcrs/huawei_solar/issues))
First occurred: 5:33:42 PM (1 occurrences)
Last logged: 5:33:42 PM

Unexpected exception while connecting via ModbusTCP
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/asyncio/tasks.py", line 507, in wait_for
    return await fut
           ^^^^^^^^^
  File "/usr/local/lib/python3.13/asyncio/locks.py", line 213, in wait
    await fut
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/config/custom_components/huawei_solar/config_flow.py", line 508, in async_step_setup_network
    info = await validate_network_setup_auto_slave_discovery(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<3 lines>...
    )
    ^
  File "/config/custom_components/huawei_solar/config_flow.py", line 100, in validate_network_setup_auto_slave_discovery
    bridge = await create_tcp_bridge(
             ^^^^^^^^^^^^^^^^^^^^^^^^
    ...<3 lines>...
    )
    ^
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/bridge.py", line 659, in create_tcp_bridge
    return await _create(await AsyncHuaweiSolar.create(host, port, slave_id), slave_id)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/bridge.py", line 685, in _create
    product_info = await HuaweiSolarProductInfo.retrieve_from_device(client, slave_id)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/bridge.py", line 67, in retrieve_from_device
    ) = await client.get_multiple(
        ^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<8 lines>...
    )
    ^
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/huawei_solar.py", line 292, in get_multiple
    response = await self._read_registers(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<3 lines>...
    )
    ^
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/huawei_solar.py", line 422, in _read_registers
    async with self._communication_lock():
               ~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/local/lib/python3.13/contextlib.py", line 214, in __aenter__
    return await anext(self.gen)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/huawei_solar.py", line 141, in _communication_lock
    await asyncio.wait_for(
    ...<2 lines>...
    )
  File "/usr/local/lib/python3.13/asyncio/tasks.py", line 506, in wait_for
    async with timeouts.timeout(timeout):
               ~~~~~~~~~~~~~~~~^^^^^^^^^
  File "/usr/local/lib/python3.13/asyncio/timeouts.py", line 116, in __aexit__
    raise TimeoutError from exc_val
TimeoutError

Roving-Ronin, 2025-02-11:
Initial support for EMMA has been added with the release of v1.5.x, however SmartGuard is a different product to EMMA, for which I believe support has NOT been added to WLCRS integration.
I believe that we are still pending finding the Modbus register details for SmartGuard, that would be required to allow adding support for SmartGuard.

As SmartGuard is the new version of the Backup Box, your install overall should be able to work with WLCRS, if connecting to the inverters via the sDongle / EMMA, albeit without the status / controls of the SmartGuard indicating being operating On/Off Grid etc.

menyhartda, 2025-02-11:
I saw you other response the problem is due to the fact that i also have
the Smartguard with EMMA inside of it, so i tought it will work as the
EMMA, but apparently not.

Thanks.

Roving-Ronin, 2025-02-12:
The SmartGuard has an EMMA 'bundled' into it, but from a wiring perspective you should be connecting to the IP address / Modbus port of the EMMA. IMHO I would think it should work, just that WLCRS would not currently be able to 'understand' or know which Modbus registers to pot serially query, to talk with the BackupBox component of SmartGuard, so it could now if it's on grid power / grid offline etc..?

menyhartda, 2025-02-12:
> I can confirm that it doesnt work (i expected the same as you). It doesnt
> connect.

i did manage to connect via the IP address of the Smartguard. But the data
coming back (probably as only theese are programmed within the Modbus
registers) is only from the MAP0 inverter and the Batteries. Nothing in the
EMMA and it smartmeter and nothing on the SMARTGUARD. However the fact the
the system is on-gid or off grid can be seen in the Inverter registers.

vschild, 2025-02-12:
> i did manage to connect via the IP address of the Smartguard. But the data
> coming back (probably as only theese are programmed within the Modbus
> registers) is only from the MAP0 inverter and the Batteries. Nothing in the
> EMMA and it smartmeter and nothing on the SMARTGUARD. However the fact the
> the system is on-gid or off grid can be seen in the Inverter registers.
> […](#)
> 
>  I can confirm that it doesnt work (i expected the same as you). It doesnt
>  connect.
> 
>  <[#499 (comment)](https://github.com/wlcrs/huawei_solar/issues/499#issuecomment-2653460212)>,
> 
> 
> Daniel, Menyhart.

can you share how did you connect to the Emma? I followed the instructions of the wiki and continue getting the error of the log

Soelli5, 2025-03-29:
@menyhartda I'd also be interested in how you managed to at least get those data out of the Smart Guard EMMA.

I've had also the SUN2000-8K-MAP0 installed earlier (including the Smart Meter and SDongle) and Huawei_solar was working perfect via the SDongle ModbusTCP as long as the internet connection was established. 
Here already a great thank you to all who brought the integration that far already!

Unfortunately (from the  integration perspective) I got a SmartGuard-63A-T0 installed (which does perfectly what it's intended for) and I also struggled by not getting any connection with HA (the inverter ModbusID moved in my case to 12 now), but modifying that did not help so far.

What I also recognized in comparison to the initial configuration with the MAP0 was that the EMMA and Inverter are communicating on 115200 BaudRate compared to the earlier 9600. I'm not sure if that may be a related reason already.

I have for both devices installer access and would also happily help in testing in case you need something tested (e.g. few registers to start with on the EMMA there).
Current (latest according to Huawei App) FW Versions are in my case:
EMMA SmartHEMS V100R024C00SPC100
Backup Box V100R024C00SPC101
Inverter V200R024C00SPC102

Best, Stefan

ichimmariusadrian, 2025-04-17:
Any chance on scheduling a solution for this?
I am getting the same error as vschild when trying to connect to the EMMA of the Smartguard.
EMMA SmartHEMS V100R024C00SPC100
Backup Box V100R024C00SPC101
Huawei Solar: 1.5.4

`Logger: custom_components.huawei_solar.config_flow
Source: custom_components/huawei_solar/config_flow.py:508
integration: Huawei Solar ([documentation](https://github.com/wlcrs/huawei_solar/wiki), [issues](https://github.com/wlcrs/huawei_solar/issues))
First occurred: 9:20:37 PM (1 occurrences)
Last logged: 9:20:37 PM

Unexpected exception while connecting via ModbusTCP
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/asyncio/tasks.py", line 507, in wait_for
    return await fut
           ^^^^^^^^^
  File "/usr/local/lib/python3.13/asyncio/locks.py", line 213, in wait
    await fut
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/config/custom_components/huawei_solar/config_flow.py", line 508, in async_step_setup_network
    info = await validate_network_setup_auto_slave_discovery(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<3 lines>...
    )
    ^
  File "/config/custom_components/huawei_solar/config_flow.py", line 100, in validate_network_setup_auto_slave_discovery
    bridge = await create_tcp_bridge(
             ^^^^^^^^^^^^^^^^^^^^^^^^
    ...<3 lines>...
    )
    ^
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/bridge.py", line 659, in create_tcp_bridge
    return await _create(await AsyncHuaweiSolar.create(host, port, slave_id), slave_id)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/bridge.py", line 685, in _create
    product_info = await HuaweiSolarProductInfo.retrieve_from_device(client, slave_id)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/bridge.py", line 67, in retrieve_from_device
    ) = await client.get_multiple(
        ^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<8 lines>...
    )
    ^
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/huawei_solar.py", line 292, in get_multiple
    response = await self._read_registers(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<3 lines>...
    )
    ^
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/huawei_solar.py", line 422, in _read_registers
    async with self._communication_lock():
               ~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/local/lib/python3.13/contextlib.py", line 214, in __aenter__
    return await anext(self.gen)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/huawei_solar/huawei_solar.py", line 141, in _communication_lock
    await asyncio.wait_for(
    ...<2 lines>...
    )
  File "/usr/local/lib/python3.13/asyncio/tasks.py", line 506, in wait_for
    async with timeouts.timeout(timeout):
               ~~~~~~~~~~~~~~~~^^^^^^^^^
  File "/usr/local/lib/python3.13/asyncio/timeouts.py", line 116, in __aexit__
    raise TimeoutError from exc_val
TimeoutError`

Soelli5, 2025-04-17:
Hi ichimmariusadrian,
I also struggled a lot but learned by trial and error that you have to enter the exact IP address of your modbus client. In my case the 0.0.0.0 does not work anymore. Second in my case I had the EMMA on port 0 and inverter at port 3 (found by AUTO) even if the inverter was set to 12. I had issues for some time but by coincidence then restarted SmartGuard due to other issues and suddenly HA could connect and was working fine. Maybe this helps for you too. There can only be one connection to the one set IP address.

Besides that, it seems to me that SmartGuard takes over the control of the battery management. After connecting EMMA in my case the inverter/batteries controls (input fields) do not receive values anymore and remain empty. 
But till now I had unfortunately no time to check this manually with huawei_solar library (which was my first trial to connect after many errors in HA) if I can still read values from those specific registers.

c0delama, 2025-06-10:
I'd like to +1 this issue. My system is composed of the inverter SUN2000-12K-MAP0, battery LUNA2000-21-S1 and SmartGuard with EMMA-A02. I'm connected via TCP to EMMA. The integration works in general and i can see quite some values, however i cannot set any value, which is quite a bummer. Also the optimisers don't show up at all and i also don't see a few not so important values.

Liipong, 2025-07-01:
Hello. I have a SUN2000 m1 inverter and a LUNA2000 7 kW battery. And before buying SMART BOX, I successfully controlled my energy system from Home assistant. But after adding Smart box, my energy system turned to stone. Because it became impossible to control it from Home assistant. I was able to connect to EMMA via a local network. I can see most of the sensors, but I cannot change the setting values. Because all the input fields are now inactive and it is impossible to enter data in them. I experimented with different settings and tried different versions of EMMA firmware, batteries and inverter, it does not change anything. Therefore, buying EMMA was a big disappointment for me. After reading everything that is written in this thread, I understand that there is no quick solution. Or is there another way to control the Huawei energy system from Home assistant?

tobiasburkart, 2025-07-03:
Any news on this issue?
It is not possible for me to connect to EMMA/Smartguard via HomeAssistant.

Modusbus is active and the IP from Home Assistant ist entered, also TLS deactivated.

Home Assistant show the error "No inverter was found".

What I am doing wrong?

carlosdominique, 2025-07-05:
> Before yesterday I also struggled to connect to EMMA (smartguard). I tried a few times but it doesn’t work.
> 
> Then I put Modbus off and reconnect the network connection in EMMA interface (fusion app, lokal connect)… turned Modbus on and I directly had a connection in the next try with home assistant.
> 
> The most sensor what I need are available and I am happy.

I think I’m getting a similar issue. The integration returns an “unexpected error” message when I try to configure the connection with the Emma (installed inside a Smartguard).
Can you detail step by step how to put off and reconnect turning off and off Modbus with Fusion Solar?
I have installer permissions.
Thanks in advance!

bogd, 2025-07-07:
> The integration returns an “unexpected error” message when I try to configure the connection with the Emma (installed inside a Smartguard). 

Sounds exactly like my problem - I got the same "unexpected error" from the integration when it could not connect to port 502 on the EMMA. And the HAss logs showed a "connect call failed error:
```
Logger: pymodbus.logging
Source: custom_components/huawei_solar/config_flow.py:100
integration: huawei_solar ([documentation](https://github.com/wlcrs/huawei_solar/wiki), [issues](https://github.com/wlcrs/huawei_solar/issues))
First occurred: July 5, 2025 at 11:51:44 PM (1 occurrence)
Last logged: July 5, 2025 at 11:51:44 PM

Failed to connect [Errno 111] Connect call failed ('192.168.1.101', 502)
```

@tobiasburkart - can you please clarify the steps you took to get EMMA to respond to connection requests?

larrybml, 2025-08-12:
Inverter : SUN5000-12K-MAP0 - latest firmware
SmarGuard-63a-t0 - latest firmware
EMMA-A02 - latest firmware

I can confirm that connecting to SmartGuard-63a-to with EMMA PRO is working. After playing with settings and network cable, like it was exemplified before in this thread , I managed to connect. The problem is that a lot of inverter data are missing (inverter temperature, individual MPPT status, voltage, power, etc) also optimizers are not visible . Is there any way to connect directly to inverter via EMMA network ?
I leave here an screenshot of my EMMA MODBUS communications settings.

[image]

Soelli5, 2025-08-12:
To me this looks good, if your client (e.g. HA instance is connecting from IP 192.168.17.27. I also struggled with the inverter initially. 

Two learnings I had (I think that was solved with FW update above similar) was that for the initial connection you either have to upload a new FW (=> Reboot of Emma) or Power cycle it (Inverter off and Grid off => everything off; Helped in my case for initial connect). 

Once you are connected to EMMA in HA, EMMA is typically on address 0, and in my case the Inverter was found at 3 (even it is set to 11 or 12 in the Huawei settings). But that was revealed with "Configure new" or in German "Neu Konfigurieren" directly in the Huawei integration. Just make sure you enter AUTO then and let the integration search. Otherwise you may try 0,3 or 0,4 etc. 

What doesn't work for me is e.g. setting the charging power for the batteries as it was with the smart dongle. That works with the WLCRS in python directly but not in HA anymore. Here I installed as workaround now the Modbus integration and am writing the registers directly to e.g. limit charging in the morning and keeping the feed into the grid during peak power under the network provider limits to charge the battery.
You can experiment first with Developer Tools -> Actions -> Write Modbus register -> and then make a script from what you tried.

An example script from my configuration to set charging to 500W in the Inverter (in my case Address 3; and Battery is controlled by Inverter):

```
action: modbus.write_register
metadata: {}
data:
  hub: modbus_hub
  address: 47075 # Modbus register you want to write
  slave: 3   # My inverter ID
  value:
    - 0        # First UINT16 of the UINT32 if I understood correct
    - 500    # Second UINT16 to indicate the 500W => I use it up to 3500W and the second line is sufficient
```

Reading works as well, if the battery configuration can not be accessed anymore.
In case somebody has found a solution for the native usage of the Battery config; I'm happy to hear as well ;)

Criaga, 2025-08-12:
Hi everyone! 
I'm about to switch from Dongle to SmartGuard (single-phase). I just read that the battery checks are gone. 
Will our @wlcrs  work some miracles in this regard? 
Thanks.

Criaga, 2025-08-17:
Hi everyone. 
Yesterday I installed SmartGuard 63A-S0. I had the dongle before. I updated to the latest firmware version, and on HA I have all the entities I had before. I also enabled elevated permissions and can monitor battery charge/discharge... I can even see the optimizers.

I read that many people have problems with entities…maybe I was lucky 😅

[image]![Uploading 478770676...]
[image]
[image]
[image]

larrybml, 2025-08-20:
Can you please tell us what firmware has your : SmartGuard, EMMA, Inverter … and what version of HA and Huawei Solar do you have?
I tried updating everything and it didn’t work.

Carletto86, 2025-08-20:
> Can you please tell us what firmware has your : SmartGuard, EMMA, Inverter … and what version of HA and Huawei Solar do you have? I tried updating everything and it didn’t work. I had to downgrade Iverter firmware so I can read it again. And still about oprimizers, I can see only the number of them, but I can not see each one statistics.

I also didn't see the optimizers before, then I updated the Huawei Solar component and it showed them to me

Carletto86, 2025-08-20:
> > Hi, I also see the optimizers, but I noticed that they update with a 2-hour delay. Does this happen to you too?
> > Also, did you install both the dongle and Emma at the same time?
> 
> Italiano?
> 
> I used to have the Dongle but it was removed because it was not compatible with Emma.

Si, sono italiano.

Out of curiosity, did you use the default passwords for your Huawei devices during setup or did you change them?

larrybml, 2025-08-20:
I've set an installer password when installed solar system.

bogd, 2025-08-20:
> What version you have on inverter? it is SUN5000 ..MAP0 also? You can see optimizers individually ?

V200R024C00SPC102 . It is a SUN2000-10K-MAP0.

I am connecting to the EMMA IP using Modbus-over-TCP, and reading everything from there.

I do not know about the optimizers - I have only one ( :) ), and I did not even try to see if it is visible in HAss.

bogd, 2025-09-13:
> It's connected via my wifi network (set by my installer), but i don't have any 502 or 6607 port open, and i can't find any "modbus tcp" option (with regular or installer account).

Here is how I did it:
- In the Huawei FusionSolar app, go to Services (at the bottom) / Device Commissioning
- Scan the QR code on the backup box/EMMA (not the one on the inverter!), or connect to the HEMS WiFi network directly, using the SSID/password on the label
- Select "installer account", and enter your installer password
- Go to Set / Communication Settings / Modbus TCP
- Perform the needed configuration (check "connection", uncheck "TLS", specify client IP.

skrew, 2025-09-13:
Thanks @bogd . Unfortunately, my installer created an installer account for me, but I don't have the password that was apparently entered directly into the devices. I tried some generic passwords for Huawei found online, but none of them worked. I've written to him and I'm waiting for Monday.

skrew, 2025-09-14:
@bogd Big thanks! Reset was the solution. I have finally enabled modbus-tcp and integrated it to HASS ! 😅