wlcrs/huawei_solar #505: [Feature Request]: Calling service to a specific register.

home409ca, 2023-10-12:
### Describe your feature request

Solarman integration has this solution.
I could then call a specific register and set a parameter that is not included by default in your integration.

[image]

### Proper usage

- [X] I confirm that this is not a bug report or support request
- [X] I confirm that this feature request is within the stated scope of the integration

Roving-Ronin, 2023-10-13:
@home409ca 
Just wondering as to an example of where and what register you'd want to change this way?
Adding this takes the integration from something that is relatively KISS (keep it simple stupid) and adds the ability for someone without the knowledge (or with, but enters a typo) to potentially screw up or toast their Huawei setup.

home409ca, 2023-10-13:
for expaple "overvoltage protection" value

wlcrs, 2023-10-13:
It would be quite hard to correctly implement this: 
- some settings require you to write to more than 1 register at once;
- there is a mix of signed and unsigned registers.

If a power user wants to achieve this functionality, I suggest that he/she uses the [Modbus proxy addon](https://github.com/Akulatraxas/ha-modbusproxy) to multiplex the connection to the inverter and uses the [Modbus integration](https://www.home-assistant.io/integrations/modbus) to correctly define and change the registers that he/she wants to update. This forces the user to properly think about all the subtilties of correctly reading and setting a Modbus register.

closed 2023-10-13 by wlcrs