wlcrs/huawei_solar #507: [Bug]: Wong values from integration

spac3h, 2023-10-16:
### Describe the issue

Hi,

The integration is reporting values totally different from the Fusion Solar app. I am connecting to modbus via inverter AP, and i am also using the fusion solar app trough AP.

[image]

[image]

Also my inverter is a SUN2000-5KTL-L1 and the integration reports that i have installed a SUN2000-2KTL-L1

Any idea on what can i misconfigured or be missing?

### Bescribe your Huawei Solar Setup

Inverter Type:
Inverter Firmware version:
SDongle present: yes / no
Power meter present: three phase / single phase / no
Battery: LG RESU xxkWh / LUNA2000 xxkWh / None
Battery Firmware version:

### How do you connect to the inverter?

Via the `SUN2000-<serial_no>` WiFi

### Upload your Diagnostics File

[config_entry-huawei_solar-94e4b0aa9768003bbc27e7d151ef7830.json.txt](https://github.com/wlcrs/huawei_solar/files/12915861/config_entry-huawei_solar-94e4b0aa9768003bbc27e7d151ef7830.json.txt)

### Upload your relevant debug logs

```text
2023-10-16 11:36:46.052 INFO (MainThread) [backoff] Backing off _do_read(...) for 1.0s (TimeoutError)
2023-10-16 11:36:47.055 ERROR (MainThread) [huawei_solar.huawei_solar] Modbus client is not connected to the inverter.
NoneType: None
2023-10-16 11:36:47.056 INFO (MainThread) [backoff] Backing off _do_read(...) for 2.0s (huawei_solar.exceptions.ConnectionInterruptedException: Modbus client is not connected to the inverter.)
2023-10-16 11:36:48.095 ERROR (MainThread) [custom_components.huawei_solar] Timeout fetching HV21A0255655_data_update_coordinator data
2023-10-16 11:37:23.094 ERROR (MainThread) [huawei_solar.huawei_solar] Timeout while waiting for connection. Reconnecting...
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/asyncio/tasks.py", line 490, in wait_for
    return fut.result()
           ^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/asyncio/locks.py", line 213, in wait
    await fut
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 141, in _communication_lock
    await asyncio.wait_for(self._client.connected_event.wait(), WAIT_FOR_CONNECTION_TIMEOUT)
  File "/usr/local/lib/python3.11/asyncio/tasks.py", line 492, in wait_for
    raise exceptions.TimeoutError() from exc
TimeoutError
2023-10-16 11:40:08.903 INFO (MainThread) [backoff] Backing off _do_read(...) for 1.0s (TimeoutError)
2023-10-16 11:40:09.905 ERROR (MainThread) [huawei_solar.huawei_solar] Modbus client is not connected to the inverter.
NoneType: None
2023-10-16 11:40:09.905 INFO (MainThread) [backoff] Backing off _do_read(...) for 2.0s (huawei_solar.exceptions.ConnectionInterruptedException: Modbus client is not connected to the inverter.)
2023-10-16 11:40:11.907 ERROR (MainThread) [huawei_solar.huawei_solar] Modbus client is not connected to the inverter.
NoneType: None
2023-10-16 11:40:11.912 INFO (MainThread) [backoff] Backing off _do_read(...) for 4.0s (huawei_solar.exceptions.ConnectionInterruptedException: Modbus client is not connected to the inverter.)
2023-10-16 11:40:15.914 ERROR (MainThread) [huawei_solar.huawei_solar] Modbus client is not connected to the inverter.
NoneType: None
2023-10-16 11:40:15.914 INFO (MainThread) [backoff] Backing off _do_read(...) for 8.0s (huawei_solar.exceptions.ConnectionInterruptedException: Modbus client is not connected to the inverter.)
2023-10-16 11:40:17.096 ERROR (MainThread) [custom_components.huawei_solar] Timeout fetching HV21A0255655_data_update_coordinator data
2023-10-16 11:41:32.418 INFO (MainThread) [backoff] Backing off _do_read(...) for 1.0s (TimeoutError)
2023-10-16 11:41:33.420 ERROR (MainThread) [huawei_solar.huawei_solar] Modbus client is not connected to the inverter.
NoneType: None
[... 115 lines cut ...]
2023-10-16 11:46:03.500 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x7f\xff', b'\xff\xff']
2023-10-16 11:46:03.500 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x7f\xff', b'\xff\xff']
2023-10-16 11:46:03.500 DEBUG (MainThread) [pymodbus.logging] handle: [b'\xff\xff', b'\xff\x17']
2023-10-16 11:46:03.500 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x7f\xff', b'\xff\xff']
2023-10-16 11:46:03.501 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x7f\xff', b'\xff\xff']
2023-10-16 11:46:03.501 DEBUG (MainThread) [pymodbus.logging] handle: [b'\xff\xff', b'\xfeC']
2023-10-16 11:46:03.501 DEBUG (MainThread) [pymodbus.logging] handle: [b'\xff\xff', b'\xfe\xbd']
2023-10-16 11:46:03.501 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x03', b'\x07R']
2023-10-16 11:46:03.501 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x06', b'\x97P']
2023-10-16 11:46:03.501 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b'\x00\x00']
2023-10-16 11:46:03.501 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x7f\xff', b'\xff\xff']
2023-10-16 11:46:03.501 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x7f\xff', b'\xff\xff']
2023-10-16 11:46:03.501 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x7f\xff', b'\xff\xff']
2023-10-16 11:46:03.501 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x7f\xff', b'\xff\xff']
2023-10-16 11:46:03.501 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x7f\xff', b'\xff\xff']
2023-10-16 11:46:03.501 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x7f\xff', b'\xff\xff']
2023-10-16 11:46:03.501 DEBUG (MainThread) [custom_components.huawei_solar] Finished fetching HV21A0255655_data_update_coordinator data in 3.409 seconds (success: True)
```

### Please confirm the following:

- [X] The problem is still present in the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

wlcrs, 2023-10-16:
I've never seen an inverter misreporting its model number before. Please verify that you are not mixing up two unrelated inverters. The SN of the inverter of which you've included the logs is HV21A0255655 . Please verify that SN in your FusionSolar app and on the sticker on your inverter.

spac3h, 2023-10-16:
Hi, the error was on me, a neighbor has an inverter which i can connect to with same credentials and i was connecting to his inverter 🤦 . Thanks for your help.

closed 2023-10-16 by spac3h