wlcrs/huawei_solar #512: [Bug]: Integration continuously failing after SW update

9-ck, 2023-10-21:
### Describe the issue

I've had the SW of my SUN2000 10KTL-M1, SUN2000 3KTL-L1 and the Dongle BT22… updated a couple of days ago. I've experienced issues with the integration since. It fails to initialize / setup. I've noticed that it is "lagging" during restart.

I thought that it had something to do with my configuration.yaml file since when I deleted newly added parts and rebooted HA the integration would setup. But working in the configuration.yaml (without the parts that I previously thought was the cause), the integration would not initialize after a simple restart.

Prior to this instance I've not had a single issue with either HA nor the integration.

I will upload the diagnostics file when / if i get the integration up and running. I could reinstal the integration if that would make any sence?

[image]

### Bescribe your Huawei Solar Setup

Inverter Type: SUN2000 10KTL-M1 / SUN2000 3KTL-L1 
Inverter Firmware version: V100R001C00SPC156 / V200R001C00SPC132
SDongle present: yes (Firmware version: V200R001C00SPC132)
Power meter present: three phase
Battery: LUNA2000 10 kWh
Battery Firmware version: V100R002C00SPC119

### How do you connect to the inverter?

Please select your connection method

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.

### Upload your relevant debug logs

```text
2023-10-21 11:50:07.669 WARNING (SyncWorker_2) [homeassistant.loader] We found a custom integration watchman which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-10-21 11:50:07.670 WARNING (SyncWorker_2) [homeassistant.loader] We found a custom integration huawei_solar which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-10-21 11:50:07.671 WARNING (SyncWorker_2) [homeassistant.loader] We found a custom integration nordpool which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-10-21 11:50:07.672 WARNING (SyncWorker_2) [homeassistant.loader] We found a custom integration energidataservice which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-10-21 11:50:07.672 WARNING (SyncWorker_2) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-10-21 11:50:07.673 WARNING (SyncWorker_2) [homeassistant.loader] We found a custom integration solcast_solar which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-10-21 11:50:18.647 ERROR (MainThread) [homeassistant.components.sensor] Platform template does not generate unique IDs. ID house_load_yield_saving already exists - ignoring sensor.house_load_yield_saving
2023-10-21 11:50:57.032 ERROR (MainThread) [huawei_solar.huawei_solar] Aborting client creation due to error.
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 178, in create
    await huawei_solar._initialize()
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 107, in _initialize
    self.time_zone = (await self.get(rn.TIME_ZONE)).value
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 241, in get
    return (await self.get_multiple([name], slave))[0]
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 276, in get_multiple
    response = await self._read_registers(registers[0].register, total_length, slave)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 398, in _read_registers
    return await _do_read()
           ^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/backoff/_async.py", line 151, in retry
    ret = await target(*args, **kwargs)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/backoff/_async.py", line 151, in retry
    ret = await target(*args, **kwargs)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 351, in _do_read
    response = await self._client.read_holding_registers(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/client/base.py", line 210, in async_execute
    raise ModbusIOException(
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] ERROR: No response received after 3 retries
2023-10-21 11:50:57.045 WARNING (MainThread) [homeassistant.config_entries] Config entry 'SUN2000-10KTL-M1    EESOLAR-10' for huawei_solar integration not ready yet; Retrying in background
2023-10-21 11:50:57.080 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.energy_battery_charge_capacity (<class 'homeassistant.components.template.sensor.SensorTemplate'>) is using native unit of measurement 'kWh' which is not a valid unit for the device class ('power') it is using; expected one of ['kW', 'W']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+template%22
2023-10-21 11:50:57.080 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.energy_battery_discharge_capacity (<class 'homeassistant.components.template.sensor.SensorTemplate'>) is using native unit of measurement 'kWh' which is not a valid unit for the device class ('power') it is using; expected one of ['kW', 'W']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+template%22
[... 628 lines cut ...]
2023-10-21 12:11:25.456 INFO (MainThread) [backoff] Backing off _do_read(...) for 16.0s (huawei_solar.exceptions.ConnectionInterruptedException: Modbus client is not connected to the inverter.)
2023-10-21 12:11:41.457 ERROR (MainThread) [huawei_solar.huawei_solar] Modbus client is not connected to the inverter.
NoneType: None
2023-10-21 12:11:41.458 ERROR (MainThread) [backoff] Giving up _do_read(...) after 6 tries (huawei_solar.exceptions.ConnectionInterruptedException: Modbus client is not connected to the inverter.)
2023-10-21 12:11:41.459 ERROR (MainThread) [huawei_solar.huawei_solar] Aborting client creation due to error.
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/backoff/_async.py", line 151, in retry
    ret = await target(*args, **kwargs)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/backoff/_async.py", line 151, in retry
    ret = await target(*args, **kwargs)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 349, in _do_read
    raise ConnectionInterruptedException(message)
huawei_solar.exceptions.ConnectionInterruptedException: Modbus client is not connected to the inverter.

During handling of the above exception, another exception occurred:
```

### Please confirm the following:

- [X] The problem is still present in the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [ ] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

9-ck, 2023-10-21:
Here is the Full Log
[config_entry-huawei_solar-1a6c1dee2434020362aa55e51f2e97c8.json.txt](https://github.com/wlcrs/huawei_solar/files/13060546/config_entry-huawei_solar-1a6c1dee2434020362aa55e51f2e97c8.json.txt)

9-ck, 2023-10-21:
Here is the Huawei Solar debug log
[home-assistant_huawei_solar_2023-10-21T09-43-36.482Z.log](https://github.com/wlcrs/huawei_solar/files/13060548/home-assistant_huawei_solar_2023-10-21T09-43-36.482Z.log)

SirInsky12, 2023-10-21:
I have exactly the same problem

Wintherman, 2023-10-21:
I have the same problem.
But where can I find config.yaml?
I have searched in the huawie_solar directory but cannot find.

9-ck, 2023-10-21:
Thanks @SirInsky12! I'm sorry, but I'm not sure which config.yaml you're referring too. Is it th configuration.yaml? I have no modbus entries here. Or is it maybe the config_flow.py file inside the Huawei_solar folder? That’s some file to go through.

The funny thing is that the integration has been stable since my post. I re-edited some sensors with what I believe is "older" code (not to solve the issue, just to clean up). Deleted the integration, rebooted, installed Huawei Solar again. Also moved all my templates to a templates.yaml file.

Here is a small piece of the code I copied…
          {% set price = float(states('sensor.energi_data_service'),0) %}
          {% set meter = float(states('sensor.energy_battery_charge_grid'),0) %}
          {% set meter = float(states('sensor.energy_battery_charge_grid'),0) %}

I changed it to this…
          {% set price = states('sensor.energi_data_service')|float(0) %}
          {% set meter = states('sensor.energy_battery_charge_grid')|float(0) %}
          {% set delta = meter - this.attributes.last|float(0) %}

I am no wiz, more like a novice, so I'm not sure it´s "old" code and I'm just telling you what happened. Hard for me to say if it has any relevance at all.

9-ck, 2023-10-22:
OK - the Huawei Solar integration crashed this morning while I was editing my templated.yaml file. Attached is the debug log.

[home-assistant_huawei_solar_2023-10-22T07-54-30.767Z.log](https://github.com/wlcrs/huawei_solar/files/13062674/home-assistant_huawei_solar_2023-10-22T07-54-30.767Z.log)

9-ck, 2023-10-22:
Reading the debug log, trying to understand and correct / clean. Rebooted and installed the integration again. I now see that the naming of the inverter input sensors are back to what they were before I had issues…? When I first installed the integration the name was "sensor.inverter_input_power_1" - then it changed to "sensor.inverter_1_input_power_1" and now it's back to "sensor.inverter_input_power_1".

Wintherman, 2023-10-22:
Hi,
I am up and running again after Delete and installed to the latest version 1.2.7 I think and reboot.
I could run my little script forcing the battery to charge from the grid up to 70% with "huawei_solar.forcible_charge_soc"
But still it would be nice to know where config.yaml is placed.

SirInsky12, 2023-10-22:
Please check your configuration.yaml and edit your Modbus entries, is not allowed to use count for that.

9-ck, 2023-10-22:
> Please check your configuration.yaml in Homeassistant and edit your Modbus entries, is not allowed to use count for that.
Thanks @SirInsky12 but I don't have any modbus entries in my configuration.yaml file? If you check the lates debug log I uploaded in the thread you'll notice that (at least to my eyes) the Huawei Solar integration has some issues with bad formatting in my templates - I use DKK and kWh/kWp. I did clean this up, but just had another crash.

9-ck, 2023-10-22:
Hmm... The CPU load on VM for HA is around 85-90% constantly and 5% user on top of that. That seems a bit to squeezed - no?
I'm running HA-core (latest ver. 2023.10.5) in an Oracle VM VirtualBox Manager on an Intel NUCi5 with 16 GB ram and 8 processors. The VM for HA has 4GB ram and 2 processors allocated. The only other thin running on the NUC is PLEX, also in a VM (4 GB ram and 4 processors allocated to PLEX).

My HA iso image is approximately 30 GB. Would you consider the CPU load normal?

Admittedly I experience the latest crash while working in HA, playing with views several views.

SheikhMishu, 2023-10-22:
Same issue, I have added some logs if that helps. 

[image]

[huawei_solar.huawei_solar_1.log](https://github.com/wlcrs/huawei_solar/files/13064606/huawei_solar.huawei_solar_1.log)
[huawei_solar.huawei_solar_2.log](https://github.com/wlcrs/huawei_solar/files/13064607/huawei_solar.huawei_solar_2.log)

SirInsky12, 2023-10-23:
Go to developer tools and check the config and look what happens

9-ck, 2023-10-23:
> Same issue, I have added some logs if that helps.
> 
> [image]
> 
> [huawei_solar.huawei_solar_1.log](https://github.com/wlcrs/huawei_solar/files/13064606/huawei_solar.huawei_solar_1.log) [huawei_solar.huawei_solar_2.log](https://github.com/wlcrs/huawei_solar/files/13064607/huawei_solar.huawei_solar_2.log)

As said I'm no wiz, but it seems you have time out errorsAs said I'm no wiz, but it seems you have time out errors

9-ck, 2023-10-23:
@SirInsky12 Appreciate your trying to help but where do you mean? In Developer Tools I have YAML / STATES / SERVICES / TEMPLATE/ EVENTS / STATISTICS and ASSIST.

In the STATISTICS I have a bunch of issues with regards to "units" some that I have created myself, some which come from other integrations. Seams strange that HA wouldn't know kWh, m and °C - also why doesn't HA know the Danish currency DKK? Im unsure what wil happen if I pres "Fix Issue".

[image]

BTW, in my VM setup I have increased the numbers of processors to 4 and ram to 6GB. But still I am experiencing dropouts now and then. HA will run fine for a while and suddenly drop out.

9-ck, 2023-10-25:
Could someone please have a look at this. Is it OK with the "errors"?
[image]

matt7py, 2023-10-25:
The same problem was working normally, but suddenly, it would not accept connections. I attempted to resolve it by deleting the integration and reinstalling it. Additionally, I tried reconnecting via WiFi, but the issue persists. !!

[image]

9-ck, 2023-10-26:
This issue continues. It seams like frequent restarts in connection with working in the configuration file will provoke a crash in my case.