wlcrs/huawei_solar #515: [Feature Request]:  Off Grid Switch

MattiaFontanaErrepinet, 2023-10-27:
### Describe your feature request

Hi, it is possible to add a switch that activates and deactivates the Off Grid mode

### Proper usage

- [X] I confirm that this is not a bug report or support request
- [X] I confirm that this feature request is within the stated scope of the integration

wlcrs, 2023-11-20:
I don't understand the official documentation for this feature. It only documents what at '0' value means, but nothing else:

[image]

Can you please test and report back how that register should be used? I don't have a backup box, so I cannot test.

ObviousInvestor, 2023-12-11:
I would like to request this feature too. I should have the backup box installed next week, but I'm not a developer. If you can tell me what you need and how to get it, I'll provide any information you need.

WiebKastanje, 2023-12-11:
Have a Backup Box recently installed. 
Not tested yet.

The integration shows conviently below info in HA, but I do not know if this is register 47604.
If any testing is needed, let me know what and how to do that.

[image]

ObviousInvestor, 2023-12-11:
There is also a switch in the app under "Parameters" for the inverter for enabling/disabling the feature.
[image]

ObviousInvestor, 2024-01-06:
Hello,

I finally have my backup box installed so if you need me to provide any information to make this feature a reality, please let me know.

It would also be helpful to be able to access "Inverter Device Status" somehow as it's available to display but not to access in the automation section.

Thanks for this wonderful integration!

wlcrs, 2024-01-07:
I've reverse engineered the app to check if I could see what it writes to register 47604. However, I didn't find any evidence of the app writing to that register.

I don't feel confident to add a feature I do not understand. You can reach out to Huawei support to ask how that Modbus register should be used. You can include the screenshot I've shared earlier in this thread to help them understand what you are asking.

The most likely value that needs to be written is `1` instead of `0` to switch between on and off-grid. If you want to try totally untested code, **for which I take absolutely no responsibility for and which can damage your installation**, then you can try adding the following entity to the `ENERGY_STORAGE_SWITCH_DESCRIPTIONS` list in [switch.py](https://github.com/wlcrs/huawei_solar/blob/main/switch.py#L52C7-L52C7):

```python
   HuaweiSolarSwitchEntityDescription(
        key=rn.BACKUP_SWITCH_TO_OFF_GRID,
        name="Switch to Off-Grid",
        entity_category=EntityCategory.CONFIG,
    ),

```

Before toggling the switch in HA, please try to activate the off-grid function from the FusionSolar app, restart your HA (as a change to the value can take up to 5 minutes to propagate otherwise) and report back the state of the switch before and after going off-grid.

If this switch behaves as expected when toggled externally, you can try toggling it from HA to check if it works as expected. If you didn't see it's status change when switching between modes from the FusionSolar app, then I would strongly advise against doing so.

Please report back your findings.

ObviousInvestor, 2024-01-07:
Thanks for the feedback and the help.

I'm starting to think this is a non starter. The Switch I showed in the previous post does not take the inverter off grid, it just enables off grid mode, but it doesn't actually take it off grid. Now I have the box installed, it's enabled all of the time. To put the inverter in to off grid mode, the grid power has to be cut to the box it appears. So even if we control the software switch, it will only put the SUN2000 in to a state that it "can" go off grid, but it won't actually take it off grid until the power goes out (hope that makes sense?).
I think the easiest way now to automatically put the inverter into off grid mode is to install a wifi breaker on the feed before the box. That way I can cut the power whenever I need to. The breaker would obviously need to be powered from the other side of the box.

wlcrs, 2024-01-07:
Ok, in that case I'm closing this issue. Thank you for your insights!

closed 2024-01-07 by wlcrs