wlcrs/huawei_solar #520: Add entity and service translations

wlcrs, 2023-10-29:
Hi @vpont, @herostrat, @alexdelli, @xtala, @mkornat, @LeandroIssa,

I'm mentioning you because you've already done the translation of the Huawei Solar integration in the past. I've updated the integration to make the entities and services translatable. The number of lines in the translation json-file has grown from 51 to 658, so I hope you are up for the challenge to translate them 🥵

All new strings to be translated start in the [`en.json` file starting from line 51](https://github.com/wlcrs/huawei_solar/blob/main/translations/en.json#L51).

There are quite a few duplicates like 'voltage' and 'current' to be found, so a few "replace all" operations in your favorite text editor should already fix quite a few translations.

alexdelli, 2023-10-29:
Yeah very pleasure to help with translation!

mkornat, 2023-10-29:
I'm willing to help in my spare time too

vpont, 2023-10-29:
Here you go! https://github.com/wlcrs/huawei_solar/pull/522

alexdelli, 2023-10-31:
The italian job  [#526](https://github.com/wlcrs/huawei_solar/pull/526)

xtala, 2023-11-02:
Hi,

I've started working on this as well. Is it possible that some fields are missing? There don't seem to be fields for pv_11_voltage and pv_21_voltage.

Regards,
Evert

wlcrs, 2023-11-02:
These are indeed missing. Well spotted! Feel free to add them yourself, I'll make sure that it is also fixed in the other translations.

mkornat, 2023-11-21:
Here you go #545

alexdelli, 2023-12-18:
mmmm my sensors are changed. All with italian name, but non so clearly:
could any other check?
thank you
[image]