wlcrs/huawei_solar #530: [Bug]:

ragesoft, 2023-11-02:
### Describe the issue

After Updating to HA 2023.11.0 i get error that huyawei solar could not been loaded...
See logs below

Core 2023.11.0
Supervisor 2023.10.1
Operating System 11.1
Frontend 20231030.1

 ERROR (MainThread) [homeassistant.loader] Unexpected exception importing component custom_components.huawei_solar

ERROR (MainThread) [homeassistant.setup] Setup failed for custom integration huawei_solar: Unable to import component

`2023-11-02 23:04:39.769 ERROR (MainThread) [homeassistant.loader] Unexpected exception importing component custom_components.huawei_solar
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/loader.py", line 816, in get_component
ComponentProtocol, importlib.import_module(self.pkg_path)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.11/importlib/__init__.py", line 126, in import_module
return _bootstrap._gcd_import(name[level:], package, level)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
File "<frozen importlib._bootstrap_external>", line 940, in exec_module
File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
File "/config/custom_components/huawei_solar/__init__.py", line 23, in <module>
from huawei_solar import (
File "/usr/local/lib/python3.11/site-packages/huawei_solar/__init__.py", line 6, in <module>
from .bridge import HuaweiSolarBridge # noqa
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.11/site-packages/huawei_solar/bridge.py", line 16, in <module>
from .huawei_solar import DEFAULT_BAUDRATE, DEFAULT_SLAVE, DEFAULT_TCP_PORT, AsyncHuaweiSolar, Result
File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 17, in <module>
from pymodbus.payload import BinaryPayloadBuilder, BinaryPayloadDecoder
File "/usr/local/lib/python3.11/site-packages/pymodbus/payload.py", line 28, in <module>
class BinaryPayloadBuilder:
File "/usr/local/lib/python3.11/site-packages/pymodbus/payload.py", line 42, in BinaryPayloadBuilder
self, payload=None, byteorder=Endian.LITTLE, wordorder=Endian.BIG, repack=False
^^^^^^^^^^^^^
AttributeError: type object 'Endian' has no attribute 'LITTLE'
2023-11-02 23:04:39.798 ERROR (MainThread) [homeassistant.setup] Setup failed for custom integration huawei_solar: Unable to import component: Exception importing custom_components.huawei_solar
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/loader.py", line 816, in get_component
ComponentProtocol, importlib.import_module(self.pkg_path)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.11/importlib/__init__.py", line 126, in import_module
return _bootstrap._gcd_import(name[level:], package, level)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
File "<frozen importlib._bootstrap_external>", line 940, in exec_module
File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
File "/config/custom_components/huawei_solar/__init__.py", line 23, in <module>
from huawei_solar import (
File "/usr/local/lib/python3.11/site-packages/huawei_solar/__init__.py", line 6, in <module>
from .bridge import HuaweiSolarBridge # noqa
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.11/site-packages/huawei_solar/bridge.py", line 16, in <module>
from .huawei_solar import DEFAULT_BAUDRATE, DEFAULT_SLAVE, DEFAULT_TCP_PORT, AsyncHuaweiSolar, Result
File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 17, in <module>
from pymodbus.payload import BinaryPayloadBuilder, BinaryPayloadDecoder
File "/usr/local/lib/python3.11/site-packages/pymodbus/payload.py", line 28, in <module>
class BinaryPayloadBuilder:
File "/usr/local/lib/python3.11/site-packages/pymodbus/payload.py", line 42, in BinaryPayloadBuilder
self, payload=None, byteorder=Endian.LITTLE, wordorder=Endian.BIG, repack=False
[... 79 lines cut ...]
from .huawei_solar import DEFAULT_BAUDRATE, DEFAULT_SLAVE, DEFAULT_TCP_PORT, AsyncHuaweiSolar, Result
File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 17, in <module>
from pymodbus.payload import BinaryPayloadBuilder, BinaryPayloadDecoder
File "/usr/local/lib/python3.11/site-packages/pymodbus/payload.py", line 28, in <module>
class BinaryPayloadBuilder:
File "/usr/local/lib/python3.11/site-packages/pymodbus/payload.py", line 42, in BinaryPayloadBuilder
self, payload=None, byteorder=Endian.LITTLE, wordorder=Endian.BIG, repack=False
^^^^^^^^^^^^^
AttributeError: type object 'Endian' has no attribute 'LITTLE'
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/setup.py", line 215, in _async_setup_component
component = integration.get_component()
^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/src/homeassistant/homeassistant/loader.py", line 824, in get_component
raise ImportError(f"Exception importing {self.pkg_path}") from err
ImportError: Exception importing custom_components.huawei_solar
```

### Please confirm the following:

- [X] The problem is still present in the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [ ] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [ ] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

wlcrs, 2023-11-03:
Please update this integration to the latest version

closed 2023-11-03 by wlcrs

ragesoft, 2023-11-03:
> Please update this integration to the latest version

I have.
1.2.8
Still get this error... So why close?

wlcrs, 2023-11-03:
Because if you don't do the effort to search for existing issues, I don't do effort to write a big explanation

cfr. https://github.com/wlcrs/huawei_solar/issues/470#issuecomment-1727258019