wlcrs/huawei_solar #531: [Feature Request]: Change Capacity Control max average peak Wattage

driesvandamme, 2023-11-02:
### Describe your feature request

Would it be possible to change the maximum average peak wattage in Capacity Control mode from Home Assistant?

Currently it's only possible using the FusionSolar app. - Could be useful to automate setting it higher during winter months when the average consumption is higher (and the "normal" peak isn't sufficient)

### Proper usage

- [X] I confirm that this is not a bug report or support request
- [X] I confirm that this feature request is within the stated scope of the integration

wlcrs, 2023-11-03:
The only information I have available to implement the peak shaving is the following:

[image]

I already have reverse-engineered the 'Peak power' register 47956, and this contains the following: 14 possible periods with the following settings:

- start time
- end time
- peak value
- days of the week where the period is active

If you can provide me with the necessary info on how I can set this parameter via some Modbus register, I will gladly implement it. But I'm not aware of any way how to do it. 

Can you provide screenshots of where you do it in the app? Maybe I can also reverse engineer it from there? (but that will be on a dreary winter day with nothing else on my to-do-list)

driesvandamme, 2023-11-03:
Hi Thijs,

Unfortunately I don't have any clue about the modbus register (I'm not that experienced with Modbus myself)

I am able to provide a screenshot though:
[image]

_How to get there:_
Inverter -> Connected Devices
- Battery
- - 4 dots in the top right: Parameter Settings
- - - Tab: Peak Shaving

wlcrs, 2023-11-03:
Everything you describe is already available? cfr. https://github.com/wlcrs/huawei_solar/wiki/Capacity-control

closed 2023-11-03 by wlcrs