wlcrs/huawei_solar #532: Update Frequency ignored if set too low

SvenAbels, 2023-11-03:
### Describe the issue

Hi,

I'm trying to update the "Active power" sensor from the Power Meter more often as I'd like tomuse it in a near real-time dashboard. It's a DTSU666-H, which is connected to the Sun2000 inverter and I query it using the SDongle with a wired LAN connection. I have setup the automation script of the "Inverter polling frequency" for the "Active Power" sensor and it works well when setting it to 15 seconds.

But I noticed that it only works is I set the frequency to around 15 seconds or more. If I use "seconds: "/5" or even "seconds: /1", then the value only gets updated with the automatic refresh every 30 seconds.

Is there a minimum limit or 15 seconds or am I doing something wrong?
I'm not sure if this is a bug or simply the normal behaviour but I didn't find anything in the discussions so far.

### Bescribe your Huawei Solar Setup

Inverter Type: SUN2000-10KTL-M1
SDongle present: yes
Power meter present: yes, three phase 
Battery: LUNA2000 15kWh

### How do you connect to the inverter?

Via the SDongle, wired connection

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.

### Upload your relevant debug logs

```text
.
```

### Please confirm the following:

- [X] The problem is still present in the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [ ] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [ ] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

SvenAbels, 2023-11-03:
I believe that the following discussion covers it already:
https://github.com/wlcrs/huawei_solar/discussions/434

My understanding from the post in this discussion is that every update reading already takes around 12 seconds which is why a lower frequency is not working. May it would be a good idea to update the example on the readme file to set it to e.g. 15 instead of 10 seconds and to mention the minimum frequency there?

What I don't understand yet: Is the time caused because the WLCRS extension always reads all MODBUS registers and would it in principle be possible to read the Active Power entry more often? Or is this a general problem with the MODBUS protocol only allowing to read all entries?

wlcrs, 2023-11-03:
Due to the way Home Assistant implements batched sensor updates with the `DataUpdateCoordinator`, your request to update the "Active Power" sensor triggers an update of all registers that this integration offers (excluding the configuration entities and optimizer entities). This is a HA limitation as there is no way to know which register the user requested to be updated.

I would assume that fetching that one register will be faster than updating all registers. However, it is not possible to implement this in this integration without removing the batched sensor updates with the `DataUpdateCoordinator`. This would certainly deal a big performance hit to the integration in its whole. (This is even the reason I started developing this integration in the first place, as the original integration developed by Emilv2 did not do the batching of requests)

I would suggest putting a Modbus Proxy between the inverter and this integration, and then using the [built-in modbus integration](https://www.home-assistant.io/integrations/modbus/) to fetch that one modbus register which you want more frequently. This is the relevant modbus register (I think):

[image]

SvenAbels, 2023-11-03:
Thanks  a lot. This helps me a lot to understand the issue.

I'll play a bit with the modbus proxy :)

Best regards,

Sven

closed 2023-11-03 by SvenAbels