wlcrs/huawei_solar #537: Unable to set battery working mode through the REST API

stefaanv, 2023-11-08:
### Describe the issue

Changing the `battery_working_mode` configuration over the REST API doesn't work.
The API does reflect the new state back, but the working mode hasn't changed in reality.
After th api call, the drop-down list in the lovelace interface for the `battery working mode` configuration is empty.

I'm using these calls :
POST to `http://homeassistant.local:8123/api/states/select.battery_working_mode`
body
```json
{
  "state": "Maximise Self Consumption"
}
```
returns
```json
{
  "entity_id": "select.battery_working_mode",
  "state": "Maximise Self Consumption",
  "attributes": {},
  "last_changed": "2023-11-08T17:02:42.497145+00:00",
  "last_updated": "2023-11-08T17:02:42.497145+00:00",
  "context": {
    "id": "01...9P",
    "parent_id": null,
    "user_id": "dca...32"
  }
}
```

GET to `http://homeassistant.local:8123/api/states/select.battery_working_mode`
returns
```json
{
  "entity_id": "select.battery_working_mode",
  "state": "Maximise Self Consumption",
  "attributes": {},
  "last_changed": "2023-11-08T17:02:42.497145+00:00",
  "last_updated": "2023-11-08T17:02:42.497145+00:00",
  "context": {
    "id": "01...T9P",
    "parent_id": null,
    "user_id": "dca...b32"
  }
}
```

### Bescribe your Huawei Solar Setup

Inverter Type: SUN2000-4.6KTL-L1
Inverter Firmware version:V200R001C00SPC130
SDongle present: yes / no
Power meter present:  single phase 
Battery: LUNA2000 10kWh 

### How do you connect to the inverter?

Please select your connection method

### Upload your Diagnostics File

{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.11.1",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.11.6",
    "docker": true,
[... 415 lines cut ...]
      ]
    }
  }
}

### Upload your relevant debug logs

```text
Logger: custom_components.huawei_solar
Source: helpers/update_coordinator.py:322
Integration: Huawei Solar (documentation, issues)
First occurred: 7 november 2023 om 22:11:37 (23 occurrences)
Last logged: 17:32:51

Error fetching HV20C0137463_data_update_coordinator data: Could not update HV20C0137463 values: Got error while reading from register 32064 with length 52: Exception Response(131, 3, GatewayNoResponse)
Error fetching HV20C0137463_configuration_update_coordinator data: Could not update HV20C0137463 values: Got error while reading from register 47075 with length 14: Exception Response(131, 3, GatewayNoResponse)
Error fetching HV20C0137463_data_update_coordinator data: Could not update HV20C0137463 values: Got error while reading from register 37760 with length 28: Exception Response(131, 3, GatewayNoResponse)
```

### Please confirm the following:

- [X] The problem is still present in the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

wlcrs, 2023-11-20:
Please read the INFO-box in the documentation of HA:

[image]

closed 2023-11-20 by wlcrs

stefaanv, 2023-11-20:
Then how can I set the Battery WorkingMode ?
Which is the correct URL/post data to change it?  It is  a state, so it should be possible to change.
Changing it in Developer Tools changes the state but that change isn't written to the inverter according to the Huawei app.