wlcrs/huawei_solar #538: [Feature Request]: enable/stop grid feed (for those with dynamic tarrifs)

broodro0ster, 2023-11-10:
### Describe your feature request

Would it be possible to enable/disable grid feed in from the integration?
I'm looking to switch to a dynamic tarrif with hourly prices. When feeding to the grid, you get paid the actual spot price but when it's sunny and windy at the same time, we often have negative prices. That means you pay to send electricity to the grid instead of getting paid.
I would be awesome if we could stop feeding electricity to the grid when prices are negative and enable it again when the prices are positive again. Huawei doesn't support dynamic tarrifs yet as far as I know, but with Home Assistant and the Nordpool addon it should be quite easy to achieve this if we could switch the inverter mode.

Thanks in advance

### Proper usage

- [X] I confirm that this is not a bug report or support request
- [X] I confirm that this feature request is within the stated scope of the integration

jires, 2023-11-10:
Is already [implemented](https://github.com/wlcrs/huawei_solar/wiki/Changing-Active-Power-Control). You need to create an automation, to call this action, preferred to your needs.

broodro0ster, 2023-11-10:
Oh wow, it see it now. It's even on the homepage of the repo. I was looking in the wrong place...
Thanks a lot!

closed 2023-11-10 by broodro0ster