wlcrs/huawei_solar #544: [Feature Request]: negative Max. grid feed-in power (kW): -0.020 kW

MayBankCH, 2023-11-21:
### Describe your feature request

[Feature Request]: negative Max. grid feed-in power (kW): -0.020 kW
API error when input negative Max. grid feed-in power.

[image]
[image]
Max. grid feed-in power >=0 is ok.

### Proper usage

- [X] I confirm that this is not a bug report or support request
- [X] I confirm that this feature request is within the stated scope of the integration

MayBankCH, 2023-12-21:
I create another power check script for negative Max. grid feed-in power. 
[image]
[image]

Roving-Ronin, 2023-12-26:
Also wondering why the maximum to feed into the grid is 5kW, if this is implemented at the sDongle+Consumption meter level (as opposed to the limit at the individual inverter level), this stops users with multiple/cascaded inverters from being able to choose a combined limit that is greater than a single inverter.

i.e. 2 x 6kW inverters, capable of 13.2kW Active Power between them, and being able to limit export to grid to say 10kW.

wlcrs, 2023-12-26:
The minimum value that you can enter is already set to -1000. The integration does not enforce a maximum.

Cfr. https://github.com/wlcrs/huawei_solar/blob/bbf441d3ec508fcad0c3434925b69c33855508b0/services.py#L118

MayBankCH, 2023-12-27:
Noted with thanks. BTW -1000 is -1000 W or -1000 kW

MayBankCH, 2023-12-27:
The minimun value is set to -1000. 
However, power call "_validate_power_value" function 
then check assert Power>=0 and 0 <=Power <= Max

[image]

[image]

[image]

commit 644dd0ed (2023-12-29): Don't validate the minimum power value twice

Fixes #544
closed 2023-12-29 by wlcrs