wlcrs/huawei_solar #546: [Bug]: Checkbox for elevate permissions ticked but no question for password

firebowl, 2023-11-24:
### Describe the issue

I never got asked for the admin password. I uninstalled, restarted HASS an reinstalled the integration but never got asked for the password. I am using a TCP connection. Everything is working fine with the integration, except the features that need the installer account and password.

### Bescribe your Huawei Solar Setup

Inverter Type: SUN2000-5KTL-M1GIN300-5KTL-M1
Inverter Firmware version: V100R001C00SPC153
SDongle present: yes
SDongle Firmware version: V100R001C00SPC133
Power meter present: three phase / single phase / no
Battery: LUNA2000 10kWh
Battery Firmware version: V100R002C00SPC119

### How do you connect to the inverter?

Via the SDongle, wireless connection

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.
[config_entry-huawei_solar-ea0a96c289924ded6c267b9cc54f5191.json (1).txt](https://github.com/wlcrs/huawei_solar/files/13455523/config_entry-huawei_solar-ea0a96c289924ded6c267b9cc54f5191.json.1.txt)

### Upload your relevant debug logs

```text
2023-11-23 06:40:42.789 WARNING (SyncWorker_3) [homeassistant.loader] We found a custom integration iphonedetect which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-11-23 06:40:42.790 WARNING (SyncWorker_3) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-11-23 06:40:42.790 WARNING (SyncWorker_3) [homeassistant.loader] We found a custom integration meross_lan which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-11-23 06:40:42.791 WARNING (SyncWorker_3) [homeassistant.loader] We found a custom integration fusion_solar which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-11-23 06:40:42.792 WARNING (SyncWorker_3) [homeassistant.loader] We found a custom integration huawei_solar which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-11-23 06:40:42.792 WARNING (SyncWorker_3) [homeassistant.loader] We found a custom integration ecoflow_cloud which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-11-23 06:40:42.793 WARNING (SyncWorker_3) [homeassistant.loader] We found a custom integration toyota which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-11-23 06:41:34.556 WARNING (MainThread) [homeassistant.components.sensor] Setup of sensor platform huawei_solar is taking over 10 seconds.
2023-11-23 06:41:36.680 ERROR (MainThread) [homeassistant.helpers.event] Error while processing template: Template<template=({{ (states('sensor.inverter_total_yield') | float) - (states('sensor.battery_total_discharge') | float) + (states('sensor.battery_total_charge') | float) }}) renders=2>
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 1925, in forgiving_float_filter
    return float(value)
           ^^^^^^^^^^^^
ValueError: could not convert string to float: 'unknown'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 568, in async_render
    render_result = _render_with_context(self.template, compiled, **kwargs)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 2231, in _render_with_context
    return template.render(**kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/jinja2/environment.py", line 1301, in render
    self.environment.handle_exception()
  File "/usr/local/lib/python3.11/site-packages/jinja2/environment.py", line 936, in handle_exception
    raise rewrite_traceback_stack(source=source)
  File "<template>", line 1, in top-level template code
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 1928, in forgiving_float_filter
    raise_no_default("float", value)
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 1626, in raise_no_default
    raise ValueError(
ValueError: Template error: float got invalid input 'unknown' when rendering template '{{ (states('sensor.inverter_total_yield') | float) - (states('sensor.battery_total_discharge') | float) + (states('sensor.battery_total_charge') | float) }}' but no default was specified

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 694, in async_render_to_info
    render_info._result = self.async_render(
                          ^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 570, in async_render
    raise TemplateError(err) from err
homeassistant.exceptions.TemplateError: ValueError: Template error: float got invalid input 'unknown' when rendering template '{{ (states('sensor.inverter_total_yield') | float) - (states('sensor.battery_total_discharge') | float) + (states('sensor.battery_total_charge') | float) }}' but no default was specified
[... 627 lines cut ...]
2023-11-24 06:54:22.695 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b'\x0f\xbf']
2023-11-24 06:54:22.695 DEBUG (MainThread) [pymodbus.logging] handle: [b'\xff\xff', b'\xff\x82']
2023-11-24 06:54:22.695 DEBUG (MainThread) [pymodbus.logging] handle: [b'\xff\xff', b'\xff\x9e']
2023-11-24 06:54:22.695 DEBUG (MainThread) [pymodbus.logging] handle: [b'\xff\xff', b'\xffP']
2023-11-24 06:54:22.746 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37760 with length 28 from slave 1
2023-11-24 06:54:22.746 DEBUG (MainThread) [pymodbus.logging] send: 0x37 0x99 0x0 0x0 0x0 0x6 0x1 0x3 0x93 0x80 0x0 0x1c
2023-11-24 06:54:22.747 DEBUG (MainThread) [pymodbus.logging] Adding transaction 14233
2023-11-24 06:54:23.400 DEBUG (MainThread) [pymodbus.logging] recv: 0x37 0x99 0x0 0x0 0x0 0x3b 0x1 0x3 0x38 0x0 0x32 0x0 0x0 0x0 0x4 0xff 0xff 0x7f 0xff 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x3 0x82 0x62 0x0 0x3 0x6a 0xf2 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 old_data:  addr=None
2023-11-24 06:54:23.400 DEBUG (MainThread) [pymodbus.logging] Processing: 0x37 0x99 0x0 0x0 0x0 0x3b 0x1 0x3 0x38 0x0 0x32 0x0 0x0 0x0 0x4 0xff 0xff 0x7f 0xff 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x3 0x82 0x62 0x0 0x3 0x6a 0xf2 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0
2023-11-24 06:54:23.401 DEBUG (MainThread) [pymodbus.logging] Factory Response[ReadHoldingRegistersResponse': 3]
2023-11-24 06:54:23.401 DEBUG (MainThread) [pymodbus.logging] Getting transaction 14233
2023-11-24 06:54:23.401 DEBUG (MainThread) [pymodbus.logging] [50, 0, 4, 65535, 32767, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 3, 33378, 3, 27378, 0, 0, 0, 0]
2023-11-24 06:54:23.401 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b'\x00\x00']
2023-11-24 06:54:23.401 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x03', b'\x82b']
2023-11-24 06:54:23.401 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x03', b'j\xf2']
2023-11-24 06:54:23.401 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b'\x00\x00']
2023-11-24 06:54:23.401 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b'\x00\x00']
```

### Please confirm the following:

- [X] The problem is still present in the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

firebowl, 2023-11-30:
Help? Nobody an idea? 🥲

Kugelfang666, 2023-12-02:
I don’t remember having to input the PW. I’m also connecting via TCP to the dongle with elevated permissions and ist working fine with me

firebowl, 2023-12-03:
Can you change settings like min SOC?
If I change them in HASS, nothing happens in the fusion app. 
Charge / discharge mode is greyed out.

Kugelfang666, 2023-12-03:
yep I can both min SoC and AC charging are adjustable from within HA for me

firebowl, 2024-01-10:
Does no one really have any ideas?
I have now uninstalled the integration again, restarted HASS and reinstalled and set it up again.
I just don't get a question about the password.
Can I enter the password manually in a config?
I have found an entry for user and password in constant.py, but is it enough to adjust the entry there?

[image]

To me, this reads like default values and not like the current ones.
I also have to select slave 1 instead of 0 during the installation (network) in order to be able to address my inverter. In the const.py the default value is still 0, so I assume that these are only default values and not the current values.

jires, 2024-01-14:
If you want to change parameters, you have to do it inside config entry here: /homeassistant/.storage/core.config_entries

When I installed mine, long time ago, I've ticked elevated permission also, and never been asked for user and password, but all functions working as expected. 
My config looks like this: 
```
      {
        "entry_id": "cfc256bc12e768c9e9f3379cb7f0f2f3",
        "version": 1,
        "domain": "huawei_solar",
        "title": "SUN2000-10KTL-M1\u000001074300-002",
        "data": {
          "host": "192.168.[REDACTED]",
          "port": 502,
          "slave_ids": [
            2
          ],
          "enable_parameter_configuration": true,
          "username": null,
          "password": null
        },
        "options": {},
        "pref_disable_new_entities": false,
        "pref_disable_polling": false,
        "source": "user",
        "unique_id": "HV[REDACTED]",
        "disabled_by": null
      },
```

firebowl, 2024-01-14:
For example, if I activate the "Charge from grid" option, HA shows it as activated but nothing happens. This means that the battery is not charged from the grid. When I look on the Fusion Solar website, the option has not been activated at all. I think this is because no password has been requested and therefore access is read-only. Same behavior when I adjust the minimum SOC, for example. The new value is displayed in the HA but not on the website and the battery charge level also remains unchanged.

jires, 2024-01-14:
One more thing. You will be asked for password for elevate permissions only, if you have changed default password (00000a).
If you connecting via Sdongle, and not directly to inverter wifi, password is still default, even if is changed via device commissioning of inverter. Sdongle has different access control and not sharing password with inverter. So, if you want to check, which password is used, you can log in to sdongle directly (unplug sdongle, wait a while and plug in, when led becomes green, go to device commissioning on the app and scan qr code on sdongle). If default password will be accepted, you don't need to type password to elevate permissions in integration setup.

firebowl, 2024-01-14:
Ok, this is important information. I have changed the default password of the inverter but I connect via LAN using the SDongle. Then it should work without a password. But why are the changes I make in the HA not applied?

pkeotr, 2024-05-09:
Any solution with this case? I have the exact same problem and with hours of searching I cannot find a way to get the elevate permissions to work. On router side I obviously see the Dongle IP address and it seems that I can connect to it with slave ID 1. I can tick mark the elevate permissions box but now configuration services are created or password request popped up. And yes, I have changed the original password for inverter. I get the integration to work with and only configuration I can have is inverter on/off. How to get the others work?

closed 2024-06-21 by github-actions[bot]

RenanAz, 2025-10-03:
> If you want to change parameters, you have to do it inside config entry here: /homeassistant/.storage/core.config_entries
>
> When I installed mine, long time ago, I've ticked elevated permission also, [...]

Hardcoding these worked to gain Elevated Permissions! Thanks for the tip, i was banging my head.