wlcrs/huawei_solar #550: [Bug]: Modbus client is not connected to the inverter

oglok, 2023-12-01:
### Describe the issue

With latest Home Assistant version extension is not connecting to the inverter:

Home Assistant 2023.10.3
Frontend 20231005.0 - latest

### Bescribe your Huawei Solar Setup

Inverter Type:
Inverter Firmware version: 
SDongle present: no
Power meter present: single phase 

### How do you connect to the inverter?

Via the `SUN2000-<serial_no>` WiFi

### Upload your Diagnostics File

I cannot get the integration done from scratch. I don't have the integration configured to get logs from any device.
However, my home assistant has the 192.168.200.100 address and the inverter .1 is reachable.

### Upload your relevant debug logs

```text
2023-12-01 18:32:40.249 ERROR (MainThread) [huawei_solar.huawei_solar] Aborting client creation due to error.
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/backoff/_async.py", line 151, in retry
    ret = await target(*args, **kwargs)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/backoff/_async.py", line 151, in retry
    ret = await target(*args, **kwargs)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/deps/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 349, in _do_read
    raise ConnectionInterruptedException(message)
huawei_solar.exceptions.ConnectionInterruptedException: Modbus client is not connected to the inverter.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/config/deps/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 178, in create
    await huawei_solar._initialize()
  File "/config/deps/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 107, in _initialize
    self.time_zone = (await self.get(rn.TIME_ZONE)).value
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/deps/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 241, in get
    return (await self.get_multiple([name], slave))[0]
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/deps/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 276, in get_multiple
    response = await self._read_registers(registers[0].register, total_length, slave)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/deps/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 398, in _read_registers
    return await _do_read()
           ^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/backoff/_async.py", line 159, in retry
    await _call_handlers(on_giveup, **details, exception=e)
  File "/usr/local/lib/python3.11/site-packages/backoff/_async.py", line 37, in _call_handlers
    await handler(details)
  File "/usr/local/lib/python3.11/site-packages/backoff/_async.py", line 16, in f
    return coro_or_func(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/deps/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 327, in backoff_giveup
    raise ReadException(f"Failed to read register {register} after {details['tries']} tries")
huawei_solar.exceptions.ReadException: Failed to read register 43006 after 6 tries
```

### Please confirm the following:

- [X] The problem is still present in the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

artur90226, 2023-12-06:
Plus. Me too.

wlcrs, 2023-12-08:
Please do not use the issue tracker for connectivity problems.

closed 2023-12-08 by wlcrs

lakatusch, 2024-02-20:
hi there,
got the same issue since yesterday 1pm. In the middle of the day the log shows :

```
024-02-19 13:37:33.064 ERROR (MainThread) [huawei_solar.huawei_solar] Modbus client is not connected to the inverter.
NoneType: None
2024-02-19 13:37:33.065 INFO (MainThread) [backoff] Backing off _do_read(...) for 1.0s (huawei_solar.exceptions.ConnectionInterruptedException: Modbus client is not connected to the inverter.)
2024-02-19 13:37:33.928 ERROR (MainThread) [tibber.home] Could not find consumption data.
2024-02-19 13:37:34.075 ERROR (MainThread) [huawei_solar.huawei_solar] Modbus client is not connected to the inverter.
NoneType: None
2024-02-19 13:37:34.076 INFO (MainThread) [backoff] Backing off _do_read(...) for 2.0s (huawei_solar.exceptions.ConnectionInterruptedException: Modbus client is not connected to the inverter.)
2024-02-19 13:37:36.077 ERROR (MainThread) [huawei_solar.huawei_solar] Modbus client is not connected to the inverter.
NoneType: None
2024-02-19 13:37:36.077 INFO (MainThread) [backoff] Backing off _do_read(...) for 4.0s (huawei_solar.exceptions.ConnectionInterruptedException: Modbus client is not connected to the inverter.)
2024-02-19 13:37:40.079 ERROR (MainThread) [huawei_solar.huawei_solar] Modbus client is not connected to the inverter.
NoneType: None
2024-02-19 13:37:40.079 INFO (MainThread) [backoff] Backing off _do_read(...) for 8.0s (huawei_solar.exceptions.ConnectionInterruptedException: Modbus client is not connected to the inverter.)
2024-02-19 13:37:48.081 ERROR (MainThread) [huawei_solar.huawei_solar] Modbus client is not connected to the inverter.
NoneType: None
2024-02-19 13:37:48.082 INFO (MainThread) [backoff] Backing off _do_read(...) for 16.0s (huawei_solar.exceptions.ConnectionInterruptedException: Modbus client is not connected to the inverter.)
2024-02-19 13:38:02.170 WARNING (MainThread) [homeassistant.components.device_tracker] Setup of device_tracker platform cupra_we_connect is taking over 10 seconds.
2024-02-19 13:38:04.085 ERROR (MainThread) [huawei_solar.huawei_solar] Modbus client is not connected to the inverter.
NoneType: None
2024-02-19 13:38:04.086 ERROR (MainThread) [backoff] Giving up _do_read(...) after 6 tries (huawei_solar.exceptions.ConnectionInterruptedException: Modbus client is not connected to the inverter.)
2024-02-19 13:38:04.086 ERROR (MainThread) [huawei_solar.huawei_solar] Aborting client creation due to error.
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/backoff/_async.py", line 151, in retry
    ret = await target(*args, **kwargs)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/backoff/_async.py", line 151, in retry
    ret = await target(*args, **kwargs)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 349, in _do_read
    raise ConnectionInterruptedException(message)
huawei_solar.exceptions.ConnectionInterruptedException: Modbus client is not connected to the inverter.
```

How does i fix this issue?

Thanks a lot

regards
Daniel

josejn, 2024-03-04:
Same problem here...
Working like a charm for many weeks until today.
Home Assistant says (huawei_solar.exceptions.ConnectionInterruptedException: Modbus client is not connected to the inverter.)
But the ew11 is working perfectly and publishing data through the the mqtt queue.

lakatusch, 2024-03-04:
wait an hour and than do a SYSTEM reboot of HA... this work for me...

Schrolli91, 2025-02-26:
> Edit:
> Rebooting the ew11 module fixed the problem :/

That's also true for me. But I'm not sure, why this is the case.
It is working perfectly until my HA Instance has a Crash. After the Reboot i can't connect to the Elfin Dongle anymore until I restart it by unplugging it.
Very strange behavior ...