wlcrs/huawei_solar #551: [Bug]: module 'asyncio' has no attribute 'timeout'

Cartoon-Dude, 2023-12-02:
### Describe the issue

After updating to the version 1.2.8, the message "module 'asyncio' has no attribute 'timeout'" appeared.

### Bescribe your Huawei Solar Setup

Inverter Type: SUN2000-8KTL
Inverter Firmware version:
SDongle present: yes
Power meter present: three phase / single phase / no
Battery: LG RESU xxkWh / LUNA2000 xxkWh / None
Battery Firmware version:

### How do you connect to the inverter?

Via the SDongle, wireless connection

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.
[home-assistant_huawei_solar_2023-12-02T08-14-02.400Z.log](https://github.com/wlcrs/huawei_solar/files/13533807/home-assistant_huawei_solar_2023-12-02T08-14-02.400Z.log)

### Upload your relevant debug logs

```text
Réessayer la configuration: module 'asyncio' has no attribute 'timeout'
```

### Please confirm the following:

- [X] The problem is still present in the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

Cartoon-Dude, 2023-12-02:
Downgraded to 1.2.7 and it is working fine again.

wlcrs, 2023-12-08:
Caused by an incompatible Python version.

closed 2023-12-08 by wlcrs