wlcrs/huawei_solar #554: Every sensor shows correct data, but after 30 seconds every sensor change to unavailable

nkk67, 2023-12-10:
### Describe the issue

Connection to my inverter works perfect Modbus TCP Port 502. And after installation or reloading every sensor shows the correct value. But after about 15 to 30 seconds every sensor become "unavailable". After reload all sensors show again the correct values.

### Bescribe your Huawei Solar Setup

Inverter Type: SUN2000-12KTL-M5
Inverter Firmware version: V200R022C10SPC104
SDongle present: yes
Power meter present: no
Battery: None
Battery Firmware version:

### How do you connect to the inverter?

Please select your connection method

### Upload your Diagnostics File

[config_entry-huawei_solar-6db82e9829ae9bc4f775474457781231.json.txt](https://github.com/wlcrs/huawei_solar/files/13628892/config_entry-huawei_solar-6db82e9829ae9bc4f775474457781231.json.txt)

### Upload your relevant debug logs

```text
2023-12-10 13:55:47.172 INFO (MainThread) [custom_components.huawei_solar.switch] Skipping switch setup, as parameter configuration is not enabled
2023-12-10 13:55:47.172 INFO (MainThread) [custom_components.huawei_solar.select] Skipping select setup, as parameter configuration is not enabled
2023-12-10 13:55:47.219 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064 with length 52 from slave 1
2023-12-10 13:55:47.220 DEBUG (MainThread) [pymodbus.logging] send: 0x0 0xb 0x0 0x0 0x0 0x6 0x1 0x3 0x7d 0x40 0x0 0x34
2023-12-10 13:55:47.220 DEBUG (MainThread) [pymodbus.logging] Adding transaction 11
2023-12-10 13:55:47.380 DEBUG (MainThread) [pymodbus.logging] recv: 0x0 0xb 0x0 0x0 0x0 0x6b 0x1 0x3 0x68 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0xe 0x77 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x1 0x24 0xb 0xb8 0xa0 0x0 0x0 0x0 0x65 0x75 0x7f 0x6a 0x65 0x75 0xdf 0x8 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x5 0x8e 0x0 0x0 0x6 0x23 0x65 0x76 0x17 0xbb 0x0 0x0 0x0 0x0 0x0 0x0 0x2 0xbe old_data:  addr=None
2023-12-10 13:55:47.380 DEBUG (MainThread) [pymodbus.logging] Processing: 0x0 0xb 0x0 0x0 0x0 0x6b 0x1 0x3 0x68 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0xe 0x77 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x1 0x24 0xb 0xb8 0xa0 0x0 0x0 0x0 0x65 0x75 0x7f 0x6a 0x65 0x75 0xdf 0x8 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x5 0x8e 0x0 0x0 0x6 0x23 0x65 0x76 0x17 0xbb 0x0 0x0 0x0 0x0 0x0 0x0 0x2 0xbe
2023-12-10 13:55:47.380 DEBUG (MainThread) [pymodbus.logging] Factory Response[ReadHoldingRegistersResponse': 3]
2023-12-10 13:55:47.380 DEBUG (MainThread) [pymodbus.logging] Getting transaction 11
2023-12-10 13:55:47.381 DEBUG (MainThread) [pymodbus.logging] [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 3703, 0, 0, 0, 0, 0, 0, 0, 292, 3000, 40960, 0, 25973, 32618, 25973, 57096, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1422, 0, 1571, 25974, 6075, 0, 0, 0, 702]
2023-12-10 13:55:47.381 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b'\x00\x00']
2023-12-10 13:55:47.381 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b'\x00\x00']
2023-12-10 13:55:47.381 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b'\x00\x00']
2023-12-10 13:55:47.381 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b'\x00\x00']
2023-12-10 13:55:47.382 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b'\x0ew']
2023-12-10 13:55:47.382 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b'\x00\x00']
2023-12-10 13:55:47.382 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b'\x00\x00']
2023-12-10 13:55:47.382 DEBUG (MainThread) [pymodbus.logging] handle: [b'eu', b'\x7fj']
2023-12-10 13:55:47.382 DEBUG (MainThread) [pymodbus.logging] handle: [b'eu', b'\xdf\x08']
2023-12-10 13:55:47.382 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b'\x05\x8e']
2023-12-10 13:55:47.382 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b'\x02\xbe']
2023-12-10 13:55:47.435 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32000 with length 24 from slave 1
2023-12-10 13:55:47.436 DEBUG (MainThread) [pymodbus.logging] send: 0x0 0xc 0x0 0x0 0x0 0x6 0x1 0x3 0x7d 0x0 0x0 0x18
2023-12-10 13:55:47.437 DEBUG (MainThread) [pymodbus.logging] Adding transaction 12
2023-12-10 13:55:47.688 DEBUG (MainThread) [pymodbus.logging] recv: 0x0 0xc 0x0 0x0 0x0 0x33 0x1 0x3 0x30 0x0 0x1 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x8b 0xb8 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 old_data:  addr=None
2023-12-10 13:55:47.688 DEBUG (MainThread) [pymodbus.logging] Processing: 0x0 0xc 0x0 0x0 0x0 0x33 0x1 0x3 0x30 0x0 0x1 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x8b 0xb8 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0
2023-12-10 13:55:47.689 DEBUG (MainThread) [pymodbus.logging] Factory Response[ReadHoldingRegistersResponse': 3]
2023-12-10 13:55:47.689 DEBUG (MainThread) [pymodbus.logging] Getting transaction 12
2023-12-10 13:55:47.690 DEBUG (MainThread) [pymodbus.logging] [1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 35768, 0, 0, 0, 0, 0, 0, 0, 0]
2023-12-10 13:55:47.690 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b'\x00\x00']
2023-12-10 13:55:47.691 DEBUG (MainThread) [custom_components.huawei_solar] Finished fetching HV2320227320_data_update_coordinator data in 0.512 seconds (success: True)
2023-12-10 13:55:57.693 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064 with length 52 from slave 1
2023-12-10 13:55:57.694 DEBUG (MainThread) [pymodbus.logging] send: 0x0 0xd 0x0 0x0 0x0 0x6 0x1 0x3 0x7d 0x40 0x0 0x34
2023-12-10 13:55:57.696 DEBUG (MainThread) [pymodbus.logging] Adding transaction 13
2023-12-10 13:55:57.853 DEBUG (MainThread) [pymodbus.logging] recv: 0x0 0xd 0x0 0x0 0x0 0x6b 0x1 0x3 0x68 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0xe 0x77 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x1 0x24 0xb 0xb8 0xa0 0x0 0x0 0x0 0x65 0x75 0x7f 0x6a 0x65 0x75 0xdf 0x8 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x5 0x8e 0x0 0x0 0x6 0x23 0x65 0x76 0x17 0xc5 0x0 0x0 0x0 0x0 0x0 0x0 0x2 0xbe old_data:  addr=None
2023-12-10 13:55:57.854 DEBUG (MainThread) [pymodbus.logging] Processing: 0x0 0xd 0x0 0x0 0x0 0x6b 0x1 0x3 0x68 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0xe 0x77 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x1 0x24 0xb 0xb8 0xa0 0x0 0x0 0x0 0x65 0x75 0x7f 0x6a 0x65 0x75 0xdf 0x8 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x5 0x8e 0x0 0x0 0x6 0x23 0x65 0x76 0x17 0xc5 0x0 0x0 0x0 0x0 0x0 0x0 0x2 0xbe
2023-12-10 13:55:57.854 DEBUG (MainThread) [pymodbus.logging] Factory Response[ReadHoldingRegistersResponse': 3]
2023-12-10 13:55:57.855 DEBUG (MainThread) [pymodbus.logging] Getting transaction 13
2023-12-10 13:55:57.855 DEBUG (MainThread) [pymodbus.logging] [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 3703, 0, 0, 0, 0, 0, 0, 0, 292, 3000, 40960, 0, 25973, 32618, 25973, 57096, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1422, 0, 1571, 25974, 6085, 0, 0, 0, 702]
2023-12-10 13:55:57.856 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b'\x00\x00']
2023-12-10 13:55:57.856 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b'\x00\x00']
2023-12-10 13:55:57.857 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b'\x00\x00']
2023-12-10 13:55:57.857 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b'\x00\x00']
2023-12-10 13:55:57.857 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b'\x0ew']
2023-12-10 13:55:57.857 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b'\x00\x00']
2023-12-10 13:55:57.858 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b'\x00\x00']
[... 62 lines cut ...]
2023-12-10 13:56:58.132 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32000 with length 24 from slave 1
2023-12-10 13:56:58.133 DEBUG (MainThread) [pymodbus.logging] send: 0x0 0x12 0x0 0x0 0x0 0x6 0x1 0x3 0x7d 0x0 0x0 0x18
[... cut ...]
2023-12-10 13:57:47.915 DEBUG (MainThread) [custom_components.huawei_solar] Finished fetching HV2320227320_data_update_coordinator data in 20.002 seconds (success: False)
```

### Please confirm the following:

- [X] The problem is still present in the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

Andurilll, 2023-12-14:
i have the same now

johnbrecon, 2023-12-14:
I have the same, started 13 Dec, about the same time as upgrade from Huawei?

miguelalonso, 2023-12-15:
Same for me, Unvavailable problems starting one month ago after more than 1 year working OK. I use an Elfin EW11 TCP/RS485 converter.

wlcrs, 2023-12-18:
This issue seems to be caused by an unstable connection to the inverter. This integration already has already implemented a lot of logic to automatically recover from unstable connections. The two most important ones are:
- each Modbus request is retried for up to 6 times, with increasing waits between each attempt to allow the connection to recover;
- after 3 failed Modbus requests the connection is reset to fix hanging connections.

Please understand that I cannot help you solve your unstable connection. Consider [an alternative connection method](https://github.com/wlcrs/huawei_solar/wiki/Connecting-to-the-inverter).
If you are convinced that there is no connectivity issue, then I invite you to debug and fix the problem yourself and submit a pull request to the underlying [huawei-solar Python library](https://gitlab.com/Emilv2/huawei-solar). Your input would be greatly appreciated.

As there is nothing actionable for me to do in this issue, I'm closing it. You are welcome to open a topic in [Discussions: Connecting to the inverter](https://github.com/wlcrs/huawei_solar/discussions/categories/connecting-to-the-inverter) to get community support for your problem.

closed 2023-12-18 by wlcrs

Andurilll, 2023-12-22:
Thanks for the Update and sad to hear you close this thread.

I thank you anyway for the Integration and reconnection logic implemented and hope to solve it somehow with a alternative Solution. Now i am connected with a USB-Wifi Stick to the inverter 

I am new to python and have no idea how to Programm. I try to Start my Master of Advanced Studies in Information technology this Fall and there i will learn python. May you know a good Way to learn it by myself now? (Tutorial?)

nkk67, 2024-01-03:
Hi, after a lot of experimenting with no success, today I restricted the modbus tcp (before it was unrestricted) to the ip of my home assistant and it is working. No more unavailable sind about 15 minutes. I do not know why, but now everything is fine.

stefanhapper, 2024-01-03:
> Hi, after a lot of experimenting with no success, today I restricted the modbus tcp (before it was unrestricted) to the ip of my home assistant and it is working.

At what level and how did you restrict the modbus access?

nkk67, 2024-01-03:
In the configuration of my inverter sun2000-12ktl-M5 there is a menu where I can disable Modbus TCP, Enable unrestricted and enable restricted. When restricted I can add one Ip address (it is only on possible). After this it works for me until know without any problem.

stefanhapper, 2024-01-04:
> In the configuration of my inverter sun2000-12ktl-M5 there is a menu where I can disable Modbus TCP, Enable unrestricted and enable restricted. When restricted I can add one Ip address (it is only one possible). After this it works for me until know without any problem.

Thanks for pointing me to that option. I realised that for me the Modbus access was simply turned off (I guess by Huawei after an automatic firmware update as I have never changed this myself). I turned it simply on and the issue was resolved.

Joao-Sousa-71, 2024-07-16:
> In the configuration of my inverter sun2000-12ktl-M5 there is a menu where I can disable Modbus TCP, Enable unrestricted and enable restricted. When restricted I can add one Ip address (it is only one possible). After this it works for me until know without any problem.

That looks interesting!