wlcrs/huawei_solar #557: [Bug]: Loosing connection to Inverter

Andurilll, 2023-12-14:
### Describe the issue

Had everything working fine.

a few days ago i keep loosing the connection to the Inverter after around 3 hours.
Restart fixes it for short this 3 hours. after this the Inverter, Power meter & Battery are Offline.

[image]

### Bescribe your Huawei Solar Setup

Inverter Type: SUN2000-4KTL-M1
Inverter Firmware version: V100R001C00SPC154
SDongle present: yes (V100R001C00SPC133)
Power meter present: three phase 
Battery: LUNA2000 10kWh 
Battery Firmware version: V100R002C00SPC120/V100R002C00SPC117 (updating now to ....123)

### How do you connect to the inverter?

Via the `SUN2000-<serial_no>` WiFi

### Upload your Diagnostics File

[config_entry-huawei_solar-6a3069b428534f4f59ecdcfa309f00fe.json (1).txt](https://github.com/wlcrs/huawei_solar/files/13677876/config_entry-huawei_solar-6a3069b428534f4f59ecdcfa309f00fe.json.1.txt)

### Upload your relevant debug logs

```text
2023-12-14 20:50:30.595 ERROR (MainThread) [huawei_solar.huawei_solar] Could not read register value, has another device interrupted the connection?
2023-12-14 20:50:30.605 INFO (MainThread) [backoff] Backing off _do_read(...) for 1.0s (huawei_solar.exceptions.ConnectionInterruptedException: Could not read register value, has another device interrupted the connection?)
2023-12-14 20:50:30.606 DEBUG (MainThread) [huawei_solar.huawei_solar] Received <class 'huawei_solar.exceptions.ConnectionInterruptedException'>: backing off reading for 1.0 seconds after 1 tries
2023-12-14 20:50:31.607 ERROR (MainThread) [huawei_solar.huawei_solar] Modbus client is not connected to the inverter.
NoneType: None
2023-12-14 20:50:31.607 INFO (MainThread) [backoff] Backing off _do_read(...) for 2.0s (huawei_solar.exceptions.ConnectionInterruptedException: Modbus client is not connected to the inverter.)
2023-12-14 20:50:31.607 DEBUG (MainThread) [huawei_solar.huawei_solar] Received <class 'huawei_solar.exceptions.ConnectionInterruptedException'>: backing off reading for 2.0 seconds after 2 tries
2023-12-14 20:50:31.862 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.zhimi_meb1_4041_pm10_density_2 (<class 'custom_components.xiaomi_miot.sensor.MiotSensorSubEntity'>) is using native unit of measurement 'None' which is not a valid unit for the device class ('pm10') it is using; expected one of ['µg/m³']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/al-one/hass-xiaomi-miot/issues
2023-12-14 20:50:33.608 ERROR (MainThread) [huawei_solar.huawei_solar] Modbus client is not connected to the inverter.
NoneType: None
2023-12-14 20:50:33.609 INFO (MainThread) [backoff] Backing off _do_read(...) for 4.0s (huawei_solar.exceptions.ConnectionInterruptedException: Modbus client is not connected to the inverter.)
2023-12-14 20:50:33.609 DEBUG (MainThread) [huawei_solar.huawei_solar] Received <class 'huawei_solar.exceptions.ConnectionInterruptedException'>: reconnecting and backing off reading for 4.0 seconds after 3 tries
2023-12-14 20:50:33.609 DEBUG (MainThread) [pymodbus.logging] Connecting to 192.168.200.1:6607.
2023-12-14 20:50:33.609 DEBUG (MainThread) [pymodbus.logging] Connecting comm
2023-12-14 20:50:33.615 DEBUG (MainThread) [pymodbus.logging] Connected to comm
2023-12-14 20:50:33.615 DEBUG (MainThread) [pymodbus.logging] callback_connected called
2023-12-14 20:50:33.617 DEBUG (MainThread) [huawei_solar.modbus] Waiting for 1500 milliseconds after connection before performing operations
2023-12-14 20:50:34.785 ERROR (MainThread) [custom_components.huawei_solar] Timeout fetching BT2270251564_data_update_coordinator data
2023-12-14 20:50:34.792 INFO (MainThread) [huawei_solar.huawei_solar] Waiting for connection ...
2023-12-14 20:50:35.118 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 47075 with length 14 from slave 0
2023-12-14 20:50:35.118 DEBUG (MainThread) [pymodbus.logging] send: 0x0 0x24 0x0 0x0 0x0 0x6 0x0 0x3 0xb7 0xe3 0x0 0xe
2023-12-14 20:50:35.119 DEBUG (MainThread) [pymodbus.logging] Adding transaction 36
2023-12-14 20:50:35.130 DEBUG (MainThread) [pymodbus.logging] recv: 0x0 0x24 0x0 0x0 0x0 0x1f 0x0 0x3 0x1c 0x0 0x0 0x13 0x88 0x0 0x0 0x11 0x30 0x7f 0xff 0xff 0xff 0x3 0x84 0x0 0x64 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x2 0x0 0x0 0x3 0x84 old_data:  addr=None
2023-12-14 20:50:35.131 DEBUG (MainThread) [pymodbus.logging] Processing: 0x0 0x24 0x0 0x0 0x0 0x1f 0x0 0x3 0x1c 0x0 0x0 0x13 0x88 0x0 0x0 0x11 0x30 0x7f 0xff 0xff 0xff 0x3 0x84 0x0 0x64 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x2 0x0 0x0 0x3 0x84
2023-12-14 20:50:35.131 DEBUG (MainThread) [pymodbus.logging] Factory Response[ReadHoldingRegistersResponse': 3]
2023-12-14 20:50:35.131 DEBUG (MainThread) [pymodbus.logging] Getting transaction 36
2023-12-14 20:50:35.131 DEBUG (MainThread) [pymodbus.logging] [0, 5000, 0, 4400, 32767, 65535, 900, 100, 0, 0, 0, 2, 0, 900]
2023-12-14 20:50:35.132 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b'\x13\x88']
2023-12-14 20:50:35.132 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b'\x110']
2023-12-14 20:50:35.161 ERROR (MainThread) [custom_components.huawei_solar] Timeout fetching BT2270251564_configuration_update_coordinator data
```

### Please confirm the following:

- [X] The problem is still present in the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

approx3145, 2023-12-16:
We use a sun2000 8kW and and a Luna 5kWh

when the sun is strong the power meter loosing connection with the Inverter. Always around 12am

Andurilll, 2023-12-17:
> We use a sun2000 8kW and and a Luna 5kWh
> 
> when the sun is strong the power meter loosing connection with the Inverter. Always around 12am

i guess this is a different issue. my setup looses connection to everything. what does your log say?

wlcrs, 2023-12-18:
This issue seems to be caused by an unstable connection to the inverter. This integration already has already implemented a lot of logic to automatically recover from unstable connections. The two most important ones are:
- each Modbus request is retried for up to 6 times, with increasing waits between each attempt to allow the connection to recover;
- after 3 failed Modbus requests the connection is reset to fix hanging connections.

Please understand that I cannot help you solve your unstable connection. Consider [an alternative connection method](https://github.com/wlcrs/huawei_solar/wiki/Connecting-to-the-inverter).
If you are convinced that there is no connectivity issue, then I invite you to debug and fix the problem yourself and submit a pull request to the underlying [huawei-solar Python library](https://gitlab.com/Emilv2/huawei-solar). Your input would be greatly appreciated.

As there is nothing actionable for me to do in this issue, I'm closing it. You are welcome to open a topic in [Discussions: Connecting to the inverter](https://github.com/wlcrs/huawei_solar/discussions/categories/connecting-to-the-inverter) to get community support for your problem.

closed 2023-12-18 by wlcrs

Andurilll, 2024-01-27:
To anybody Setting Here searching For a Solution: after upgrading my inverter, battery & dongle it seems very stable

It could be caused by 2 Different Software versions in the 2 Batteries itself