wlcrs/huawei_solar #558: [Bug]: Error with update from 1.2.8 to 1.2.9 reading via modbus

danise76, 2023-12-18:
### Describe the issue

Justa after updating to 1.2.9 from 1.2.8, the instegration stopeed working.
If we go back to 1.2.8, it works again.

We have 2 inverters coneccted in cascade but modbus, and being readed via modbus with a serial usb adapter

In the log we get with 1.2.9:

```
2023-12-18 18:31:09.068 ERROR (MainThread) [homeassistant.components.sensor] Error while setting up huawei_solar platform for sensor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 361, in _async_setup_platform
    await asyncio.shield(task)
  File "/config/custom_components/huawei_solar/sensor.py", line 735, in async_setup_entry
    entity.add_name_suffix(f" #{idx+1}")
  File "/config/custom_components/huawei_solar/__init__.py", line 461, in add_name_suffix
    self._attr_name = f"{self.name}{suffix}"
                         ^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 477, in name
    return self._name_internal(None, {})
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 438, in _name_internal
    and (name_translation_key := self._name_translation_key)
                                 ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/backports/functools.py", line 70, in __get__
    val = self.func(instance)
          ^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 424, in _name_translation_key
    f"component.{platform.platform_name}.entity.{platform.domain}"
                 ^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'platform_name'
```

### Bescribe your Huawei Solar Setup

Inverter Type: SUN2000-5KTL-L1  and SUN2000-2KTL-L1
SDongle present: no
Power meter present: single phase 

### How do you connect to the inverter?

Please select your connection method

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.

### Upload your relevant debug logs

```text
2023-12-18 18:31:03.076 DEBUG (MainThread) [pymodbus.logging] Connecting to /dev/serial/by-id/usb-1a86_USB2.0-Ser_-if00-port0.
2023-12-18 18:31:03.077 DEBUG (MainThread) [pymodbus.logging] Connecting comm
2023-12-18 18:31:03.081 DEBUG (MainThread) [pymodbus.logging] Connected to comm
2023-12-18 18:31:03.081 DEBUG (MainThread) [pymodbus.logging] callback_connected called
2023-12-18 18:31:03.081 DEBUG (MainThread) [huawei_solar.modbus] Waiting for 1500 milliseconds after connection before performing operations
2023-12-18 18:31:04.084 INFO (MainThread) [huawei_solar.huawei_solar] Waiting for connection ...
2023-12-18 18:31:04.583 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 43006 with length 1 from slave 1
2023-12-18 18:31:04.584 DEBUG (MainThread) [pymodbus.logging] send: 0x1 0x3 0xa7 0xfe 0x0 0x1 0xc6 0x8e
2023-12-18 18:31:04.584 DEBUG (MainThread) [pymodbus.logging] Adding transaction 1
2023-12-18 18:31:04.618 DEBUG (MainThread) [pymodbus.logging] recv: 0x1 0x3 0x2 0x0 0x3c 0xb8 0x55 old_data:  addr=None
2023-12-18 18:31:04.618 DEBUG (MainThread) [pymodbus.logging] Processing: 0x1 0x3 0x2 0x0 0x3c 0xb8 0x55
2023-12-18 18:31:04.619 DEBUG (MainThread) [pymodbus.logging] Getting Frame - 0x3 0x2 0x0 0x3c
2023-12-18 18:31:04.619 DEBUG (MainThread) [pymodbus.logging] Factory Response[ReadHoldingRegistersResponse': 3]
2023-12-18 18:31:04.619 DEBUG (MainThread) [pymodbus.logging] Frame advanced, resetting header!!
2023-12-18 18:31:04.619 DEBUG (MainThread) [pymodbus.logging] Getting transaction 1
2023-12-18 18:31:04.620 DEBUG (MainThread) [pymodbus.logging] [60]
2023-12-18 18:31:04.672 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 47000 with length 1 from slave 1
2023-12-18 18:31:04.672 DEBUG (MainThread) [pymodbus.logging] send: 0x1 0x3 0xb7 0x98 0x0 0x1 0x22 0x51
2023-12-18 18:31:04.673 DEBUG (MainThread) [pymodbus.logging] Adding transaction 1
[... 143 lines cut ...]
2023-12-18 18:31:06.030 DEBUG (MainThread) [pymodbus.logging] Processing: 0x2 0x3 0x2 0x0 0x3 0xbc 0x45
2023-12-18 18:31:06.030 DEBUG (MainThread) [pymodbus.logging] Getting Frame - 0x3 0x2 0x0 0x3
2023-12-18 18:31:06.030 DEBUG (MainThread) [pymodbus.logging] Factory Response[ReadHoldingRegistersResponse': 3]
2023-12-18 18:31:06.031 DEBUG (MainThread) [pymodbus.logging] Frame advanced, resetting header!!
2023-12-18 18:31:06.031 DEBUG (MainThread) [pymodbus.logging] Getting transaction 2
2023-12-18 18:31:06.031 DEBUG (MainThread) [pymodbus.logging] [3]
2023-12-18 18:31:06.084 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 47000 with length 1 from slave 2
2023-12-18 18:31:06.084 DEBUG (MainThread) [pymodbus.logging] send: 0x2 0x3 0xb7 0x98 0x0 0x1 0x22 0x62
2023-12-18 18:31:06.085 DEBUG (MainThread) [pymodbus.logging] Adding transaction 2
2023-12-18 18:31:06.119 DEBUG (MainThread) [pymodbus.logging] recv: 0x2 0x3 0x2 0x0 0x0 0xfc 0x44 old_data:  addr=None
2023-12-18 18:31:06.119 DEBUG (MainThread) [pymodbus.logging] Processing: 0x2 0x3 0x2 0x0 0x0 0xfc 0x44
2023-12-18 18:31:06.119 DEBUG (MainThread) [pymodbus.logging] Getting Frame - 0x3 0x2 0x0 0x0
2023-12-18 18:31:06.119 DEBUG (MainThread) [pymodbus.logging] Factory Response[ReadHoldingRegistersResponse': 3]
2023-12-18 18:31:06.119 DEBUG (MainThread) [pymodbus.logging] Frame advanced, resetting header!!
2023-12-18 18:31:06.120 DEBUG (MainThread) [pymodbus.logging] Getting transaction 2
2023-12-18 18:31:06.120 DEBUG (MainThread) [pymodbus.logging] [0]
2023-12-18 18:31:06.172 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 47089 with length 1 from slave 2
```

### Please confirm the following:

- [X] The problem is still present in the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

opt12, 2023-12-18:
Experienced the same problem.
After upgrading to 1.29 the integration stopped working.
At startup I see these 
`AttributeError: 'NoneType' object has no attribute 'platform_name'`
errors in the logs. 

My setup also contains two inverters, a battery and a powermeter. However, these are read out via Modbus TCP via the SDongle.

Apart from that nuisance right at the moment: Kudos for this great integration.

Cheers.

wlcrs, 2023-12-18:
Hi,

This seems to be caused by an incompatibility between the translation logic
and the code to support multiple inverters.

I don't have a second inverter in my own installation, which explains why I
didn't detect this issue myself.

I'll work on a solution.

baybod, 2023-12-18:
same here. it worked very good before, now it does have the above mentioned issue

r13ssh, 2023-12-19:
same here. worked until I upgraded earlier today - downgrade does not seem to rectify the issue aswell :-( Thanks @wlcrs for all your hard work, it is really appreciated!

scrichab, 2023-12-19:
i have the same problem - how can i downgrade to 1.2.8 ?

wlcrs, 2023-12-19:
Can you please test with 1.3.0 beta 1 ? If this works as expected, I'll do a proper release.

closed 2023-12-19 by wlcrs

r13ssh, 2023-12-19:
Hi @wlcrs 

the integration restarts correctly for me using 1.3.0b1, I do see 1 error in the log (see below), but maybe that was just a interim issue during restart, I _do_ see "Daily Yield #1" and "Daily Yield #2" figures getting updated (obviously with no value changes, give the sun has set already :-))

```
Logger: homeassistant.components.utility_meter.sensor
Source: components/utility_meter/sensor.py:474
Integration: Utility Meter (documentation, issues)
First occurred: 7:17:01 PM (1 occurrences)
Last logged: 7:17:01 PM

PV Inverter Daily yield #1+2 (Utility, daily) received an invalid new state from sensor.pv_inverter_daily_yield_1_2 : unknown
```

opt12, 2023-12-19:
Hello Thijs,
thnaks for the sudden service! I can confirm, that the new 1.3.0.b1 starts up properly on my system and all values I'm aware of are available.
I'll have an eye on it toorrow during daytime.

In the startuop log I only have the usual "huawei_solar takes ore than 10 seconds". However, this is normal as it takes its time to establish the modbus communication.

KUDOS!

scrichab, 2023-12-20:
didn´t find the 'key' for install beta or other/older version - didn´t have to chose other version on reload button 
instead the actual version will be loaded

[image]

r13ssh, 2023-12-20:
Just a quick follow-up: The 1.3.0.b1 release seems to be working fine, also reporting current solar production,... without any issues.

opt12, 2023-12-20:
@scrichab 
To downgrade or use the beta release, you may follow these steps:
- In HACS search for the Huawei integration. 
- open it by clicking.
- in the Three-Dots-Menu, chose "reload" (or whatever it is called in your language)
[image]
- In the dialog activate the "Beta channel"
[image]
- Chose the proper version
- Enjoy

scrichab, 2023-12-20:
thank you for the quick help

danise76, 2023-12-20:
Hi.

Just installed 1.3.0b1 and seems to work with no probelms.

Thanks a lot.

scrichab, 2023-12-20:
the same here 
thank you

tp-1400, 2023-12-21:
I was fighting with this way too long. Only to find out that you have solved it already. Thank you!

commit 7b65137f (2023-12-19): Remove suffix logic (broken due to translation support)

Fixes #558