wlcrs/huawei_solar #563: [Bug]: Nach Update auf V1.2.9 funktioniert huawei-solar nicht mehr, Neuinstallation legt keinen Entitäten an

TMoersch, 2023-12-21:
### Describe the issue

Hallo,
hatt eien funktionierenden V1.2.8. Nach dem Update auf 1.2.9 stellte huawei_solar die Funktion vollständig ein. Nach einer Neuinstallation findet er laut Diagnosinfomationen alle WR, Batterien usw. aber er legt keine Entitäten mehr an, so das mein Dashboard nicht mehr funktionieren kann. Ich habe auch gepfüft ob die Entitäten umbenannt wurden, sie sind nicht vorhanden.
Slave ID sind geprüft und korrekt.

Was kann ich tun?

### Bescribe your Huawei Solar Setup

Alle Komponenten haben die aktuellste im FS Portal freigegebene SW Version
2x WR kaskadiert: SUN 2000 10KTL-M! und 4-KTL-M1 mit SW Version 158
2x LUNA 2000 mit SW Version 123
7 Optimierer mit SW Version 019
MBUS SW Version 330

### How do you connect to the inverter?

Via the SDongle, wired connection

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.
[config_entry-huawei_solar-0d4e9fa5775f5c756dd42952fc66211c.json.txt](https://github.com/wlcrs/huawei_solar/files/13741507/config_entry-huawei_solar-0d4e9fa5775f5c756dd42952fc66211c.json.txt)

Wie man sieht findet er alle Geräte und Entitäten und hat auch die passende Einträge, aber er legt die Entitäten in HA nicht an

[home-assistant_huawei_solar_2023-12-21T12-03-05.430Z.log](https://github.com/wlcrs/huawei_solar/files/13741566/home-assistant_huawei_solar_2023-12-21T12-03-05.430Z.log)

[image]

### Upload your relevant debug logs

```text
sind angehängt
```

### Please confirm the following:

- [X] The problem is still present in the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

stefanhapper, 2023-12-21:
Es gibt wohl einen Bug in der Version 1.2.9 wen man mehrere Wechselrichter hat. Im Moment gibt es eine Beta 1.3.0 die das wohl behebt.

---- On Thu, 21 Dec 2023 13:05:43 +0100 TMoersch ***@***.***> wrote ---

Describe the issue

Hallo,
 hatt eien funktionierenden V1.2.8. Nach dem Update auf 1.2.9 stellte huawei_solar die Funktion vollständig ein. Nach einer Neuinstallation findet er laut Diagnosinfomationen alle WR, Batterien usw. aber er legt keine Entitäten mehr an, so das mein Dashboard nicht mehr funktionieren kann. Ich habe auch gepfüft ob die Entitäten umbenannt wurden, sie sind nicht vorhanden.
 Slave ID sind geprüft und korrekt.
Was kann ich tun?

Bescribe your Huawei Solar Setup

Alle Komponenten haben die aktuellste im FS Portal freigegebene SW Version
 2x WR kaskadiert: SUN 2000 10KTL-M! und 4-KTL-M1 mit SW Version 158
 2x LUNA 2000 mit SW Version 123
 7 Optimierer mit SW Version 019
 MBUS SW Version 330
How do you connect to the inverter?

Via the SDongle, wired connection

Upload your Diagnostics File

Drag & Drop your Diagnostics File here.
 https://github.com/wlcrs/huawei_solar/files/13741507/config_entry-huawei_solar-0d4e9fa5775f5c756dd42952fc66211c.json.txt
Wie man sieht findet er alle Geräte und Entitäten und hat auch die passende Einträge, aber er legt die Entitäten in HA nicht an

https://github.com/wlcrs/huawei_solar/files/13741566/home-assistant_huawei_solar_2023-12-21T12-03-05.430Z.log

https://github.com/wlcrs/huawei_solar/assets/115359669/b4022719-0fef-42bb-be4d-85b931c2c443

Upload your relevant debug logs

sind angehängt

Please confirm the following:

The problem is still present in the latest release of this integration.

I did not find an existing issue describing this problem.

I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'

I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

TMoersch, 2023-12-21:
Ja, gerade gesehen das unter "Closed" eine Anleitung steht wie man an die neuen Beta kommt. Werde die jetzt testen,

TMoersch, 2023-12-21:
Lösung ist bereits veröffentlicht. sorry

closed 2023-12-21 by TMoersch