wlcrs/huawei_solar #565: [Bug]: After Update from 1.2.8 to 1.3.0,  Battery and meter ok, but inverter is offline: "unexpected keyword argument 'serial_number'"

jires, 2023-12-22:
### Describe the issue

After upgrading to latest version 1.3.0 from 1.2.7, integration starts, battery and power meter entities are ok and updating, but all inverter's entities are unavailable. It looks like my inverter's serial number is not recognized properly. Same issue when upgrading to 1.2.9.
Latest working version for me is 1.2.8
I can provide serial number of my inverter via PM or mail, if it can help.

### Bescribe your Huawei Solar Setup

Inverter Type: SUN2000-10KTL-M1
Inverter Firmware version: V100R001C00SPC158
SDongle present: yes 
Power meter present: three phase 
Battery:  LUNA2000 25kWh
Battery Firmware version: V100R002C00SPC123 

### How do you connect to the inverter?

Please select your connection method

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.
[config_entry-huawei_solar-cfc256bc12e768c9e9f3379cb7f0f2f3.json (1).txt](https://github.com/wlcrs/huawei_solar/files/13747173/config_entry-huawei_solar-cfc256bc12e768c9e9f3379cb7f0f2f3.json.1.txt)

### Upload your relevant debug logs

```text
https://www.dropbox.com/scl/fi/ou5tapclcmzwb5uiwpyl2/home-assistant_huawei_solar_2023-12-22T00-12-35.205Z.log?rlkey=dxieovm29aasuat6wbiw7vqh4&dl=0
```

### Please confirm the following:

- [X] The problem is still present in the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

wlcrs, 2023-12-22:
Can you please verify that the problem is still present when you upgrade your HA to the latest version?

I was very confused about this error as the keyword argument 'serial_number' does exist in the function that is complaining: https://github.com/home-assistant/core/blob/7ef20c4431b6a3f1e262272d639940ddd7c92456/homeassistant/helpers/device_registry.py#L501 . However, looking at the history of that function, there was a change only 2 months ago to that function. So I suspect that the problem is caused by an incompatibility between your older HA and this integration.

I'm not entirely sure what change has caused the compatibility issue, as I didn't touch any code directly related to the problematic area for a long time 🤔.

tscherf, 2023-12-22:
I had the same issue and can confirm that after upgrading HA to version 2023.12.3 everything is working ok again.

commit 6b27c428 (2023-12-22): Update HA version requirement

Fixes #565
closed 2023-12-22 by wlcrs