wlcrs/huawei_solar #566: [Feature Request]: Disable translation of sensor names (only friendly_name should be translated)

higur123, 2023-12-22:
### Describe your feature request

After updating to 1.2.9 / 1.3.0 all my sensors got added with new names (german translation).
Now all my automations / scripts / template sensors doesn't work anymore and I have to rename everything.

Furthermore all information based on sensor name will not work in every HA installation (e.g. information / templates / dashboards provided on the Wiki page or other people). Everybody have to adjust all sensor names to their own installation (if its not in english).

Is it possible to disable the translation of the sensor name instead just add translation to "friendly_name". In my opinion the name should be treated purely as a "technical name" and doesn't need any translation. 

### Proper usage

- [X] I confirm that this is not a bug report or support request
- [X] I confirm that this feature request is within the stated scope of the integration

wlcrs, 2023-12-22:
This is something that is out of my hands as an integration developer: AFAIK HA automatically generates the entity names based on the friendly name.

You're welcome to point me into the right direction on this, but after reviewing the [Developers documentation on entities](https://developers.home-assistant.io/docs/core/entity/), I did not find any way to achieve this.

alexdelli, 2023-12-22:
I have changed every formula with italian language. A big work, but I don't know if I did the best choice. :(

xm4rcell0x, 2023-12-25:
i have the exact same problem
my automations doesn't work anymore, even if i adjust the entity name...

xm4rcell0x, 2023-12-25:
Logger: homeassistant.components.automation.attivazione_autoconsumo
Source: helpers/script.py:468
Integration: Automazione (documentation, issues)
First occurred: 12:08:27 (3 occurrences)
Last logged: 12:17:04

Attivazione autoconsumo: Error executing script. Unexpected error for call_service at pos 1: Option massimizzare autoconsumo not valid for select.battery_working_mode
Attivazione autoconsumo: Error executing script. Unexpected error for call_service at pos 1: Option Massimizzare autoconsumo not valid for select.battery_working_mode
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 468, in _async_step
    await getattr(self, handler)()
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 704, in _async_call_service_step
    response_data = await self._async_run_long_action(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 666, in _async_run_long_action
    return long_task.result()
           ^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2067, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2104, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 272, in handle_service
    return await service.entity_service_call(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 878, in entity_service_call
    single_response = await _handle_entity_call(
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 948, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/select/__init__.py", line 105, in async_select_option
    raise ValueError(f"Option {option} not valid for {entity.entity_id}")
ValueError: Option massimizzare autoconsumo not valid for select.battery_working_mode

xm4rcell0x, 2023-12-27:
> This is something that is out of my hands as an integration developer: AFAIK HA automatically generates the entity names based on the friendly name.
> 
> You're welcome to point me into the right direction on this, but after reviewing the [Developers documentation on entities](https://developers.home-assistant.io/docs/core/entity/), I did not find any way to achieve a 'stable' entity name.
> 
> One possible workaround: when adding the integration to HA, you can temporarily switch your HA installation to English. Once the entities have been created, you can switch back to your language of choice. This works because the entity name is generated upon the first appearance in your HA installation and then doesn't change once you switch languages (again).

I have done a test with 2023.11.3
with this version i have under the entity: select.battery_working_mode 3 options
1) Maximize Self Consumption
2) Time Of Use
3) Fully Fed To Grid

in this setup my automation works, and i can manually call the service to test it.
-------------------
With 2023.12.x under the entity: select.battery_working_mode i have 5 options (translated in italian, even if i have the english language in general settings)

1) Massimizzare autoconsumo
2) Adattivo
3) Orario di utilizzo
4 & 5 don't remember

in this setup no matter how i write the options, it doesn't works anymore. I have already tried to name with the old options name but nothing changed if i call the service!
It only works if i change manually in the dropdown menu, no automation and no manually service call works.

-------------------
Logger: homeassistant.helpers.script.websocket_api_script
Source: helpers/script.py:468
First occurred: 15:29:09 (13 occurrences)
Last logged: 15:34:48

websocket_api script: Error executing script. Unexpected error for call_service at pos 1: Option Massimizzare_autoconsumo not valid for select.battery_working_mode
websocket_api script: Error executing script. Unexpected error for call_service at pos 1: Option massimizzare_autoconsumo not valid for select.battery_working_mode
websocket_api script: Error executing script. Unexpected error for call_service at pos 1: Option massimizzare autoconsumo not valid for select.battery_working_mode
websocket_api script: Error executing script. Unexpected error for call_service at pos 1: Option Time of use not valid for select.battery_working_mode
websocket_api script: Error executing script. Unexpected error for call_service at pos 1: Option Orario di utilizzo not valid for select.battery_working_mode
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 468, in _async_step
    await getattr(self, handler)()
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 704, in _async_call_service_step
    response_data = await self._async_run_long_action(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 666, in _async_run_long_action
    return long_task.result()
           ^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2067, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2104, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 272, in handle_service
    return await service.entity_service_call(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 878, in entity_service_call
    single_response = await _handle_entity_call(
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 948, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/select/__init__.py", line 105, in async_select_option
    raise ValueError(f"Option {option} not valid for {entity.entity_id}")
ValueError: Option Time Of Use not valid for select.battery_working_mode

kristianolofsson, 2024-01-04:
I had to use the following states to make it work (Including the underscore).

options:
  - adaptive
  - fixed_charge_discharge
  - maximise_self_consumption
  - fully_fed_to_grid
  - time_of_use_luna2000

I also wonder why the time_of_use is now including the suffix "luna200"

xm4rcell0x, 2024-01-05:
> * time_of_use_luna2000

You are my hero! thank you so much! 
This options works!!!

closed 2024-01-07 by wlcrs

smoki3, 2024-01-16:
Should it not work with coding unique ids?
[https://developers.home-assistant.io/docs/entity_registry_index?_highlight=entity&_highlight=id#unique-id](url)

The is something mentiones with:

`Being registered has the advantage that the same entity will always get the same entity ID. It will also prevent other entities from using that entity ID.`