wlcrs/huawei_solar #573: [Bug]: Got error while reading from register 37760 with length 28: Exception Response(131, 3, IllegalAddress)

DanieleCar, 2023-12-27:
### Describe the issue

The integration give me this error 
Got error while reading from register 37760 with length 28: Exception Response(131, 3, IllegalAddress)

same as https://github.com/wlcrs/huawei_solar/issues/35#issue-1156872651 , but i cannot reopen the bug

### Bescribe your Huawei Solar Setup

Inverter Type: Sun2000L 5ktl
Inverter Firmware version: V100R001C00SPC335
SDongle present: no
Power meter present: single phase
Battery: LG RESU 9.8kWh
Battery Firmware version: na

### How do you connect to the inverter?

Please select your connection method

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.
[home-assistant_huawei_solar_2023-12-27T10-16-43.714Z.log](https://github.com/wlcrs/huawei_solar/files/13777789/home-assistant_huawei_solar_2023-12-27T10-16-43.714Z.log)

### Upload your relevant debug logs

```text
2023-12-27 11:16:35.881 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37760 with length 28 from slave 0
2023-12-27 11:16:35.881 DEBUG (MainThread) [pymodbus.logging] send: 0x0 0x11 0x0 0x0 0x0 0x6 0x0 0x3 0x93 0x80 0x0 0x1c
2023-12-27 11:16:35.881 DEBUG (MainThread) [pymodbus.logging] Adding transaction 17
2023-12-27 11:16:35.888 DEBUG (MainThread) [pymodbus.logging] recv: 0x0 0x11 0x0 0x0 0x0 0x3 0x0 0x83 0x2 old_data:  addr=None
2023-12-27 11:16:35.888 DEBUG (MainThread) [pymodbus.logging] Processing: 0x0 0x11 0x0 0x0 0x0 0x3 0x0 0x83 0x2
2023-12-27 11:16:35.888 DEBUG (MainThread) [pymodbus.logging] Factory Response[131]
2023-12-27 11:16:35.889 DEBUG (MainThread) [pymodbus.logging] Getting transaction 17
2023-12-27 11:16:35.889 DEBUG (MainThread) [custom_components.huawei_solar] Finished fetching 2101073796HVHB005666_data_update_coordinator data in 0.243 seconds (success: False)
```

### Please confirm the following:

- [ ] The problem is still present in the latest release of this integration.
- [ ] I did not find an existing issue describing this problem.
- [ ] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [ ] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

wlcrs, 2023-12-27:
You are possibly the first person with an LG battery + 1st gen Huawei inverter using this integration.

Can you please download the latest firmware for your inverter from the Huawei website? The zip file contains a pdf with the modbus registers that your inverter supports. Can you upload that pdf? I only have access to the firmware files of my own 2nd gen inverter.

DanieleCar, 2023-12-27:
[SUN2000L V100R001 MODBUS Interface Definitions.pdf](https://github.com/wlcrs/huawei_solar/files/13778698/SUN2000L.V100R001.MODBUS.Interface.Definitions.pdf)

Thanks for your support , here the pdf

DanieleCar, 2023-12-27:
> [SUN2000L V100R001 MODBUS Interface Definitions.pdf](https://github.com/wlcrs/huawei_solar/files/13778698/SUN2000L.V100R001.MODBUS.Interface.Definitions.pdf)
> 
> Thanks for your support , here the pdf

Can i help in any way?

wlcrs, 2023-12-29:
I've done some extra research, and your inverter can be upgraded to **V200**R001C00 firmware according to the "Upgrade Guide" pdf which comes with the release notes of these newer firmware versions like [SUN2000L V200R001C00SPC136](https://support.huawei.com/enterprise/en/fusionsolar/sun2000l-pid-22027611/software/261342745):

[image]

Please upgrade your inverter to the latest **V200** firmware to make it compatible with this integration. I have no intention to make this integration compatible with obsolete firmware versions.

Please note that I cannot provide any support or further assistance for this upgrade process. Proceed at your own risk.

closed 2023-12-29 by wlcrs

DanieleCar, 2023-12-29:
thanks for you response , i tryed the firmware upgrade but the app say that v200 firmware are not compatible with my inverter , i will write to huawei support for more information.
i'm not  a programmer , but i have some knoledge if i cannot upgrade the firmware can i work on integration  with your supervision to have this integration working ?