wlcrs/huawei_solar #580: [Feature Request]: Improved sensor for HA Energy dashboard daily_yield and total_yield in scenarios with battery

Kugelfang666, 2024-01-02:
### Describe your feature request

I‘m using this integration for my setup with two cascaded inverters and one Luna battery connected to the primary inverter. 

First of all thanks for this GREAT integration

My problem is that the the daily_yield and total_yield sensors exposed by this integration behave in a way that is not useful for the energy dashboard of HA. Basically the primary inverter that features a battery does not report any yield when it it charging the battery and the consumption of my house is covered by the secondary inverter. Then again when the primary inverter is supplying electricity of the battery to my house at night it reports it via increasing the daily_yield sensor. This completely mixes up my energy dashboard. This is especially problematic since the electricity fed to the battery is counted twice, once via battery_discharge and once via increase in daily yield by the energy dashboard. 

It would be great if this integration would expose a sensor that provides the accumulated energy production by the inverter at any given time regardless wether it is supplied to the house or fed to the battery. 

I created manual sensors following the instructions provided to integrate the active_power sensor. This works well but is a major hassle for novice users. Maybe this solution could be implemented as part of this integration

[image]

[image]

### Proper usage

- [X] I confirm that this is not a bug report or support request
- [X] I confirm that this feature request is within the stated scope of the integration

wlcrs, 2024-01-02:
I understand your request and it has already come up a few times. However, the interpretation of the values supplied by inverter is outside the scope of this integration. 

I would never be able to get identical values to what FusionSolar shows, so this would be a source of endless support requests. I prefer the current solution where a few approximation options are explained in the Wiki. You're very welcome to improve the suggested solutions and document them there.

closed 2024-01-02 by wlcrs

Kugelfang666, 2024-01-02:
ok, understood, thanks for looking into this!