wlcrs/huawei_solar #582: [Bug]: Battery total charge value too big and not coherent with total discharge

rob570, 2024-01-04:
### Describe the issue

Hi,
I'm using a SUN2000-6K-TL with a LUNA 2000 15kWh. Data are retrieved by modbus+Elfin.
All values are reasonable, except for Total charging/discharging values:
[image]

On fusion solar site, I can read only the total charged, that is about 10% lower. Don't know if they consider some power efficiency loses.
[image]
Thank for your help,
  Roberto

### Bescribe your Huawei Solar Setup

Inverter Type: SUN2000-6KTL-L1 
Inverter Firmware version: V200R001C00SPC133
SDongle present: no 
Power meter present: single phase
Battery: LUNA2000 15kWh
Battery Firmware version: V100R002C00SPC123

### How do you connect to the inverter?

Over serial, with a serial-to-WiFi stick

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.

### Upload your relevant debug logs

```text
There are no relevant logs.
I've read only these, but I think they are related to startup after restarting HA:
2024-01-04 08:37:42.938 DEBUG (MainThread) [pymodbus.logging] Connecting to 192.168.100.130:9999.
2024-01-04 08:37:42.938 DEBUG (MainThread) [pymodbus.logging] Connecting comm
2024-01-04 08:37:42.958 DEBUG (MainThread) [pymodbus.logging] Connected to comm
2024-01-04 08:37:42.959 DEBUG (MainThread) [pymodbus.logging] callback_connected called
2024-01-04 08:37:42.961 DEBUG (MainThread) [huawei_solar.modbus] Waiting for 1500 milliseconds after connection before performing operations
2024-01-04 08:37:42.972 INFO (MainThread) [huawei_solar.huawei_solar] Waiting for connection ...
```

### Please confirm the following:

- [X] The problem is still present in the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

wlcrs, 2024-01-07:
You didn't provide me with any detailed debug logs to diagnose the problem. I would need logs with modbus traffic for that.

Does the value increase or decrease over time? It looks to be very close to the maximum value that can be represented by an uint32 register: 4,294,967,295

[image]

rob570, 2024-01-07:
Hi wlcrs, here it is the debug log.
As you can see the value is increasing.
[image]

Thanks,
  Rob
[HuaweiDebugLogs.txt](https://github.com/wlcrs/huawei_solar/files/13852592/HuaweiDebugLogs.txt)

wlcrs, 2024-01-07:
I've manually decoded the values that the inverter returned for the relevant register: `6554, 27969`. This translates to `0b00011001100110100110110101000001` which is `429550913`. 

This is consistent with your screenshot. 

This is not a bug in the integration. This is your inverter returning this value.  You can contact Huawei support to ask why you get this strange value when reading register 37782.

I would suggest creating a helper entity which subtracts the difference between what HA receives and what you see in FusionSolar to get an usable result.

closed 2024-01-07 by wlcrs

rob570, 2024-01-07:
Hi wlcrs,
Thank you for your time spent.
I'll try to contact huawei support, but I don't understand why the fusionsolar website shows reasonable values.
Perhaps it doesn't read the register 37782 but it sums all the battery packs. I've seen there are specific registers for every battery pack, for example:
[image]
In the meantime I'll adopt your suggested workaround.
Thanks,
   rob.