wlcrs/huawei_solar #586: [Feature Request]:

torwag, 2024-01-07:
### Describe your feature request

Just a feature request for the documentation.

As there are several ways to connect the inverters to HA, I was wondering if a ranking among them in terms of stability, features, support, might help to make the decision which way to follow.

Furthermore a table which includes which method requires which hardware and impose which constrains would be very helpful. 

Right now, there is only the remark that a wired connection via the dongle seems unreliable. 

### Proper usage

- [X] I confirm that this is not a bug report or support request
- [X] I confirm that this feature request is within the stated scope of the integration

torwag, 2024-01-07:
Sorry I messed up to add a title... Can't change it. If a good soul of the repro owners can create one... Positiv karma will flow.

AlfaCube, 2024-01-10:
Huawei solar stopped running in my HA
Got this error:
Logger: custom_components.fusion_solar.fusion_solar.kiosk.kiosk_api

Source: custom_components/fusion_solar/fusion_solar/kiosk/kiosk_api.py:47

Integration: Fusion Solar (documentation, issues)

First occurred: 12:34:59 AM (55 occurrences)

Last logged: 9:34:59 AM

Retrieving the data failed with failCode: 0, data: neteco.pvms.systemSetting notExis
 Uninstalled the integration, restatred the inverters(got two of them and one dongle).
Now in huawei sollar integration  the power meter is missing. 

can I get some help please?

thank you

torwag, 2024-01-16:
Thanks for the detailed reply. Indeed I'm just installing three converters cascaded, each connected to a battery. I will ad a pair of wires and a tcp cable to the installation, just to be prepared. That should allow me to go all options.
I will happily report how this is going. Wondering why the network connection via the sdongle seems unstable. Nevertheless, I happily would like to avoid using the cloud, if I can get all data locally.

Is there anything known already which seems to be available via cloud but not locally?

Guess I'm ready in a few weeks and most likely will have many more questions 😉

MajoV, 2024-01-18:
Hello, I don't know if this is the right place to post my experience with connecting the inverter, but I'll do it, maybe it will help someone.
I have a similar setup as @torwag , three cascaded inverters, each with its own battery and DTSU666.

My first connection to HA was made using an S-Dongle. Yes it was unstable with frequent crashes. In addition, the initial initialization of this connection took a very long time and did not always end with a successful result.

Another connection I tried was via the serial connector, of course with the S-Dongle removed. The initial initialization was very fast, the connection was amazing without any dropouts. The only problem in this connection without the S-Dongle was that the system did not work properly, because the batteries did not know when to charge or discharge, because the entire system is controlled by the S-Dongle.

The last thing I tried, and it has been working without problems and outages for two months, was connecting to the wifi of each inverter. For this, I used three pieces of USB WiFi adapter, in my case D-Link DWA-121, each piece connected to an extension USB cable and placed each by its converter to prevent WiFi interference. The only thing you have to change in the inverter is the IP address of the internal WiFi, it should be unique for each inverter. In this case, I had to initialize the Huawei Solar for each network interface. The USB WiFi adapters are connected to an Odroid C4 running Home Assistant. I publish the telemetry of the entire Huawei system via MQTT to my master Home Assistant.

torwag, 2024-01-18:
Thanks @MajoV for your (for me) very useful response. That will indeed help me to save days of fiddling, testing and head scratching. Thanks for the picture as well. Looks very nice and clean. I am just in the middle of doing all the wiring. It is indeed a crazy amount of wires getting all connected. The box seems a tidy bit oversized for an Odroid, but that might be just the perspective of the image ;)
 
Is there a reason that you choose to use three WIFI-adapters instead of one, besides of interferences? I like the idea of running a client-HA just for the PV and make sure it runs indented of what happens to the rest of the house automation, particular as a charger unit for electric vehicles will be added to the mix and I might come up with stuff like PV and weather prediction as well as real weather data.

When you said, connection via the SDongle you mean the wired connection?
I still wonder why that is so unstable, after all the purpose of that dongle should be to establish that connection.

When you said, you connected via serial, you mean via modbus, right? That was my preferred method until I read you had troubles with it. Do you have any details why the batteries couldn't be set to charge or discharge via modbus? 

Well all my questions might high-jack the purpose of a bug/feature-tracker. If there is any other way to communicate (discord, signal, etc.), please let me know. I am from Germany, if this helps.

AlfaCube, 2024-01-18:
After two weeks of running without any problems my HA decided to play games with me.

I run HA on a Intel NUC and Huawei Solar with ModBus active after two weeks has this error:
here is the error: ogger: pymodbus.logging Source: custom_components/huawei_solar/init.py:92 Integration: Huawei Solar (documentation, issues) First occurred: 12:34:58 AM (11 occurrences) Last logged: 12:46:10 AMFailed to connect [Errno 111] Connect call failed

Reboot the wifi- no change, reboot the inverters- no change, unplug the Sdongle- no change. Delete the Huawei Solar integration- only the inverter visible on Huawei Solar, no power meter. Go back to an older HA backup- no change.

Today: Huawei Solar shoes only the SUN2000-4KTL-L1, no Power Meter..

I am not an expert in HA or ITbut I managed to see this two errors:

**_This error originated from a custom integration.

Logger: custom_components.huawei_solar
Source: helpers/update_coordinator.py:306
Integration: Huawei Solar (documentation, issues)
First occurred: 6:49:08 PM (2 occurrences)
Last logged: 6:50:39 PM

Timeout fetching TA2240030061_data_update_coordinator data_**

and this one:

_**Logger: huawei_solar.huawei_solar
Source: /usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py:348
First occurred: 6:50:36 PM (2 occurrences)
Last logged: 6:50:38 PM

Modbus client is not connected to the inverter.
NoneType: None**_

Can you guys please guide me what to do next?

After a week of testing some automations involving solar pannels and electric heater I finally managed to make them run with no problems but now I got this problem.

Thank you

Strakomania, 2024-01-25:
Hey @MajoV,
I just want to try connecting my huawei inverter the same way as you did. 
My question is how did you connect the USB wifi sticks? Is it just Plug and Play or need to install an accesspoint addon? 
And wich wifi stick you can recommend? I just bought the D-Link DWA-131 but im not sure if it will work.

Maybe you can help me. 
----------
My Setup:
Homeassistant generic x64 on a IntelNuc.

torwag, 2024-01-25:
@Strakomania I am not @MajoV but the APs come from the inverters. The reason you need a single stick (interface) for each inverter is the fact, that each converter creates its own wifi network.
The sticks itself only need to run in client-Mode, which basically is the standard mode. As all this runs under Linux, you need sticks supported under Linux (which are the majority by now). I would not think to much about it, try a usb stick and if it works get a couple more.
The network config is a bit more delicated as usual (with a single interface) as you have to setup three networks.
I did not do it by myself yet, but when it comes to it, I am curious to see if I can adjust the wifi channel on the inverters to make sure they do not interfere.

Hope that helps.

Strakomania, 2024-01-25:
Thanks for your answer. 

Ok i will try a few sticks.
The Amazon description from the stick says that it will work under linux but my HASS wont find it. 

I am from Germany as well, if you want we could connect on discord.