wlcrs/huawei_solar #590: [Bug]: Inverter Daily Yield

Ivano62, 2024-01-14:
### Describe the issue

HI,
after the last update I have the impression there is a problem with the unit of measurement of the "Inverter Daily Yield" parameter, in fact, now the displayed value has five decimal places. Simply put it appears to be expressed in MWh rather than kWh.
Thank you.

### Bescribe your Huawei Solar Setup

Inverter Type: SUN2000-6KTL-L1
Inverter Firmware version: V200R001C00SPC136 
SDongle present: yes
Power meter present: single phase
Battery: LUNA2000 15kWh
Battery Firmware version: V100R002C00SPC125
[image]

### How do you connect to the inverter?

Via the SDongle, wired connection

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.

### Upload your relevant debug logs

```text
.
```

### Please confirm the following:

- [X] The problem is still present in the latest release of this integration.
- [ ] I did not find an existing issue describing this problem.
- [ ] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [ ] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

wlcrs, 2024-01-16:
This bug currently contains absolutely nothing useful for me to work with.

What is this entity? Did you rename an entity provided by the integration? Please include a diagnostics file and debug logs so I can verify what was outputted by the inverter. What do you see in FusionSolar web at the same time?

Ivano62, 2024-01-16:
Hi,
the entity is sensor.inverter_daily_yield, this is indicated in the image I've attached.

The current value is 0,00225 kWh, at the same time in FusionSolar is 2,95 kWh.

Before latest upgrade there isn't any problem.

Thanks

--- Messaggio originale ---

Data:   16 gennaio 2024 10:16:23
A:      wlcrs/huawei_solar ***@***.***
Cc:     Ivano62 ***@***.***, Author ***@***.***

wlcrs, 2024-01-16:
Please provide everything I ask. You MUST provide me with both the diagnostics file and debug logs containing at least 10 minutes of modbus traffic, otherwise I'm unable to investigate. Otherwise I have insufficient info to work with.

Ivano62, 2024-01-16:
I'm sorry, but at the moment I couldn't provide the data you asked.
However, comparing the new and old versions of the "sensor.py" file I observe this difference

- Old (**KILO_WATT**)
     HuaweiSolarSensorEntityDescription(
         key=rn.DAILY_YIELD_ENERGY,
         icon="mdi:solar-power",
         native_unit_of_measurement=ENERGY_KILO_WATT_HOUR,
         device_class=SensorDeviceClass.ENERGY,
         state_class=SensorStateClass.TOTAL_INCREASING,
     ),

- New (**WATT**)
     HuaweiSolarSensorEntityDescription(
         key=rn.DAILY_YIELD_ENERGY,
         icon="mdi:solar-power",
         native_unit_of_measurement=UnitOfEnergy.WATT_HOUR,
         device_class=SensorDeviceClass.ENERGY,
         state_class=SensorStateClass.TOTAL_INCREASING,
     ),

I hypothesize that the problem may come from here.

Thank you anyway.

wlcrs, 2024-01-16:
Thanks for the pointer. Fixed.

As that change has not yet been released I did not take it into account as a possible source. I guess you updated outside of HACS?

closed 2024-01-16 by wlcrs

Ivano62, 2024-01-17:
Hi,
I've installed latest vesion, now it seems work fine.

Thanks a lot