wlcrs/huawei_solar #591: [Feature Request]:

Trickboxx, 2024-01-15:
### Describe your feature request

I use a Huawei SUN2000 6KTL-M1. The Installation works fine over Network but for the Home Assistant Energy Dashboard I need the "Inverter Input Power" but I cannot use it only  "Inverter Daily Yield" is available.
Why is it not possible to use the right Entity "Inverter Input Power"? The Entity "Inverter Input Power" is available but not in the Energy Dashboard for SOLAR PRODUCTION.
Thank You.

### Proper usage

- [X] I confirm that this is not a bug report or support request
- [X] I confirm that this feature request is within the stated scope of the integration

wlcrs, 2024-01-16:
This is working as intended: the energy dashboard does not use the momentarily energy output of the inverter, but uses accumulative values to calculate the production every hour.

closed 2024-01-16 by wlcrs

echosmart792, 2024-01-20:
What entities must be added for example to: 
-- Electricity grid -> Grid consumption 
-- Electricity grid -> Return to grid 
-- Solar panels -> Solar production 
to have a correct electrical map?

nurzen, 2024-01-31:
For the first two you need an extra electricty meter like a DTSU666 because your inverter cannot know that. Maybe the meter from your provider can provide that information, some of them have an infrared doide you can read the values from. You can look here for details https://esphome.io/components/sml.html
The last value is inverter_total_yield from this integration.

echosmart792, 2024-02-07:
Hi. I have a smartmeter, which communicates correctly with the inverter, but this dashboard does not allow me to add instant consumption and instant production, but only the total for the current day, and that also appears in the electrical dashboard.