wlcrs/huawei_solar #592: [Bug]: Entity ID also Translated

smoki3, 2024-01-16:
### Describe the issue

Since the introduction of the translations also the entity IDs are translated if you readd the your PV setup.
This lead to major issues. That if you have to reconfigure your setup, you have to change all you automations which are related to any entity of the Huawei setup.

For Example:
before: sensor.inverter_active_power
after: after: sensor.inverter_wirkleistung

Normale the translations are only applied to the entity name not the entity IDs

### Bescribe your Huawei Solar Setup

Inverter Type:Sun2000-6KTL-M1
SDongle present: yes 
Power meter present: three phase
Battery: LUNA2000 5kWh 

### How do you connect to the inverter?

Via the SDongle, wireless connection

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.
[config_entry-huawei_solar-2846ec22b29b09f009239187ef879c41.json (1).txt](https://github.com/wlcrs/huawei_solar/files/13947520/config_entry-huawei_solar-2846ec22b29b09f009239187ef879c41.json.1.txt)

### Upload your relevant debug logs

```text
-
```

### Please confirm the following:

- [X] The problem is still present in the latest release of this integration.
- [ ] I did not find an existing issue describing this problem.
- [ ] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [ ] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

wlcrs, 2024-01-16:
Duplicate of #566 

TL;DR : This is a "feature" of HA itself that I cannot influence. Please let me know if I'm wrong by pointing me to the developer documentation that explains what I should do differently.

closed 2024-01-16 by wlcrs

offtopic, 2024-11-01:
what about the closed PR #739 - would this be the solution?

I also have big troubles because all those sensor names are german now ... it is not funny to rename ~500 entities manually.
switching home assistant to english end re-adding the integration has not helped -> still german names .. so i assume this is read from huawei solar itself?

smoki3, 2024-11-01:
Just set HA to Englisch, delete your device and readd it. Then you can set HA to German again. The entities stays englisch