wlcrs/huawei_solar #596: [Feature Request]: Add status sensor of "forcible Charge/Discharge" services

heinemannj, 2024-01-19:
### Describe your feature request

Actually I'm implementing the "**EMHASS: Energy Management Optimization for Home Assistant**".
https://github.com/davidusb-geek/emhass

This Add-On is performing more or less **to predict the next 24h battery usage** based on:

- house-load prediction for the next 24h (https://github.com/davidusb-geek/emhass)
- PV production forecast based on SolCast integration (https://github.com/oziee/ha-solcast-solar)
- EPEXSpot price forcast based on epexspot integration (https://github.com/mampfes/ha_epex_spot)
- EPEXSpot price prediction  for the next 24h (https://github.com/davidusb-geek/emhass)

[image]

For HA Huawei Solar automations I'm actually using Node-RED:

[image]

**Sheduled every minute!**

- For the **Battery Charge from Grid part** there is an Huawei Solar **switch which can be used to monitor** the actual status.
- For the **Battery Forcible Charge part unfortunately not**.

=> leading to the situation that the **inverter is penetrated with service calls every minute**:

In FusioSolar portal Battery Forcible Charge changes are directly visible => meaning there must be inverter registers available.

`Can you please provide a solution to have this status of Battery Forcible Charge within HA?`

HA call service:

[image]

In FusionSolar above HA service call is directly visible as the following:

[image]

Stop Forcible Chare via HA service call:

[image]

Output in FusionSolar:

[image]

### Proper usage

- [X] I confirm that this is not a bug report or support request
- [X] I confirm that this feature request is within the stated scope of the integration

wlcrs, 2024-01-19:
Making that value available would require a separate extra call to the inverter: it cannot be bundled with the existing requests being made to the inverter.

I must strike a careful balance between the number of requests I make to the inverter and their utility. I seem to have reached a point at which adding extra calls will make the integration unreliable for a considerable number of users (especially those connecting via an SDongle, even more so if they have multiple inverters connected in cascade).

As you can easily store this value yourself in a helper entity when you start/stop a forcible (dis)charge, I see little utility in adding this as an extra sensor. Meanwhile, the downside is pretty big for a considerable amount of users.

Feel free to add a [custom sensor](https://github.com/wlcrs/huawei_solar/wiki/Defining-custom-sensors) yourself if you really want to read it from the inverter instead.

closed 2024-01-19 by wlcrs

MaartenDams, 2024-01-23:
@wlcrs ,
I do love the integration , just playing around with it for the last 48h.
How great it works, how stupid I feel for posting this but I can't get my head around (and can't find any decent documentation) on how to change the SOC to an entity value.
[image]

Can you please enlighten my spirit in what syntax should be used? 
It's the final piece of the charge_at_low_hours_puzzel :)
Thanks,
Maarten

heinemannj, 2024-01-24:
What's about: {{ states('sensor.your_sensor_entity') }}

MaartenDams, 2024-01-24:
@heinemannj , I tried exactly that. 
It casts back that it needs a float. (value is for example 66)

Neither  target_soc: {{ (states('sensor.oplaadniveau_batterij_eerste_uur')|float) }}
nor {{ float(states('sensor.oplaadniveau_batterij_eerste_uur')) }} works.