wlcrs/huawei_solar #598: Battery Backup power SOC

scorpio2k2, 2024-01-19:
### Describe the issue

Hello,

"Battery Backup power SOC" sync between Solar app and HAS is slow. 
I want to automate this because I don't want the battery to empty when the wife starts the washing masine at night but I noticed that when I change the value of "Battery Backup power SOC" in Solar app, it's updated very late in HAS. 

Any ideas?

Thanks

### Bescribe your Huawei Solar Setup

Inverter Type:
Inverter Firmware version:
SDongle present: yes / no
Power meter present: three phase / single phase / no
Battery: LG RESU xxkWh / LUNA2000 xxkWh / None
Battery Firmware version:

### How do you connect to the inverter?

Please select your connection method

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.

### Upload your relevant debug logs

```text
no logs available for this delay.
```

### Please confirm the following:

- [X] The problem is still present in the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

wlcrs, 2024-01-19:
The update interval for configuration changes is [15 minutes](https://github.com/wlcrs/huawei_solar/blob/main/const.py#L23). This is a trade off to keep the integration stable: more frequent updates increase the chance of overloading the inverter with requests and getting timeouts for all values queried by the integration.

Feel free to change that value in your own installation, but I will not change this globally, nor make it configurable.

closed 2024-01-19 by wlcrs

scorpio2k2, 2024-01-19:
When I'm changing the value from FusionSolar app it's changed in HA (in 15min. I'm OK with it and I understand the reasons).
Do you know why when I change the value in HA, is not also changed in FusionSolar app?

wlcrs, 2024-01-19:
No, I do not. I have no knowledge outside of how to communicate with the inverter and all the pitfalls that come with that.

scorpio2k2, 2024-01-19:
Correct, the integration communicates with the inverter. My issue is that I change something in HA and is not reflected to the inverter.

wlcrs, 2024-01-19:
Feel free to submit detailed debugging logs and a diagnostics file (ideally before and after the change to the value) if you want me to investigate. Otherwise I have nothing to work with.

scorpio2k2, 2024-01-19:
I think I understand now. The integration talks with the inverter but Huawei cloud is not checking the inverter for the configs and it expects that all configs are pushed only from App to inverter. 

So, if I choose to use the integration with HA, I should not rely on configuration seen in the app or not use the app at all.

scorpio2k2, 2024-01-20:
The real issue is that ven if "charge from grid is off", if I touch the "Battery Backup power SOC" the battery start charging  to the level I set. That's not normal.