wlcrs/huawei_solar #602: [Bug]: home assistant not starting due too this integration

Vato9394, 2024-01-26:
### Describe the issue

home assistant version 2024.1.5
integration version: 1.3.1

I noticed that home assistant was not starting, I did some troubleshooting (disabled all integrations and started them 1 by 1)
The issue seems to be this integration.
However, I cannot find anything in the logs.
my huawei sun2000L is integrated over modbus LAN.

There where some changes in the 2024.1 regarding modbus, maybe this has something to do with this issue?
Remove “swap: none” from modbus ([@janiversen](https://github.com/janiversen) - [#104713](https://github.com/home-assistant/core/pull/104713))
Deprecate and remove lazy_error from modbus ([@janiversen](https://github.com/janiversen) - [#105037](https://github.com/home-assistant/core/pull/105037))
Add Modbus fan speed support ([@cian](https://github.com/cian) - [#104577](https://github.com/home-assistant/core/pull/104577))
Reduce modbus validator for “swap” (remove special handling) ([@janiversen](https://github.com/janiversen) - [#105021](https://github.com/home-assistant/core/pull/105021))
Deprecate parameter “retries” in modbus ([@janiversen](https://github.com/janiversen) - [#105024](https://github.com/home-assistant/core/pull/105024))
Remove default value from modbus retries ([@janiversen](https://github.com/janiversen) - [#106551](https://github.com/home-assistant/core/pull/106551))
Remove default value for modbus lazy_error ([@janiversen](https://github.com/janiversen) - [#106561](https://github.com/home-assistant/core/pull/106561))

### Bescribe your Huawei Solar Setup

Inverter Type: SUN2000L
Inverter Firmware version: no clue
SDongle present: no
Power meter present: no
Battery: None
Battery Firmware version: /

### How do you connect to the inverter?

Via the `SUN2000-<serial_no>` WiFi

### Upload your Diagnostics File

no logs cannot start with integration enabled

### Upload your relevant debug logs

```text
no logs cannot start with integration enabled
```

### Please confirm the following:

- [X] I'm running the latest release of Home Assistant.
- [X] I'm running the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

Vato9394, 2024-01-26:
maybe caused by ABCCachedProperties metaclass on Entity?

Andurilll, 2024-01-27:
To provide Logs disconnect the inverter after Setting the Integration up and reboot

What Software Version are you using in the inverter? Dongle? Mbus?

Vato9394, 2024-01-27:
the inverter is connected directly to the home wifi. It was working on 2023.12. So no dongle of some sort.
I have no idea what software version the inverter is using, I am not a fan of using the chinese app.

Andurilll, 2024-01-27:
I definitely recommend updating the inverter & everything to the newest Version

Also possible from the Browser oder Website

wlcrs, 2024-02-24:
no logs = nothing for me to debug. Closing as this seems to be an isolated issue.

closed 2024-02-24 by wlcrs