wlcrs/huawei_solar #603: [Bug]: Automation with changing Battery Working Mode not longer working

pschraml, 2024-01-26:
### Describe the issue

I used autoamtion to change the working mode of the battery from Maximum self-service to Time Of use but since the update, the Working Modes in Huawei Integration have changed 

### Bescribe your Huawei Solar Setup

Inverter Type:
Inverter Firmware version:
SDongle present: yes / no
Power meter present: three phase / single phase / no
Battery: LG RESU xxkWh / LUNA2000 xxkWh / None
Battery Firmware version:

### How do you connect to the inverter?

Via the `SUN2000-<serial_no>` WiFi

### Upload your Diagnostics File

Working modes in Integration are now called different so selecting TOU does not trigger change anymore as the working mode is now called time_of_use_luna2000:

Referenced entities select.battery_working_mode are missing or not currently available
18:39:38 – (WARNING) helpers/service.py

[image]
[image]

### Upload your relevant debug logs

```text
2024-01-26 18:39:38.666 WARNING (MainThread) [homeassistant.helpers.service] Referenced entities select.battery_working_mode are missing or not currently available
2024-01-26 18:59:33.967 DEBUG (MainThread) [pymodbus.logging] send: 0x1 0x1d 0x0 0x0 0x0 0x6 0x0 0x6 0xc3 0x4f 0x0 0x1
2024-01-26 18:59:33.969 DEBUG (MainThread) [pymodbus.logging] Adding transaction 285
2024-01-26 18:59:34.007 DEBUG (MainThread) [pymodbus.logging] recv: 0x1 0x1d 0x0 0x0 0x0 0x6 0x0 0x6 0xc3 0x4f 0x0 0x1 old_data:  addr=None
2024-01-26 18:59:34.008 DEBUG (MainThread) [pymodbus.logging] Processing: 0x1 0x1d 0x0 0x0 0x0 0x6 0x0 0x6 0xc3 0x4f 0x0 0x1
2024-01-26 18:59:34.008 DEBUG (MainThread) [pymodbus.logging] Factory Response[WriteSingleRegisterResponse': 6]
2024-01-26 18:59:34.009 DEBUG (MainThread) [pymodbus.logging] Getting transaction 285
2024-01-26 18:59:34.011 DEBUG (MainThread) [huawei_solar.huawei_solar] Heartbeat succeeded
2024-01-26 18:59:49.013 DEBUG (MainThread) [pymodbus.logging] send: 0x1 0x1e 0x0 0x0 0x0 0x6 0x0 0x6 0xc3 0x4f 0x0 0x1
2024-01-26 18:59:49.015 DEBUG (MainThread) [pymodbus.logging] Adding transaction 286
2024-01-26 18:59:49.204 DEBUG (MainThread) [pymodbus.logging] recv: 0x1 0x1e 0x0 0x0 0x0 0x6 0x0 0x6 0xc3 0x4f 0x0 0x1 old_data:  addr=None
2024-01-26 18:59:49.204 DEBUG (MainThread) [pymodbus.logging] Processing: 0x1 0x1e 0x0 0x0 0x0 0x6 0x0 0x6 0xc3 0x4f 0x0 0x1
2024-01-26 18:59:49.204 DEBUG (MainThread) [pymodbus.logging] Factory Response[WriteSingleRegisterResponse': 6]
2024-01-26 18:59:49.205 DEBUG (MainThread) [pymodbus.logging] Getting transaction 286
2024-01-26 18:59:49.205 DEBUG (MainThread) [huawei_solar.huawei_solar] Heartbeat succeeded
2024-01-26 18:59:55.179 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064 with length 52 from slave 0
2024-01-26 18:59:55.179 DEBUG (MainThread) [pymodbus.logging] send: 0x1 0x1f 0x0 0x0 0x0 0x6 0x0 0x3 0x7d 0x40 0x0 0x34
2024-01-26 18:59:55.180 DEBUG (MainThread) [pymodbus.logging] Adding transaction 287
2024-01-26 18:59:55.757 DEBUG (MainThread) [pymodbus.logging] recv: 0x1 0x1f 0x0 0x0 0x0 0x6b 0x0 0x3 0x68 0x0 0x0 0x0 0x0 0x9 0xb3 0x0 0x0 0x0 0x0 0x4 0xcf 0x0 0x1 0x0 0x0 0x0 0x0 0x38 0xfd 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0xe 0x60 0x0 0x0 0xe 0xf 0x0 0x0 0x0 0x10 0x3 0xe8 0x13 0x85 0x27 0x10 0x0 0xe0 0xb 0xb8 0x2 0x0 0x0 0x0 0x65 0xb3 0x69 0x12 0xff 0xff 0xff 0xff 0x0 0x0 0xe 0x12 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x1 0xd9 0xa8 0x0 0x7 0xe0 0xe7 0x0 0x7 0x48 0x5f 0x65 0xb4 0x1 0x30 0x0 0x0 0x0 0x6f 0x0 0x0 0x5 0xc5 old_data:  addr=None
2024-01-26 18:59:55.758 DEBUG (MainThread) [pymodbus.logging] Processing: 0x1 0x1f 0x0 0x0 0x0 0x6b 0x0 0x3 0x68 0x0 0x0 0x0 0x0 0x9 0xb3 0x0 0x0 0x0 0x0 0x4 0xcf 0x0 0x1 0x0 0x0 0x0 0x0 0x38 0xfd 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0xe 0x60 0x0 0x0 0xe 0xf 0x0 0x0 0x0 0x10 0x3 0xe8 0x13 0x85 0x27 0x10 0x0 0xe0 0xb 0xb8 0x2 0x0 0x0 0x0 0x65 0xb3 0x69 0x12 0xff 0xff 0xff 0xff 0x0 0x0 0xe 0x12 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x1 0xd9 0xa8 0x0 0x7 0xe0 0xe7 0x0 0x7 0x48 0x5f 0x65 0xb4 0x1 0x30 0x0 0x0 0x0 0x6f 0x0 0x0 0x5 0xc5
2024-01-26 18:59:55.758 DEBUG (MainThread) [pymodbus.logging] Factory Response[ReadHoldingRegistersResponse': 3]
2024-01-26 18:59:55.759 DEBUG (MainThread) [pymodbus.logging] Getting transaction 287
2024-01-26 18:59:55.760 DEBUG (MainThread) [pymodbus.logging] [0, 0, 2483, 0, 0, 1231, 1, 0, 0, 14589, 0, 0, 0, 0, 0, 3680, 0, 3599, 0, 16, 1000, 4997, 10000, 224, 3000, 512, 0, 26035, 26898, 65535, 65535, 0, 3602, 0, 0, 0, 0, 0, 0, 0, 1, 55720, 7, 57575, 7, 18527, 26036, 304, 0, 111, 0, 1477]
2024-01-26 18:59:55.760 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b'\x00\x00']
2024-01-26 18:59:55.761 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b'8\xfd']
2024-01-26 18:59:55.761 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b'\x00\x00']
2024-01-26 18:59:55.761 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b'\x00\x00']
2024-01-26 18:59:55.762 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b'\x0e`']
2024-01-26 18:59:55.762 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b'\x0e\x0f']
2024-01-26 18:59:55.762 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b'\x00\x10']
2024-01-26 18:59:55.762 DEBUG (MainThread) [pymodbus.logging] handle: [b'e\xb3', b'i\x12']
2024-01-26 18:59:55.763 DEBUG (MainThread) [pymodbus.logging] handle: [b'\xff\xff', b'\xff\xff']
2024-01-26 18:59:55.763 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x07', b'\xe0\xe7']
2024-01-26 18:59:55.763 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b'\x05\xc5']
2024-01-26 18:59:55.817 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32000 with length 20 from slave 0
2024-01-26 18:59:55.818 DEBUG (MainThread) [pymodbus.logging] send: 0x1 0x20 0x0 0x0 0x0 0x6 0x0 0x3 0x7d 0x0 0x0 0x14
2024-01-26 18:59:55.819 DEBUG (MainThread) [pymodbus.logging] Adding transaction 288
2024-01-26 18:59:55.856 DEBUG (MainThread) [pymodbus.logging] recv: 0x1 0x20 0x0 0x0 0x0 0x2b 0x0 0x3 0x28 0x0 0x6 0x0 0x0 0x0 0x5 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0xfb 0xd8 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 old_data:  addr=None
2024-01-26 18:59:55.856 DEBUG (MainThread) [pymodbus.logging] Processing: 0x1 0x20 0x0 0x0 0x0 0x2b 0x0 0x3 0x28 0x0 0x6 0x0 0x0 0x0 0x5 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0xfb 0xd8 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0
2024-01-26 18:59:55.857 DEBUG (MainThread) [pymodbus.logging] Factory Response[ReadHoldingRegistersResponse': 3]
[... 114 lines cut ...]
2024-01-26 19:00:26.506 DEBUG (MainThread) [pymodbus.logging] Factory Response[ReadHoldingRegistersResponse': 3]
2024-01-26 19:00:26.506 DEBUG (MainThread) [pymodbus.logging] Getting transaction 296
2024-01-26 19:00:26.507 DEBUG (MainThread) [pymodbus.logging] [580, 0, 2, 4346, 65451, 65535, 61937, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 3, 51333, 3, 50561, 0, 547, 0, 562]
2024-01-26 19:00:26.508 DEBUG (MainThread) [pymodbus.logging] handle: [b'\xff\xff', b'\xf1\xf1']
2024-01-26 19:00:26.508 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x03', b'\xc8\x85']
2024-01-26 19:00:26.508 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x03', b'\xc5\x81']
2024-01-26 19:00:26.508 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b'\x02#']
2024-01-26 19:00:26.509 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b'\x022']
2024-01-26 19:00:26.509 DEBUG (MainThread) [custom_components.huawei_solar] Finished fetching HV2270499698_data_update_coordinator data in 0.320 seconds (success: True)
2024-01-26 19:00:34.250 DEBUG (MainThread) [pymodbus.logging] send: 0x1 0x29 0x0 0x0 0x0 0x6 0x0 0x6 0xc3 0x4f 0x0 0x1
2024-01-26 19:00:34.252 DEBUG (MainThread) [pymodbus.logging] Adding transaction 297
2024-01-26 19:00:34.290 DEBUG (MainThread) [pymodbus.logging] recv: 0x1 0x29 0x0 0x0 0x0 0x6 0x0 0x6 0xc3 0x4f 0x0 0x1 old_data:  addr=None
2024-01-26 19:00:34.291 DEBUG (MainThread) [pymodbus.logging] Processing: 0x1 0x29 0x0 0x0 0x0 0x6 0x0 0x6 0xc3 0x4f 0x0 0x1
2024-01-26 19:00:34.291 DEBUG (MainThread) [pymodbus.logging] Factory Response[WriteSingleRegisterResponse': 6]
2024-01-26 19:00:34.292 DEBUG (MainThread) [pymodbus.logging] Getting transaction 297
2024-01-26 19:00:34.294 DEBUG (MainThread) [huawei_solar.huawei_solar] Heartbeat succeeded
```

### Please confirm the following:

- [X] I'm running the latest release of Home Assistant.
- [X] I'm running the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [ ] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

pschraml, 2024-01-26:
[config_entry-huawei_solar-ce741a99bbd541890902be11edcd617f.json.txt](https://github.com/wlcrs/huawei_solar/files/14068565/config_entry-huawei_solar-ce741a99bbd541890902be11edcd617f.json.txt)

wlcrs, 2024-02-24:
You need to update your configuration to match the new values when changing the battery working mode.

This is (a rather unwanted and unintended) change due to adding support for translations. This is a limitation of HA and not something I can change in the integration.

closed 2024-02-24 by wlcrs