wlcrs/huawei_solar #616: [Bug]: doesnt work wit actual HA 2024.2.0 and 2024.2.1

ottosax, 2024-02-09:
### Describe the issue

Hello,

after update from HA 2024.1.16 the Huawei integration doesnt work wit actual HA 2024.2.0 and also not with 2024.2.1

### Bescribe your Huawei Solar Setup

Inverter Type: Sun 2000 8KTL M1 and 2x Sun 2000 10 KLT M1
Inverter Firmware version: actual official latest
SDongle present: yes and actual official latest
Power meter present: three phase 
Battery: LUNA2000 15kWh
Battery Firmware version: actual official latest

### How do you connect to the inverter?

Please select your connection method

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.

### Upload your relevant debug logs

```text
huawei_solar.exceptions.ReadException: Failed to read register 43006 after 6 tries
2024-02-09 13:32:13.287 DEBUG (MainThread) [pymodbus.logging] Connecting to 192.168.178.34:502.
2024-02-09 13:32:13.288 DEBUG (MainThread) [pymodbus.logging] Connecting comm
2024-02-09 13:32:13.289 DEBUG (MainThread) [pymodbus.logging] Connected to comm
2024-02-09 13:32:13.290 DEBUG (MainThread) [pymodbus.logging] callback_connected called
2024-02-09 13:32:13.290 DEBUG (MainThread) [huawei_solar.modbus] Waiting for 1500 milliseconds after connection before performing operations
2024-02-09 13:32:13.290 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 43006 with length 1 from slave 17
2024-02-09 13:32:13.290 DEBUG (MainThread) [pymodbus.logging] send: 0x0 0x1 0x0 0x0 0x0 0x6 0x11 0x3 0xa7 0xfe 0x0 0x1
2024-02-09 13:32:13.290 DEBUG (MainThread) [pymodbus.logging] Adding transaction 1
2024-02-09 13:32:13.292 DEBUG (MainThread) [pymodbus.logging] -> transport: received eof
2024-02-09 13:32:13.292 DEBUG (MainThread) [pymodbus.logging] Connection lost comm due to None
2024-02-09 13:32:13.292 DEBUG (MainThread) [pymodbus.logging] Getting transaction 1
2024-02-09 13:32:13.292 DEBUG (MainThread) [pymodbus.logging] Wait comm 1000000 ms before reconnecting.
2024-02-09 13:32:13.293 ERROR (MainThread) [huawei_solar.huawei_solar] Could not read register value, has another device interrupted the connection?
2024-02-09 13:32:13.293 INFO (MainThread) [backoff] Backing off _do_read(...) for 1.0s (huawei_solar.exceptions.ConnectionInterruptedException: Could not read register value, has another device interrupted the connection?)
2024-02-09 13:32:13.293 DEBUG (MainThread) [huawei_solar.huawei_solar] Received <class 'huawei_solar.exceptions.ConnectionInterruptedException'>: backing off reading for 1.0 seconds after 1 tries
2024-02-09 13:32:14.301 ERROR (MainThread) [huawei_solar.huawei_solar] Modbus client is not connected to the inverter.
NoneType: None
2024-02-09 13:32:14.302 INFO (MainThread) [backoff] Backing off _do_read(...) for 2.0s (huawei_solar.exceptions.ConnectionInterruptedException: Modbus client is not connected to the inverter.)
2024-02-09 13:32:14.302 DEBUG (MainThread) [huawei_solar.huawei_solar] Received <class 'huawei_solar.exceptions.ConnectionInterruptedException'>: backing off reading for 2.0 seconds after 2 tries
2024-02-09 13:32:16.305 ERROR (MainThread) [huawei_solar.huawei_solar] Modbus client is not connected to the inverter.
NoneType: None
2024-02-09 13:32:16.306 INFO (MainThread) [backoff] Backing off _do_read(...) for 4.0s (huawei_solar.exceptions.ConnectionInterruptedException: Modbus client is not connected to the inverter.)
2024-02-09 13:32:16.306 DEBUG (MainThread) [huawei_solar.huawei_solar] Received <class 'huawei_solar.exceptions.ConnectionInterruptedException'>: reconnecting and backing off reading for 4.0 seconds after 3 tries
2024-02-09 13:32:16.306 DEBUG (MainThread) [pymodbus.logging] Connecting to 192.168.178.34:502.
2024-02-09 13:32:16.306 DEBUG (MainThread) [pymodbus.logging] Connecting comm
2024-02-09 13:32:16.308 DEBUG (MainThread) [pymodbus.logging] Connected to comm
2024-02-09 13:32:16.308 DEBUG (MainThread) [pymodbus.logging] callback_connected called
2024-02-09 13:32:16.308 DEBUG (MainThread) [huawei_solar.modbus] Waiting for 1500 milliseconds after connection before performing operations
2024-02-09 13:32:16.312 DEBUG (MainThread) [pymodbus.logging] -> transport: received eof
2024-02-09 13:32:16.312 DEBUG (MainThread) [pymodbus.logging] Connection lost comm due to None
2024-02-09 13:32:16.312 DEBUG (MainThread) [pymodbus.logging] Wait comm 1000000 ms before reconnecting.
2024-02-09 13:32:20.309 ERROR (MainThread) [huawei_solar.huawei_solar] Modbus client is not connected to the inverter.
NoneType: None
2024-02-09 13:32:20.310 INFO (MainThread) [backoff] Backing off _do_read(...) for 8.0s (huawei_solar.exceptions.ConnectionInterruptedException: Modbus client is not connected to the inverter.)
2024-02-09 13:32:20.310 DEBUG (MainThread) [huawei_solar.huawei_solar] Received <class 'huawei_solar.exceptions.ConnectionInterruptedException'>: backing off reading for 8.0 seconds after 4 tries
2024-02-09 13:32:28.317 ERROR (MainThread) [huawei_solar.huawei_solar] Modbus client is not connected to the inverter.
NoneType: None
2024-02-09 13:32:28.317 INFO (MainThread) [backoff] Backing off _do_read(...) for 16.0s (huawei_solar.exceptions.ConnectionInterruptedException: Modbus client is not connected to the inverter.)
2024-02-09 13:32:28.317 DEBUG (MainThread) [huawei_solar.huawei_solar] Received <class 'huawei_solar.exceptions.ConnectionInterruptedException'>: backing off reading for 16.0 seconds after 5 tries
2024-02-09 13:32:44.318 ERROR (MainThread) [huawei_solar.huawei_solar] Modbus client is not connected to the inverter.
NoneType: None
2024-02-09 13:32:44.318 ERROR (MainThread) [backoff] Giving up _do_read(...) after 6 tries (huawei_solar.exceptions.ConnectionInterruptedException: Modbus client is not connected to the inverter.)
2024-02-09 13:32:44.319 ERROR (MainThread) [huawei_solar.huawei_solar] Aborting client creation due to error.
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/backoff/_async.py", line 151, in retry
    ret = await target(*args, **kwargs)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/backoff/_async.py", line 151, in retry
    ret = await target(*args, **kwargs)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 349, in _do_read
    raise ConnectionInterruptedException(message)
huawei_solar.exceptions.ConnectionInterruptedException: Modbus client is not connected to the inverter.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 178, in create
    await huawei_solar._initialize()
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 107, in _initialize
    self.time_zone = (await self.get(rn.TIME_ZONE)).value
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 241, in get
    return (await self.get_multiple([name], slave))[0]
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 276, in get_multiple
    response = await self._read_registers(registers[0].register, total_length, slave)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 398, in _read_registers
    return await _do_read()
           ^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/backoff/_async.py", line 159, in retry
    await _call_handlers(on_giveup, **details, exception=e)
  File "/usr/local/lib/python3.12/site-packages/backoff/_async.py", line 37, in _call_handlers
    await handler(details)
  File "/usr/local/lib/python3.12/site-packages/backoff/_async.py", line 16, in f
    return coro_or_func(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 327, in backoff_giveup
    raise ReadException(f"Failed to read register {register} after {details['tries']} tries")
huawei_solar.exceptions.ReadException: Failed to read register 43006 after 6 tries
```

### Please confirm the following:

- [ ] I'm running the latest release of Home Assistant.
- [ ] I'm running the latest release of this integration.
- [ ] I did not find an existing issue describing this problem.
- [ ] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [ ] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

ottosax, 2024-02-09:
I checked the log, its every time I try the same error:
huawei_solar.exceptions.ReadException: Failed to read register 43006 after 6 tries

Bevor up to HA 2024.1.16 it runs without any trouble.

j2g2com, 2024-02-09:
Same problem but connecting : Sun 2000 5KTL L1 and SmartPower DDSU666-H with Wi-Fi

baybod, 2024-02-09:
same here, no connection at all

RidersMX, 2024-02-09:
I got the same issues after updating from HA 2024.1.6 to 2024.2.1: the Huawei integration doesn't work
Unfortunatly HA 2024.1.6 from my Backup doesn't work for me now either. The integration doesn't work here to anymore.

Inverter Type: SUN2000-6KTL-M1
Inverter Firmware version: V100R001C00SPC158
SDongle present: yes, V200R022C10SPC108
Power meter present: yes, DTSU666-H
Battery: LUNA2000 5kWh
Battery Firmware version: V100R002C00SPC123

Aborting client creation due to error.
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/backoff/_async.py", line 151, in retry
    ret = await target(*args, **kwargs)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/backoff/_async.py", line 151, in retry
    ret = await target(*args, **kwargs)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 349, in _do_read
    raise ConnectionInterruptedException(message)
huawei_solar.exceptions.ConnectionInterruptedException: Modbus client is not connected to the inverter.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 178, in create
    await huawei_solar._initialize()
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 107, in _initialize
    self.time_zone = (await self.get(rn.TIME_ZONE)).value
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 241, in get
    return (await self.get_multiple([name], slave))[0]
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 276, in get_multiple
    response = await self._read_registers(registers[0].register, total_length, slave)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 398, in _read_registers
    return await _do_read()
           ^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/backoff/_async.py", line 159, in retry
    await _call_handlers(on_giveup, **details, exception=e)
  File "/usr/local/lib/python3.11/site-packages/backoff/_async.py", line 37, in _call_handlers
    await handler(details)
  File "/usr/local/lib/python3.11/site-packages/backoff/_async.py", line 16, in f
    return coro_or_func(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 327, in backoff_giveup
    raise ReadException(f"Failed to read register {register} after {details['tries']} tries")
huawei_solar.exceptions.ReadException: Failed to read register 43006 after 6 tries

Roving-Ronin, 2024-02-10:
No issues with this integration running upon 2024.1.6, 2024.2.0 or 2024.2.1 on either of 2 environments (5,000km apart):
____________________________________________________

**HA instance 1:**

Core: 2024.2.1
Supervisor : 2024.01.1
Operating System: 11.5
Frontend: 20240207.1

Inverter Type: 2 x SUN2000-6KTL-M1
Inverter Firmware version: V100R001C00SPC155  (iirc SPC158 just released, but no worthwhile changes since 155).
SDongle present: yes, V200R022C10SPC108
Power meter present: yes, DTSU666-H
Battery: NIL.
____________________________________________________

**HA instance 2:**

Core: 2024.2.0
Supervisor : 2024.01.1
Operating System: 11.5
Frontend: 20240207.0

Inverter Type: 2 x SUN2000-5KTL-L1
Inverter Firmware version: V200R001C00SPC135
SDongle present: yes, V200R022C10SPC108
Power meter present: yes, DDSU666-H
Battery: LUNA2000 15kWh
Battery Firmware version: V100R002C00SPC123

RidersMX, 2024-02-10:
> I got the same issues after updating from HA 2024.1.6 to 2024.2.1: the Huawei integration doesn't work
> 
> Inverter Type: SUN2000-6KTL-M1 Inverter Firmware version: V100R001C00SPC158 SDongle present: yes, V200R022C10SPC108 Power meter present: yes, DTSU666-H Battery: LUNA2000 5kWh Battery Firmware version: V100R002C00SPC123
> 
> Aborting client creation due to error. Traceback (most recent call last): File "/usr/local/lib/python3.11/site-packages/backoff/_async.py", line 151, in retry ret = await target(*args, **kwargs) ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/usr/local/lib/python3.11/site-packages/backoff/_async.py", line 151, in retry ret = await target(*args, **kwargs) ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 349, in _do_read raise ConnectionInterruptedException(message) huawei_solar.exceptions.ConnectionInterruptedException: Modbus client is not connected to the inverter.
> 
> During handling of the above exception, another exception occurred:
> 
> Traceback (most recent call last): File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 178, in create await huawei_solar._initialize() File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 107, in _initialize self.time_zone = (await self.get(rn.TIME_ZONE)).value ^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 241, in get return (await self.get_multiple([name], slave))[0] ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 276, in get_multiple response = await self._read_registers(registers[0].register, total_length, slave) ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 398, in _read_registers return await _do_read() ^^^^^^^^^^^^^^^^ File "/usr/local/lib/python3.11/site-packages/backoff/_async.py", line 159, in retry await _call_handlers(on_giveup, **details, exception=e) File "/usr/local/lib/python3.11/site-packages/backoff/_async.py", line 37, in _call_handlers await handler(details) File "/usr/local/lib/python3.11/site-packages/backoff/_async.py", line 16, in f return coro_or_func(*args, **kwargs) ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/usr/local/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 327, in backoff_giveup raise ReadException(f"Failed to read register {register} after {details['tries']} tries") huawei_solar.exceptions.ReadException: Failed to read register 43006 after 6 tries

This morning I tried again the Upgrade on 24.2.1 and it works fine. Didn't know exactly why.. But actually, sitll running as expected.

virtualzone, 2024-02-10:
I had issues with this integration after updating Home Assistant to 2024.2.1. After restarting Home Assistant (without changing anything else), the integration started working again.

j2g2com, 2024-02-10:
Correct after rebooting and waiting a while it worked again. Let's see with the new version.

baybod, 2024-02-12:
Still the same issue here, it's not starting, constantly initializing.., rebooting does not solve the issue

danise76, 2024-02-13:
Same problem.

HA Information - Versions:
    Core  2024.2.1
    Supervisor  2024.01.1
    Operating System 11.5

Huawei Solar version:   "1.3.1"

It work for a while after rebooting, and many times with a reload to the interegration goes back to work for a while until it stops.

```
2024-02-13 15:41:43.614 ERROR (MainThread) [custom_components.huawei_solar] Unexpected error fetching 1020AXXXXXXX_optimizer_data_update_coordinator data: 'NoneType' object has no attribute 'close'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 313, in _async_refresh
    self.data = await self._async_update_data()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/huawei_solar/__init__.py", line 371, in _async_update_data
    return await self.bridge.get_latest_optimizer_history_data()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/bridge.py", line 255, in get_latest_optimizer_history_data
    end_time = (await self.client.get(rn.SYSTEM_TIME_RAW, self.slave_id)).value
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 241, in get
    return (await self.get_multiple([name], slave))[0]
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 276, in get_multiple
    response = await self._read_registers(registers[0].register, total_length, slave)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 398, in _read_registers
    return await _do_read()
           ^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/backoff/_async.py", line 151, in retry
    ret = await target(*args, **kwargs)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/backoff/_async.py", line 151, in retry
    ret = await target(*args, **kwargs)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 351, in _do_read
    response = await self._client.read_holding_registers(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pymodbus/client/base.py", line 206, in async_execute
    self.close(reconnect=True)
  File "/usr/local/lib/python3.12/site-packages/pymodbus/client/base.py", line 153, in close
    self.connection_lost(asyncio.TimeoutError("Server not responding"))
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/modbus.py", line 40, in connection_lost
    super().connection_lost(exc)
  File "/usr/local/lib/python3.12/site-packages/pymodbus/transport/transport.py", line 300, in connection_lost
    self.transport_close(intern=True)
  File "/usr/local/lib/python3.12/site-packages/pymodbus/transport/transport.py", line 419, in transport_close
    self.transport.close()
    ^^^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'close'

2024-02-13 15:41:48.622 ERROR (MainThread) [huawei_solar.huawei_solar] Timeout while waiting for connection. Reconnecting...
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/asyncio/tasks.py", line 520, in wait_for
    return await fut
           ^^^^^^^^^
  File "/usr/local/lib/python3.12/asyncio/locks.py", line 212, in wait
    await fut
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 141, in _communication_lock
    await asyncio.wait_for(self._client.connected_event.wait(), WAIT_FOR_CONNECTION_TIMEOUT)
  File "/usr/local/lib/python3.12/asyncio/tasks.py", line 519, in wait_for
    async with timeouts.timeout(timeout):
[... 220 lines cut ...]
2024-02-13 16:05:36.255 DEBUG (MainThread) [pymodbus.logging] Frame - not ready
2024-02-13 16:05:36.291 DEBUG (MainThread) [pymodbus.logging] recv: 0x7c 0x0 0x0 0x3 0x47 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x18 0xea 0x5a 0x0 0x19 0x86 0x62 0x65 old_data:  addr=None
2024-02-13 16:05:36.291 DEBUG (MainThread) [pymodbus.logging] Processing: 0x7c 0x0 0x0 0x3 0x47 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x18 0xea 0x5a 0x0 0x19 0x86 0x62 0x65
2024-02-13 16:05:36.291 DEBUG (MainThread) [pymodbus.logging] Frame - not ready
2024-02-13 16:05:36.309 DEBUG (MainThread) [pymodbus.logging] recv: 0xcb 0x93 0x47 0x0 0x0 0x0 0x7 0x0 0x0 0x2 0x3e 0x8a 0x50 old_data:  addr=None
2024-02-13 16:05:36.309 DEBUG (MainThread) [pymodbus.logging] Processing: 0xcb 0x93 0x47 0x0 0x0 0x0 0x7 0x0 0x0 0x2 0x3e 0x8a 0x50
2024-02-13 16:05:36.309 DEBUG (MainThread) [pymodbus.logging] Frame - not ready
2024-02-13 16:05:38.603 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064 with length 52 from slave 2
2024-02-13 16:05:38.603 DEBUG (MainThread) [pymodbus.logging] send: 0x2 0x3 0x7d 0x40 0x0 0x34 0x5d 0x96
2024-02-13 16:05:38.604 DEBUG (MainThread) [pymodbus.logging] Adding transaction 2
[... cut ...]
2024-02-13 16:05:38.781 DEBUG (MainThread) [pymodbus.logging] recv: 0xcb 0x93 0x4f 0x0 0x0 0x0 0x4 0x0 0x0 0x1 0x14 0xee 0x62 old_data:  addr=None
2024-02-13 16:05:38.781 DEBUG (MainThread) [pymodbus.logging] Processing: 0xcb 0x93 0x4f 0x0 0x0 0x0 0x4 0x0 0x0 0x1 0x14 0xee 0x62
2024-02-13 16:05:38.781 DEBUG (MainThread) [pymodbus.logging] Frame - not ready
2024-02-13 16:05:44.243 DEBUG (MainThread) [custom_components.huawei_solar] Finished fetching HV225XXXXXXXXXX_data_update_coordinator data in 29.001 seconds (success: False)

```

huineng, 2024-02-13:
same here, it stopped working after the upgrade
Core 2024.2.1
Supervisor 2024.01.1
Operating System 11.5

timconinx, 2024-02-16:
Same problem where the integration suddenly stopped working... My fusionsolar app feels new, did anything change on the huawei side?

I can confirm however that the problem is not restricted to 2024.2.x: just downgraded to 2024.1.5 and got
```
Logger: huawei_solar.huawei_solar
Source: /srv/homeassistant/lib/python3.11/site-packages/huawei_solar/huawei_solar.py:184
First occurred: 6:36:45 PM (2 occurrences)
Last logged: 6:37:31 PM

Aborting client creation due to error.
Traceback (most recent call last):
  File "/usr/lib/python3.11/asyncio/tasks.py", line 490, in wait_for
    return fut.result()
           ^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/locks.py", line 213, in wait
    await fut
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/srv/homeassistant/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 178, in create
    await huawei_solar._initialize()
  File "/srv/homeassistant/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 107, in _initialize
    self.time_zone = (await self.get(rn.TIME_ZONE)).value
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/srv/homeassistant/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 241, in get
    return (await self.get_multiple([name], slave))[0]
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/srv/homeassistant/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 276, in get_multiple
    response = await self._read_registers(registers[0].register, total_length, slave)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/srv/homeassistant/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 396, in _read_registers
    async with self._communication_lock():
  File "/usr/lib/python3.11/contextlib.py", line 204, in __aenter__
    return await anext(self.gen)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/srv/homeassistant/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 145, in _communication_lock
    raise err
  File "/srv/homeassistant/lib/python3.11/site-packages/huawei_solar/huawei_solar.py", line 141, in _communication_lock
    await asyncio.wait_for(self._client.connected_event.wait(), WAIT_FOR_CONNECTION_TIMEOUT)
  File "/usr/lib/python3.11/asyncio/tasks.py", line 492, in wait_for
    raise exceptions.TimeoutError() from exc
TimeoutError
```

Jack77777777, 2024-02-16:
Can´t get the integration start up with 2024.2.2.
Hopefully a fix and new release will be available soon.

Thanks @wlcrs

Jack77777777, 2024-02-17:
Today it is working without changing anything.

ottosax, 2024-02-17:
Here the same, yesterday I tried a new test with with 2024.2.1 (!) and that Update worked, now I will go an test 2024.2.2 ... cross the fingers, I hope this worked to  ;-)

mariusnauj, 2024-02-17:
Hello,

For me the same errors with 2024.2.2v. After HA reboot it tooks for a while to show inverter data, and alwaus in logs get these errors:

Logger: huawei_solar.huawei_solar
Source: /usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py:184
First occurred: 14:38:43 (1 occurrences)
Last logged: 14:38:43

Aborting client creation due to error.
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/asyncio/tasks.py", line 520, in wait_for
    return await fut
           ^^^^^^^^^
  File "/usr/local/lib/python3.12/asyncio/locks.py", line 212, in wait
    await fut
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 178, in create
    await huawei_solar._initialize()
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 107, in _initialize
    self.time_zone = (await self.get(rn.TIME_ZONE)).value
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 241, in get
    return (await self.get_multiple([name], slave))[0]
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 276, in get_multiple
    response = await self._read_registers(registers[0].register, total_length, slave)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 396, in _read_registers
    async with self._communication_lock():
  File "/usr/local/lib/python3.12/contextlib.py", line 210, in __aenter__
    return await anext(self.gen)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 145, in _communication_lock
    raise err
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 141, in _communication_lock
    await asyncio.wait_for(self._client.connected_event.wait(), WAIT_FOR_CONNECTION_TIMEOUT)
  File "/usr/local/lib/python3.12/asyncio/tasks.py", line 519, in wait_for
    async with timeouts.timeout(timeout):
  File "/usr/local/lib/python3.12/asyncio/timeouts.py", line 115, in __aexit__
    raise TimeoutError from exc_val
TimeoutError

Logger: huawei_solar.huawei_solar
Source: /usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py:143
First occurred: 14:38:43 (1 occurrences)
Last logged: 14:38:43

Timeout while waiting for connection. Reconnecting...
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/asyncio/tasks.py", line 520, in wait_for
    return await fut
           ^^^^^^^^^
  File "/usr/local/lib/python3.12/asyncio/locks.py", line 212, in wait
    await fut
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 141, in _communication_lock
    await asyncio.wait_for(self._client.connected_event.wait(), WAIT_FOR_CONNECTION_TIMEOUT)
  File "/usr/local/lib/python3.12/asyncio/tasks.py", line 519, in wait_for
    async with timeouts.timeout(timeout):
  File "/usr/local/lib/python3.12/asyncio/timeouts.py", line 115, in __aexit__
    raise TimeoutError from exc_val
TimeoutError

huineng, 2024-02-17:
i got 3 applications that failed to load when upgrading to 2.2. today all of a sudden without doing anything, it all worked again.

timconinx, 2024-02-17:
No luck for me: I upgraded homeassistant's python to 3.12 and got slightly different errors, but still 

```
Logger: homeassistant.config_entries
Source: config_entries.py:444
First occurred: 3:46:37 PM (1 occurrences)
Last logged: 3:46:37 PM

Error setting up entry SUN2000-4KTL-M1SUN2000-5KTL-M for huawei_solar
Traceback (most recent call last):
  File "/srv/homeassistant/lib/python3.12/site-packages/homeassistant/config_entries.py", line 444, in async_setup
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/homeassistant/.homeassistant/custom_components/huawei_solar/__init__.py", line 205, in async_setup_entry
    raise err
  File "/home/homeassistant/.homeassistant/custom_components/huawei_solar/__init__.py", line 92, in async_setup_entry
    primary_bridge = await HuaweiSolarBridge.create(
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/srv/homeassistant/lib/python3.12/site-packages/huawei_solar/bridge.py", line 69, in create
    await HuaweiSolarBridge.__populate_fields(bridge)
  File "/srv/homeassistant/lib/python3.12/site-packages/huawei_solar/bridge.py", line 127, in __populate_fields
    bridge.battery_1_type = (await bridge.client.get(rn.STORAGE_UNIT_1_PRODUCT_MODEL, bridge.slave_id)).value
                             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/srv/homeassistant/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 241, in get
    return (await self.get_multiple([name], slave))[0]
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/srv/homeassistant/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 276, in get_multiple
    response = await self._read_registers(registers[0].register, total_length, slave)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/srv/homeassistant/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 398, in _read_registers
    return await _do_read()
           ^^^^^^^^^^^^^^^^
  File "/srv/homeassistant/lib/python3.12/site-packages/backoff/_async.py", line 151, in retry
    ret = await target(*args, **kwargs)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/srv/homeassistant/lib/python3.12/site-packages/backoff/_async.py", line 151, in retry
    ret = await target(*args, **kwargs)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/srv/homeassistant/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 351, in _do_read
    response = await self._client.read_holding_registers(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/srv/homeassistant/lib/python3.12/site-packages/pymodbus/client/base.py", line 207, in async_execute
    raise ModbusIOException(
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] ERROR: No response received after 3 retries
```

mariusnauj, 2024-02-17:
And always appearing this error, integration working but error appearing:

Logger: huawei_solar.huawei_solar
Source: /usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py:348
First occurred: 14:56:18 (57 occurrences)
Last logged: 20:04:27

Modbus client is not connected to the inverter.
NoneType: None

timconinx, 2024-02-18:
The plot thickens: I tried removing the integration, and readding it... which didn't work until I de-checked the "elevated permissions" box. After that I got the message that the integration was added succesfully, _and_ a new error (the integration still doesn't work)
```
Logger: homeassistant.config_entries
Source: config_entries.py:444
First occurred: 7:42:41 PM (1 occurrences)
Last logged: 7:42:41 PM

Error setting up entry SUN2000-4KTL-M1SUN2000-5KTL-M for huawei_solar
Traceback (most recent call last):
  File "/srv/homeassistant/lib/python3.12/site-packages/homeassistant/config_entries.py", line 444, in async_setup
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/homeassistant/.homeassistant/custom_components/huawei_solar/__init__.py", line 205, in async_setup_entry
    raise err
  File "/home/homeassistant/.homeassistant/custom_components/huawei_solar/__init__.py", line 92, in async_setup_entry
    primary_bridge = await HuaweiSolarBridge.create(
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/srv/homeassistant/lib/python3.12/site-packages/huawei_solar/bridge.py", line 69, in create
    await HuaweiSolarBridge.__populate_fields(bridge)
  File "/srv/homeassistant/lib/python3.12/site-packages/huawei_solar/bridge.py", line 151, in __populate_fields
    await bridge.client.get(rn.METER_STATUS, bridge.slave_id)
  File "/srv/homeassistant/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 241, in get
    return (await self.get_multiple([name], slave))[0]
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/srv/homeassistant/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 276, in get_multiple
    response = await self._read_registers(registers[0].register, total_length, slave)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/srv/homeassistant/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 398, in _read_registers
    return await _do_read()
           ^^^^^^^^^^^^^^^^
  File "/srv/homeassistant/lib/python3.12/site-packages/backoff/_async.py", line 151, in retry
    ret = await target(*args, **kwargs)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/srv/homeassistant/lib/python3.12/site-packages/backoff/_async.py", line 151, in retry
    ret = await target(*args, **kwargs)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/srv/homeassistant/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 351, in _do_read
    response = await self._client.read_holding_registers(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/srv/homeassistant/lib/python3.12/site-packages/pymodbus/client/base.py", line 162, in async_execute
    self.transport_send(packet)
  File "/srv/homeassistant/lib/python3.12/site-packages/pymodbus/transport/transport.py", line 397, in transport_send
    self.transport.write(data)  # type: ignore[attr-defined]
    ^^^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'write'
```

markusgaller1978, 2024-02-19:
Hi,
I have the same issue with the update HA Update Core 2024.2.1 and HA Update Core 2024.2.2.
With HA Core 2024.1.6 the huawei integration is working nice.

Its report the issue that the integration can't create the HuaweiSolarBridge.

baybod, 2024-02-19:
3 days ago it worked fpr one day, than it stopped working again. No connection at all....

timconinx, 2024-02-19:
You can remove and add the integration without losing your history (if you didn't rename the entities, I think). This way you can try checking/unchecking the 'elevated privileges' switch.

If connecting via network, this piece of the documentation helped me a lot:

> The port is either 502 or 6607.
When connecting to the inverter AP the host IP is typically 192.168.200.1 and the slave id is typically 0.
When connecting to an SDongle, the slave id is typically 1. Make sure to give this device a fixed IP!

I used to connect through the inverter AP port 6607, now I connect through the SDongle (fixed ip in your network) port 502. With elevated privileges but without having the provide the installer password... which I think makes sense because I can change all of the "protected" configuration parameters through the huawei fusionsolar site as well these days.

sLesage, 2024-02-19:
> You can remove and add the integration without losing your history (if you didn't rename the entities, I think). This way you can try checking/unchecking the 'elevated privileges' switch.

I think that actually might have worked. The integration isn't showing any errors for now, but I do see a lot of entities marked as 'Disabled by Integration'. I'll have another look at it later on.

baybod, 2024-02-21:
just a remark, what I do not get is that it sometimes work, and after reboot it does not work, Then you reboot serveral times and it works again, 
Now it is again not working at all....

wlcrs, 2024-02-22:
This issue seems to be caused by a connection failure or an unstable connection to the inverter. This integration already has already implemented a lot of logic to automatically recover from connection failures or unstable connections. The three most important ones are:
- automatic reconnects if the connection to the inverter failed;
- each Modbus request is retried for up to 6 times, with increasing waits between each attempt to allow the connection to recover;
- after 3 failed Modbus requests the connection is reset to fix hanging connections.

Please understand that I cannot help you solve your connection failure or unstable connection. Consider [an alternative connection method](https://github.com/wlcrs/huawei_solar/wiki/Connecting-to-the-inverter).
If you are convinced that there is no connectivity issue, then I invite you to debug and fix the problem yourself and submit a pull request to the underlying [huawei-solar Python library](https://gitlab.com/Emilv2/huawei-solar). Your input would be greatly appreciated.

As there is nothing actionable for me to do in this issue, I'm closing it. You are welcome to open a topic in [Discussions: Connecting to the inverter](https://github.com/wlcrs/huawei_solar/discussions/categories/connecting-to-the-inverter) to get community support for your problem.

closed 2024-02-22 by wlcrs