wlcrs/huawei_solar #618: [Bug]: change of units of measure - deprecated constant

forgo111, 2024-02-10:
### Describe the issue

Logger: homeassistant.const
Source: helpers/deprecation.py:205
First occurred: 16:52:34 (3 occurrences)
Last logged: 16:52:34

ENERGY_KILO_WATT_HOUR was used from huawei_solar, this is a deprecated constant which will be removed in HA Core 2025.1. Use UnitOfEnergy.KILO_WATT_HOUR instead, please create a bug report at https://github.com/wlcrs/huawei_solar/issues
POWER_WATT was used from huawei_solar, this is a deprecated constant which will be removed in HA Core 2025.1. Use UnitOfPower.WATT instead, please create a bug report at https://github.com/wlcrs/huawei_solar/issues

### Bescribe your Huawei Solar Setup

Inverter Type: SUN2000-5KTL-M1GIN300-5KTL-M1
Inverter Firmware version: V100R001C00SPC158
SDongle present: yes
Power meter present: three phase 
Battery:  / None

### How do you connect to the inverter?

Via the SDongle, wired connection

### Upload your Diagnostics File

[config_entry-huawei_solar-7a8d809078c00388a3073e8afedf35b2.json](https://github.com/wlcrs/huawei_solar/files/14229760/config_entry-huawei_solar-7a8d809078c00388a3073e8afedf35b2.json)
Drag & Drop your Diagnostics File here.

### Upload your relevant debug logs

```text
debugging not making sense, right?
```

### Please confirm the following:

- [X] I'm running the latest release of Home Assistant.
- [X] I'm running the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

forgo111, 2024-02-10:
debugging:
2024-02-10 16:51:53.996 WARNING (SyncWorker_4) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-02-10 16:51:53.997 WARNING (SyncWorker_4) [homeassistant.loader] We found a custom integration huawei_solar which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-02-10 16:52:07.301 WARNING (MainThread) [homeassistant.setup] Setup of hassio is taking over 10 seconds.
2024-02-10 16:52:07.715 WARNING (MainThread) [homeassistant.setup] Setup of bluetooth is taking over 10 seconds.
2024-02-10 16:52:15.241 WARNING (SyncWorker_5) [homeassistant.components.rpi_power.binary_sensor] Under-voltage was detected. Consider getting a uninterruptible power supply for your Raspberry Pi.
2024-02-10 16:52:34.848 WARNING (MainThread) [homeassistant.const] ENERGY_KILO_WATT_HOUR was used from huawei_solar, this is a deprecated constant which will be removed in HA Core 2025.1. Use UnitOfEnergy.KILO_WATT_HOUR instead, please create a bug report at https://github.com/wlcrs/huawei_solar/issues
2024-02-10 16:52:34.855 WARNING (MainThread) [homeassistant.const] POWER_WATT was used from huawei_solar, this is a deprecated constant which will be removed in HA Core 2025.1. Use UnitOfPower.WATT instead, please create a bug report at https://github.com/wlcrs/huawei_solar/issues
2024-02-10 16:52:34.912 WARNING (MainThread) [homeassistant.const] POWER_WATT was used from huawei_solar, this is a deprecated constant which will be removed in HA Core 2025.1. Use UnitOfPower.WATT instead, please create a bug report at https://github.com/wlcrs/huawei_solar/issues
2024-02-10 17:06:40.001 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064 with length 52 from slave 1
2024-02-10 17:06:40.002 DEBUG (MainThread) [pymodbus.logging] send: 0x0 0x95 0x0 0x0 0x0 0x6 0x1 0x3 0x7d 0x40 0x0 0x34
2024-02-10 17:06:40.003 DEBUG (MainThread) [pymodbus.logging] Adding transaction 149
2024-02-10 17:06:40.845 DEBUG (MainThread) [pymodbus.logging] recv: 0x0 0x95 0x0 0x0 0x0 0x6b 0x1 0x3 0x68 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x9 0xf0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0xff 0xff 0x0 0x0 0xb 0xb8 0xa0 0x0 0x0 0x0 0x65 0xc7 0x27 0x80 0x65 0xc7 0xa9 0x9 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x2 0xf6 0x7d 0x0 0x3 0xd 0xfd 0x65 0xc7 0xad 0x1f 0x0 0x0 0x0 0x0 0x0 0x0 0x1 0xa1 old_data:  addr=None
2024-02-10 17:06:40.846 DEBUG (MainThread) [pymodbus.logging] Processing: 0x0 0x95 0x0 0x0 0x0 0x6b 0x1 0x3 0x68 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x9 0xf0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0xff 0xff 0x0 0x0 0xb 0xb8 0xa0 0x0 0x0 0x0 0x65 0xc7 0x27 0x80 0x65 0xc7 0xa9 0x9 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x2 0xf6 0x7d 0x0 0x3 0xd 0xfd 0x65 0xc7 0xad 0x1f 0x0 0x0 0x0 0x0 0x0 0x0 0x1 0xa1
2024-02-10 17:06:40.846 DEBUG (MainThread) [pymodbus.logging] Factory Response[ReadHoldingRegistersResponse': 3]
2024-02-10 17:06:40.847 DEBUG (MainThread) [pymodbus.logging] Getting transaction 149
2024-02-10 17:06:40.848 DEBUG (MainThread) [pymodbus.logging] [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 2544, 0, 0, 0, 0, 0, 0, 65535, 0, 3000, 40960, 0, 26055, 10112, 26055, 43273, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 2, 63101, 3, 3581, 26055, 44319, 0, 0, 0, 417]
2024-02-10 17:06:40.848 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b'\x00\x00']
2024-02-10 17:06:40.849 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b'\x00\x00']
2024-02-10 17:06:40.849 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b'\x00\x00']
2024-02-10 17:06:40.850 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b'\x00\x00']
2024-02-10 17:06:40.850 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b'\t\xf0']
2024-02-10 17:06:40.850 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b'\x00\x00']
2024-02-10 17:06:40.851 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b'\x00\x00']
2024-02-10 17:06:40.851 DEBUG (MainThread) [pymodbus.logging] handle: [b'e\xc7', b"'\x80"]
2024-02-10 17:06:40.852 DEBUG (MainThread) [pymodbus.logging] handle: [b'e\xc7', b'\xa9\t']
2024-02-10 17:06:40.852 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x02', b'\xf6}']
2024-02-10 17:06:40.853 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b'\x01\xa1']
2024-02-10 17:06:40.907 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32000 with length 20 from slave 1
2024-02-10 17:06:40.908 DEBUG (MainThread) [pymodbus.logging] send: 0x0 0x96 0x0 0x0 0x0 0x6 0x1 0x3 0x7d 0x0 0x0 0x14
2024-02-10 17:06:40.910 DEBUG (MainThread) [pymodbus.logging] Adding transaction 150
2024-02-10 17:06:41.553 DEBUG (MainThread) [pymodbus.logging] recv: 0x0 0x96 0x0 0x0 0x0 0x2b 0x1 0x3 0x28 0x0 0x1 0x0 0x0 0x0 0x4 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0xa8 0xaf 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 old_data:  addr=None
[... cut ...]
2024-02-10 17:07:15.807 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b'\x0f\x8d']
2024-02-10 17:07:15.808 DEBUG (MainThread) [pymodbus.logging] handle: [b'\xff\xff', b'\xfeM']
2024-02-10 17:07:15.808 DEBUG (MainThread) [pymodbus.logging] handle: [b'\xff\xff', b'\xfec']
2024-02-10 17:07:15.808 DEBUG (MainThread) [pymodbus.logging] handle: [b'\xff\xff', b'\xfdT']
2024-02-10 17:07:15.809 DEBUG (MainThread) [custom_components.huawei_solar] Finished fetching TA2260065924_data_update_coordinator data in 2.808 seconds (success: True)

Roving-Ronin, 2024-02-12:
Please check to see if an issue has already been opened. This one is already logged, see: https://github.com/wlcrs/huawei_solar/issues/583

forgo111, 2024-02-12:
sorry

wlcrs, 2024-02-24:
Duplicate of #583

closed 2024-02-24 by wlcrs