wlcrs/huawei_solar #621: [Bug]: v1.3.1 - Inverter On/Off Switch - Entity ID Name Error

Roving-Ronin, 2024-02-17:
### Describe the issue

Looking at a clean install of  v1.3.1 at a site with 2 x L1 5kW and 1 x 15kW LUNA ESS, checking the Configuration section of the panel and instead of a switch for Inverter On/Off it now shows as 'None'.

Checking the entity ID and they have been created as  'switch.inverter_none'  and  'switch.inverter_none_2'.

In pre v1.3.1 installs these would have been 'switch.inverter_inverter_on_off' and 'switch.inverter_inverter_on_off_2', thinking that the code for creating these entity ID's in the change of naming format in v1.3.1 might have gotten overlooked? Assume the correct naming format that v.1.3.1 meant to introduce would have seen these become 'switch.inverter__on_off' (removing the duplicate 'inverter' from the name) ?

Discussion at: 

Home Assistant: 2024.2.1 (2.2 only released today..so not installed).

[image]

[image]

[image]

### Bescribe your Huawei Solar Setup

Inverter Type: 2 x L1 5KTL in a cascade
Inverter Firmware version: SPC133
SDongle present: yes
Power meter present: three phase
Battery: 1 x LUNA2000 15kWh
Battery Firmware version: latest

### How do you connect to the inverter?

Via the SDongle, wireless connection

### Upload your Diagnostics File

[config_entry-huawei_solar-e7190a773e4143ed923016cd2d3a7e21.json](https://github.com/wlcrs/huawei_solar/files/14317597/config_entry-huawei_solar-e7190a773e4143ed923016cd2d3a7e21.json)

### Upload your relevant debug logs

```text
N/A.
```

### Please confirm the following:

- [X] I'm running the latest release of Home Assistant.
- [X] I'm running the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

Roving-Ronin, 2024-02-17:
Convert to issue didn't work... adding link to original discussion thread: https://github.com/wlcrs/huawei_solar/discussions/585

Roving-Ronin, 2024-02-17:
I would assume its also linked to the v1.3.1 change to the entity ID naming format, as I see that after adding the custom extra sensors for Battery Temperature that the entity ID is 'sensor.battery_battery_1_temperature' and 'sensor.battery_battery_2_temperature'.

Would anyone know where to look to fix this up so these entity ID drop the duplicate 'battery' and following the new naming format of 'sensor.battery_temperature' (for battery 1) and 'sensor.battery_temperature_2' (for battery 2) ?

Thanks.

closed 2024-02-24 by wlcrs

j2g2com, 2024-02-27:
Hello good. One question... with this version can we connect directly via Dongle? I explain. I read the data from the inverter via Wifi, because with the Dongle, although it has a fixed IP and with Huawei cable, it does not allow it to be read... is it possible with this version that it works? Thank you

Roving-Ronin, 2024-02-28:
@j2g2com ??? All the versions of WLCRS work with using the sDongle (Wifi / Fast Ethernet) and connecting via WiFi or Ethernet (even though contrary to expectations connecting via WiFi is more reliable). See the Wiki section on connection methods and configuring, and make sure you have modbus access enabled on the sDongle.

j2g2com, 2024-03-11:
@Roving-Ronin Well, I have connected home assistant by wifi because by ethernet there is no way and as you say it should be the other way around since direct ethernet is more reliable. Thank you —

Roving-Ronin, 2024-03-11:
@j2g2com 
1) You appear to have jumped into a BUG report and be trying to hijack it for a 'how do I connect to my inverter' ?

2) Have you read the Wiki at : https://github.com/wlcrs/huawei_solar/wiki/Connecting-to-the-inverter#via-the-ip-with-which-the-inverter-is-directly-connected-to-your-home-network
that covers having your inverter connected to your home network via the sDongle (that supported WiFi or Fast-Ethernet connection methods).

3) If you can't work out how to do it / are having difficulties, then read the existing posts in the Discussion area, and if nothing is the same as your seeing, then open a new discussion for yourself.

https://github.com/wlcrs/huawei_solar/discussions/categories/connecting-to-the-inverter

commit 98b70704 (2024-02-24): Fix On/Off switch naming

Fixes #621