wlcrs/huawei_solar #624: [Feature Request]: EV charging drains LUNA2000 battery empty.

johanvh83, 2024-02-21:
### Describe your feature request

I own a Huawei converter and LUNA2000 battery. Your integrations works already for months without any problem. Apologies!
Recently I got a company eletrical vehicle (EV).
At daytime the home battery is charged by the solar panels. I want to use the stored energy for my home usage and most preferable when I have to pay the most for the energy, so during day time.
The charging of the EV is managed by the Engie Drive app. It starts the charging when the night - low tariff starts.
In the area where I live in Belgium, it's 21:00.
I realised that the EV drains quickly the home battery empty. As a result, the day after, I have to pay day high tarriff for my personnal home use! I did an investment into the home battery but due to the EV charging I don't get any profit!
Is it possible to 'disable the discharge to the home battery' in order to keep the storaged energy till the morning after?
The charging of my car should come from the grid, not from the home battery. 

### Proper usage

- [X] I confirm that this is not a bug report or support request
- [X] I confirm that this feature request is within the stated scope of the integration

Roving-Ronin, 2024-02-21:
@johanvh83 are you saying you use your LUNA battery during the day, when normally people are running of the solar panel output to power the house and also recharge the battery, because they used the battery overnight (when most countries charge higher prices for electricity) ?

If so, I think it would be best to look at the Huawei manual and refer to the Time of Use (TOU) settings, you can then configure the TOU scheduled (iirc I think you can have 5-6 dirrent time period you can manually define in the Huawei software).

wlcrs, 2024-02-22:
You can set the maximum discharge power to 0 to disable discharging.