wlcrs/huawei_solar #627: [Bug]: Start of huawei_solar - failed after upgrade HA Core 20.24.2.2

markusgaller1978, 2024-02-23:
### Describe the issue

After upgrade the start of integration failed.
The Custom integration could not be loaded by start therefore no debug logs can be taken.

### Bescribe your Huawei Solar Setup

Inverter Type: SUN2000-10KTL-M1
Inverter Firmware version: V100R001C00SPC159
SDongle present: yes 
Power meter present: three phase / single phase / no
Battery: None
Battery Firmware version: ---

### How do you connect to the inverter?

Via the SDongle, wireless connection

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.

### Upload your relevant debug logs

```text
Logger: homeassistant.setup
Source: setup.py:251
First occurred: 7:05:36 AM (1 occurrences)
Last logged: 7:05:36 AM

Setup failed for custom integration 'huawei_solar': Unable to import component: cannot import name 'HuaweiSolarBridge' from partially initialized module 'huawei_solar' (most likely due to a circular import) (/config/huawei_solar/__init__.py)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 251, in _async_setup_component
    component = integration.get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 827, in get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/importlib/__init__.py", line 90, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 994, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/huawei_solar/__init__.py", line 23, in <module>
    from huawei_solar import (
  File "/config/huawei_solar/__init__.py", line 25, in <module>
    from huawei_solar import HuaweiSolarBridge, HuaweiSolarException, InvalidCredentials
ImportError: cannot import name 'HuaweiSolarBridge' from partially initialized module 'huawei_solar' (most likely due to a circular import) (/config/huawei_solar/__init__.py)
```

### Please confirm the following:

- [X] I'm running the latest release of Home Assistant.
- [X] I'm running the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

crypto-kid, 2024-02-24:
I also have this issue

wlcrs, 2024-02-25:
The folder 'huawei_solar' must be located in the sub-folder 'custom_components' in the 'config' directory.

That does not seem to be the case here, causing it to crash.

closed 2024-02-25 by wlcrs