wlcrs/huawei_solar #636: [Bug]: Intergration not loading with 2024.3

yousaf465, 2024-02-29:
### Describe the issue

Huawei_solar integration is not loading with 2024.3b0. 
[image]

### Bescribe your Huawei Solar Setup

10KTLM1 

### How do you connect to the inverter?

Via the `SUN2000-<serial_no>` WiFi

### Upload your Diagnostics File

I can't download it as the integration is not loaded. 

### Upload your relevant debug logs

```text
I can't download it as the integration is not loaded.
```

### Please confirm the following:

- [X] I'm running the latest release of Home Assistant.
- [X] I'm running the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [ ] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [ ] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

yousaf465, 2024-02-29:
https://discord.com/channels/330944238910963714/427516175237382144/1212449004541972567 I think this might be relevant.

yousaf465, 2024-02-29:
```Logger: homeassistant.util.json
Source: util/json.py:78
First occurred: 11:30:05 (2 occurrences)
Last logged: 11:31:51

Could not parse JSON content: /config/custom_components/huawei_solar/translations/ur.json
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/util/json.py", line 78, in load_json
    return orjson.loads(fdesc.read())  # type: ignore[no-any-return]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
orjson.JSONDecodeError: unexpected character: line 595 column 131 (char 19468)

```
another error

adepedroblanes, 2024-02-29:
Me ocurre lo mismo, pero creo que es la wifi modbus del inversor que se desconecta.

The same thing happens to me, but I think it is the inverter's modbus wifi that disconnects.

yousaf465, 2024-03-01:
fixed for Urdu at least.

closed 2024-03-01 by wlcrs