wlcrs/huawei_solar #640: [Bug]: Slow integration startup time

fredck, 2024-03-07:
### Describe the issue

The Huawei Solar integration is at the top of my startup time list, with more than 2 minutes:

[image]

It makes HA restarts a really painful process.

HA fires warnings when an integration goes over 10s to startup. I believe bringing Huawei Solar to under 10s would be a first reasonable goal.

### Bescribe your Huawei Solar Setup

Inverter Type: SUN2000-10KTL-M1SUN2000-12
Inverter Firmware version: ?
SDongle present: yes
Power meter present: three phase
Battery: ?
Battery Firmware version: ?

### How do you connect to the inverter?

Via the SDongle, wired connection

### Upload your Diagnostics File

[config_entry-huawei_solar-ce1909fe29ef85ee125fb1e5486a37e2.json](https://github.com/wlcrs/huawei_solar/files/14525519/config_entry-huawei_solar-ce1909fe29ef85ee125fb1e5486a37e2.json)

### Upload your relevant debug logs

```text
2024-03-07 15:46:55.028 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration huawei_solar which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-03-07 15:46:59.734 INFO (MainThread) [homeassistant.setup] Setting up huawei_solar
2024-03-07 15:46:59.734 INFO (MainThread) [homeassistant.setup] Setup of domain huawei_solar took 0.0 seconds
2024-03-07 15:47:00.261 DEBUG (MainThread) [huawei_solar.modbus] Waiting for 1500 milliseconds after connection before performing operations
2024-03-07 15:47:00.262 INFO (MainThread) [huawei_solar.huawei_solar] Waiting for connection ...
2024-03-07 15:47:01.762 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 43006 with length 1 from slave 1
2024-03-07 15:47:02.977 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 47000 with length 1 from slave 1
2024-03-07 15:47:04.165 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 30000 with length 35 from slave 1
2024-03-07 15:47:05.690 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 30071 with length 1 from slave 1
2024-03-07 15:47:06.841 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37200 with length 1 from slave 1
2024-03-07 15:47:08.293 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 47000 with length 1 from slave 1
2024-03-07 15:47:09.117 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 47089 with length 1 from slave 1
2024-03-07 15:47:10.215 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 47954 with length 1 from slave 1
2024-03-07 15:47:10.955 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37100 with length 1 from slave 1
2024-03-07 15:47:11.427 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37125 with length 1 from slave 1
2024-03-07 15:47:12.439 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064 with length 52 from slave 1
2024-03-07 15:47:13.239 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32000 with length 20 from slave 1
2024-03-07 15:47:14.474 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37201 with length 1 from slave 1
2024-03-07 15:47:15.600 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37100 with length 38 from slave 1
2024-03-07 15:47:16.155 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37760 with length 28 from slave 1
2024-03-07 15:47:16.949 DEBUG (MainThread) [custom_components.huawei_solar] Finished fetching TA2280064304_data_update_coordinator data in 4.561 seconds (success: True)
2024-03-07 15:47:17.000 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 47075 with length 14 from slave 1
2024-03-07 15:47:18.334 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 47200 with length 46 from slave 1
2024-03-07 15:47:21.000 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 47255 with length 45 from slave 1
2024-03-07 15:47:21.736 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 47102 with length 1 from slave 1
2024-03-07 15:47:22.856 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 47954 with length 66 from slave 1
2024-03-07 15:47:23.722 DEBUG (MainThread) [custom_components.huawei_solar] Finished fetching TA2280064304_configuration_update_coordinator data in 6.773 seconds (success: True)
2024-03-07 15:47:23.774 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading file 0x45 from slave 1
2024-03-07 15:47:54.762 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 40000 with length 2 from slave 1
2024-03-07 15:47:55.265 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading file 0x44 from slave 1
2024-03-07 15:47:58.249 WARNING (MainThread) [homeassistant.bootstrap] Waiting on integrations to complete setup: huawei_solar
2024-03-07 15:48:07.474 DEBUG (MainThread) [custom_components.huawei_solar] Finished fetching TA2280064304_optimizer_data_update_coordinator data in 12.764 seconds (success: True)
2024-03-07 15:48:07.477 INFO (MainThread) [homeassistant.components.sensor] Setting up huawei_solar.sensor
2024-03-07 15:48:07.479 INFO (MainThread) [homeassistant.components.number] Setting up huawei_solar.number
2024-03-07 15:48:07.481 INFO (MainThread) [homeassistant.components.switch] Setting up huawei_solar.switch
2024-03-07 15:48:07.483 INFO (MainThread) [homeassistant.components.select] Setting up huawei_solar.select
2024-03-07 15:48:07.527 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37046 with length 2 from slave 1
2024-03-07 15:48:08.833 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064 with length 52 from slave 1
2024-03-07 15:48:10.039 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37048 with length 2 from slave 1
2024-03-07 15:48:10.679 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32000 with length 20 from slave 1
[... 22 lines cut ...]
2024-03-07 15:48:44.905 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 40000 with length 2 from slave 1
2024-03-07 15:48:46.006 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 47255 with length 45 from slave 1
2024-03-07 15:48:46.618 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading file 0x44 from slave 1
2024-03-07 15:48:58.426 DEBUG (MainThread) [custom_components.huawei_solar] Finished fetching TA2280064304_optimizer_data_update_coordinator data in 14.973 seconds (success: True)
2024-03-07 15:48:58.499 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 47102 with length 1 from slave 1
2024-03-07 15:48:59.408 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 47954 with length 66 from slave 1
2024-03-07 15:48:59.972 DEBUG (MainThread) [custom_components.huawei_solar] Finished fetching TA2280064304_configuration_update_coordinator data in 30.669 seconds (success: True)
2024-03-07 15:49:00.361 INFO (MainThread) [homeassistant.components.automation.huawei_solar_inverter_data_polling] Initialized trigger Huawei Solar inverter data polling
2024-03-07 15:49:10.706 INFO (MainThread) [homeassistant.components.automation.huawei_solar_inverter_data_polling] Huawei Solar inverter data polling: Running automation actions
2024-03-07 15:49:10.707 INFO (MainThread) [homeassistant.components.automation.huawei_solar_inverter_data_polling] Huawei Solar inverter data polling: Executing step call service
2024-03-07 15:49:10.758 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32064 with length 52 from slave 1
2024-03-07 15:49:11.391 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 32000 with length 20 from slave 1
2024-03-07 15:49:12.210 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37201 with length 1 from slave 1
2024-03-07 15:49:13.409 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37100 with length 38 from slave 1
2024-03-07 15:49:14.322 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37760 with length 28 from slave 1
2024-03-07 15:49:15.058 DEBUG (MainThread) [custom_components.huawei_solar] Finished fetching TA2280064304_data_update_coordinator data in 4.301 seconds (success: True)
```

### Please confirm the following:

- [X] I'm running the latest release of Home Assistant.
- [X] I'm running the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

robertaccess, 2024-03-07:
you are right mine take 169.20 sec.

wlcrs, 2024-03-12:
The speed is mainly dictated by how fast the inverter replies to the queries from this inverter. This integration does this as quickly and efficiently as possible, mainly by batching requests where possible. I specifically started working on this integration as I was not happy by the inefficiencies of the old integration (https://github.com/Emilv2/huawei_solar). 

As you don't provide anything useful in this bug report, I'm closing it. Please refrain from opening gratuitous reports like these in the future. 

You are of course invited to create a PR with code changes which reduce the startup time.

closed 2024-03-12 by wlcrs

fredck, 2024-03-13:
> As you don't provide anything useful in this bug report, I'm closing it. Please refrain from opening gratuitous reports like these in the future.

What do you mean by that? The usefulness of a bug report is to have a report of a bug. What else is required?

wlcrs, 2024-03-13:
Is it a bug, or is it just a consequence of how the system works? As already explained: it's the latter.

To put it bluntly, this issue lacked critical thinking and thorough investigation. I'm asking you to do better.