wlcrs/huawei_solar #641: [Bug]: failing to load after upgrade to HA Core 2024.3.0

themaninspain, 2024-03-08:
### Describe the issue

Integration fails to load after upgrade to HA 2024.3.0
Cannot download diagnostics as integration is not loaded
Integration was working before upgrade, but after trying to downgrade (using a backup) it still didn't load

### Bescribe your Huawei Solar Setup

Inverter Type: SUN2000-6KTL-L1
Inverter Firmware version: V200R001C00SPC109
SDongle present: no
Power meter present: single phase
Battery: None
Battery Firmware version:

### How do you connect to the inverter?

Via the `SUN2000-<serial_no>` WiFi

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.

### Upload your relevant debug logs

```text
Logger: homeassistant.util.package
Source: util/package.py:122
First occurred: 9:51:12 AM (2 occurrences)
Last logged: 9:57:25 AM

Unable to install package huawei-solar==2.2.9: WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7fc1027ad6d0>: Failed to establish a new connection: [Errno 101] Network unreachable')': /packages/fc/34/6aedede927e2228ced95bcd84caa032968fe6bd8b0c0b6aec28c0a2beac3/huawei_solar-2.2.9-py3-none-any.whl.metadata WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7fc1027ada60>: Failed to establish a new connection: [Errno 101] Network unreachable')': /packages/fc/34/6aedede927e2228ced95bcd84caa032968fe6bd8b0c0b6aec28c0a2beac3/huawei_solar-2.2.9-py3-none-any.whl.metadata WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7fc1027adc10>: Failed to establish a new connection: [Errno 101] Network unreachable')': /packages/fc/34/6aedede927e2228ced95bcd84caa032968fe6bd8b0c0b6aec28c0a2beac3/huawei_solar-2.2.9-py3-none-any.whl.metadata WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7fc1027addf0>: Failed to establish a new connection: [Errno 101] Network unreachable')': /packages/fc/34/6aedede927e2228ced95bcd84caa032968fe6bd8b0c0b6aec28c0a2beac3/huawei_solar-2.2.9-py3-none-any.whl.metadata WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7fc1027adee0>: Failed to establish a new connection: [Errno 101] Network unreachable')': /packages/fc/34/6aedede927e2228ced95bcd84caa032968fe6bd8b0c0b6aec28c0a2beac3/huawei_solar-2.2.9-py3-none-any.whl.metadata ERROR: Could not install packages due to an OSError: HTTPSConnectionPool(host='files.pythonhosted.org', port=443): Max retries exceeded with url: /packages/fc/34/6aedede927e2228ced95bcd84caa032968fe6bd8b0c0b6aec28c0a2beac3/huawei_solar-2.2.9-py3-none-any.whl.metadata (Caused by NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7fc1027ae0f0>: Failed to establish a new connection: [Errno 101] Network unreachable'))
Unable to install package huawei-solar==2.2.9: WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7fca6184d1c0>: Failed to establish a new connection: [Errno 101] Network unreachable')': /packages/fc/34/6aedede927e2228ced95bcd84caa032968fe6bd8b0c0b6aec28c0a2beac3/huawei_solar-2.2.9-py3-none-any.whl.metadata WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7fca6184daf0>: Failed to establish a new connection: [Errno 101] Network unreachable')': /packages/fc/34/6aedede927e2228ced95bcd84caa032968fe6bd8b0c0b6aec28c0a2beac3/huawei_solar-2.2.9-py3-none-any.whl.metadata WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7fca6184ddc0>: Failed to establish a new connection: [Errno 101] Network unreachable')': /packages/fc/34/6aedede927e2228ced95bcd84caa032968fe6bd8b0c0b6aec28c0a2beac3/huawei_solar-2.2.9-py3-none-any.whl.metadata WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7fca6184dee0>: Failed to establish a new connection: [Errno 101] Network unreachable')': /packages/fc/34/6aedede927e2228ced95bcd84caa032968fe6bd8b0c0b6aec28c0a2beac3/huawei_solar-2.2.9-py3-none-any.whl.metadata WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7fca6184e090>: Failed to establish a new connection: [Errno 101] Network unreachable')': /packages/fc/34/6aedede927e2228ced95bcd84caa032968fe6bd8b0c0b6aec28c0a2beac3/huawei_solar-2.2.9-py3-none-any.whl.metadata ERROR: Could not install packages due to an OSError: HTTPSConnectionPool(host='files.pythonhosted.org', port=443): Max retries exceeded with url: /packages/fc/34/6aedede927e2228ced95bcd84caa032968fe6bd8b0c0b6aec28c0a2beac3/huawei_solar-2.2.9-py3-none-any.whl.metadata (Caused by NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7fca6184e270>: Failed to establish a new connection: [Errno 101] Network unreachable'))
```

### Please confirm the following:

- [X] I'm running the latest release of Home Assistant.
- [X] I'm running the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [ ] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [ ] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

dominikandreas, 2024-03-08:
Same here. Logs:

```
Logger: huawei_solar.huawei_solar
Source: /usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py:184
First occurred: 2:47:05 PM (7 occurrences)
Last logged: 2:50:00 PM

Aborting client creation due to error.
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/asyncio/tasks.py", line 520, in wait_for
    return await fut
           ^^^^^^^^^
  File "/usr/local/lib/python3.12/asyncio/locks.py", line 212, in wait
    await fut
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 178, in create
    await huawei_solar._initialize()
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 107, in _initialize
    self.time_zone = (await self.get(rn.TIME_ZONE)).value
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 241, in get
    return (await self.get_multiple([name], slave))[0]
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 276, in get_multiple
    response = await self._read_registers(registers[0].register, total_length, slave)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 396, in _read_registers
    async with self._communication_lock():
  File "/usr/local/lib/python3.12/contextlib.py", line 210, in __aenter__
    return await anext(self.gen)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 145, in _communication_lock
    raise err
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 141, in _communication_lock
    await asyncio.wait_for(self._client.connected_event.wait(), WAIT_FOR_CONNECTION_TIMEOUT)
  File "/usr/local/lib/python3.12/asyncio/tasks.py", line 519, in wait_for
    async with timeouts.timeout(timeout):
  File "/usr/local/lib/python3.12/asyncio/timeouts.py", line 115, in __aexit__
    raise TimeoutError from exc_val
TimeoutError
```

Juanig, 2024-03-09:
Same here, any help?

dominikandreas, 2024-03-09:
I noticed my home assistant was connected to the hotspot of the inverter, but wlan0 had no assigned ip address. I fixed it using `sudo ip link set wlan0 down && sudo ip link set wlan0 up` (rebooting home assistant should have the same effect) and then reloaded the application. Now it's working again

Maybe it would be better to just use the assigned ip address as a static one in the settings of home assistant to prevent this issue in the future.

I also downgraded to the previous version of home assistant core, but not sure if that had anything to do with it.

Juanig, 2024-03-09:
I managed to solve it, for any reason my wifi adapter settings were changed, IPv4 and IPv6 were set to "disabled". Just changed to "Auto", restarted and its fixed!
Thanks.

themaninspain, 2024-03-09:
Unfortunately that is not my case. Network connection is perfect, I can even ping the inverter from my HA system

themaninspain, 2024-03-12:
> /
> This issue seems to be caused by a connection failure or an unstable connection to the inverter. This integration already has already implemented a lot of logic to automatically recover from connection failures or unstable connections. The three most important ones are:
>
> automatic reconnects if the connection to the inverter failed;
> each Modbus request is retried for up to 6 times, with increasing waits between each attempt to allow the connection to recover;
> after 3 failed Modbus requests the connection is reset to fix hanging connections.
> Please understand that I cannot help you solve your connection failure or unstable connection. Consider an alternative connection method <https://github.com/wlcrs/huawei_solar/wiki/Connecting-to-the-inverter>.
> If you are convinced that there is no connectivity issue, then I invite you to debug and fix the problem yourself and submit a pull request to the underlying huawei-solar Python library <https://gitlab.com/Emilv2/huawei-solar>. Your input would be greatly appreciated.
>
> As there is nothing actionable for me to do in this issue, I'm closing it. You are welcome to open a topic in Discussions: Connecting to the inverter <https://github.com/wlcrs/huawei_solar/discussions/categories/connecting-to-the-inverter> to get community support for your problem.

What I am having difficulty understanding is why this has worked perfectly for over 2 years and only stopped working after I upgraded to the latest version of Home Assistant.
The cut off point for readings from the inverter is the exact moment I performed the upgrade.
I have never had any connectivity problems before and the inverter network connection is shown as active and stable (no dropouts or errors)

themaninspain, 2024-03-12:
I finally resolved this problem and need to eat a plate of humble pie.
Turns out that port 6607, which was what i had originally configured was now closed and port 502 was open.
After changing the port in the configuration I am once more connected to the inverter and getting readings.