wlcrs/huawei_solar #654: [Bug]: No optimizers details

techwolfpl, 2024-03-16:
### Describe the issue

No optimizers data. Logs says
Cannot create optimizer sensor entities as the integration has insufficient permissions. Consider enabling elevated permissions to get more optimizer data
But i have checked elevate permissions and provided instalator password

### Bescribe your Huawei Solar Setup

Inverter Type:SUN2000-5KTL-M1GIN300-5KTL-M1
Inverter Firmware version: V100R001C00SPC159
SDongle present: yes - V100R001C00SPC123
Power meter present: bo
Battery: none

### How do you connect to the inverter?

Via the SDongle, wireless connection

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.
[config_entry-huawei_solar-17f074925b12f9f9edffd0e76268cc81.json](https://github.com/wlcrs/huawei_solar/files/14623409/config_entry-huawei_solar-17f074925b12f9f9edffd0e76268cc81.json)

### Upload your relevant debug logs

```text
2024-03-16 11:18:01.041 DEBUG (MainThread) [pymodbus.logging] Getting transaction 16
2024-03-16 11:18:01.042 DEBUG (MainThread) [pymodbus.logging] [6, 0, 7, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 52732, 4141, 137, 0, 0]
2024-03-16 11:18:01.042 DEBUG (MainThread) [pymodbus.logging] handle: [b'\x00\x00', b'\x00\x00']
2024-03-16 11:18:01.093 DEBUG (MainThread) [huawei_solar.huawei_solar] Reading register 37201 with length 1 from slave 1
2024-03-16 11:18:01.093 DEBUG (MainThread) [pymodbus.logging] send: 0x0 0x11 0x0 0x0 0x0 0x6 0x1 0x3 0x91 0x51 0x0 0x1
2024-03-16 11:18:01.093 DEBUG (MainThread) [pymodbus.logging] Adding transaction 17
2024-03-16 11:18:01.127 DEBUG (MainThread) [pymodbus.logging] recv: 0x0 0x11 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0xc old_data:  addr=None
2024-03-16 11:18:01.127 DEBUG (MainThread) [pymodbus.logging] Processing: 0x0 0x11 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0xc
2024-03-16 11:18:01.127 DEBUG (MainThread) [pymodbus.logging] Factory Response[ReadHoldingRegistersResponse': 3]
2024-03-16 11:18:01.127 DEBUG (MainThread) [pymodbus.logging] Getting transaction 17
2024-03-16 11:18:01.128 DEBUG (MainThread) [pymodbus.logging] [12]
2024-03-16 11:18:01.128 DEBUG (MainThread) [custom_components.huawei_solar] Finished fetching HV2140107737_data_update_coordinator data in 3.432 seconds (success: True)
2024-03-16 11:18:01.128 DEBUG (MainThread) [custom_components.huawei_solar] Finished fetching HV2140107737_configuration_update_coordinator data in 0.000 seconds (success: True)
2024-03-16 11:18:01.179 DEBUG (MainThread) [huawei_solar.huawei_solar] Logging in
2024-03-16 11:18:01.179 DEBUG (MainThread) [pymodbus.logging] send: 0x0 0x12 0x0 0x0 0x0 0x5 0x1 0x41 0x24 0x1 0x0
2024-03-16 11:18:01.180 DEBUG (MainThread) [pymodbus.logging] Adding transaction 18
2024-03-16 11:18:01.187 DEBUG (MainThread) [pymodbus.logging] recv: 0x0 0x12 0x0 0x0 0x0 0x15 0x1 0x41 0x24 0x11 0xe4 0x95 0x62 0x62 0x8a 0x40 0xf8 0xec 0xb6 0x23 0x93 0x7d 0xdb 0xcc 0x56 0x67 0x1 old_data:  addr=None
2024-03-16 11:18:01.188 DEBUG (MainThread) [pymodbus.logging] Processing: 0x0 0x12 0x0 0x0 0x0 0x15 0x1 0x41 0x24 0x11 0xe4 0x95 0x62 0x62 0x8a 0x40 0xf8 0xec 0xb6 0x23 0x93 0x7d 0xdb 0xcc 0x56 0x67 0x1
2024-03-16 11:18:01.188 DEBUG (MainThread) [pymodbus.logging] Factory Response[PrivateHuaweiModbusResponse': 65]
2024-03-16 11:18:01.188 DEBUG (MainThread) [pymodbus.logging] Getting transaction 18
2024-03-16 11:18:01.240 DEBUG (MainThread) [pymodbus.logging] send: 0x0 0x13 0x0 0x0 0x0 0x3f 0x1 0x41 0x25 0x3b 0xb4 0xb1 0xc1 0x5 0x53 0xb 0x7c 0x39 0x61 0x48 0xc9 0xbb 0x68 0x8b 0x0 0xdf 0x9 0x69 0x6e 0x73 0x74 0x61 0x6c 0x6c 0x65 0x72 0x20 0xae 0xad 0x20 0x45 0xbd 0x2c 0x7b 0xbc 0xf6 0xe3 0x22 0x8 0xce 0x78 0x33 0xb7 0xc3 0x54 0xf4 0x5b 0x79 0xfc 0x9d 0x45 0x88 0x72 0x78 0x1e 0xfa 0x90 0x6f 0x95
2024-03-16 11:18:01.241 DEBUG (MainThread) [pymodbus.logging] Adding transaction 19
2024-03-16 11:18:02.567 DEBUG (MainThread) [pymodbus.logging] recv: 0x0 0x13 0x0 0x0 0x0 0x58 0x1 0x41 0x25 0x54 0x0 0x20 0x44 0xa8 0x3b 0xcc 0x75 0x31 0xed 0x8a 0xb3 0x20 0xe5 0x4f 0x1e 0x84 0x18 0x58 0xdd 0x1 0x7b 0xc9 0xaf 0xa6 0xa3 0xe6 0x79 0xc7 0x1 0x1f 0xb5 0x30 0x0 0xda 0x0 0x0 0x51 0xa6 0xab 0x4 0xa5 0xbf 0xc7 0xf7 0xcf 0x31 0xc4 0x9c 0x14 0x13 0x75 0x67 0x10 0x7f 0xd3 0x65 0x12 0x19 0x8c 0x7f 0x4a 0x5a 0x8d 0x4 0xee 0xd6 0xe0 0x24 0xc5 0xa4 0x35 0x42 0xa3 0x6d 0x8f 0x16 0xc9 0xeb 0x66 0x20 0x72 0xe5 0x4a 0xd old_data:  addr=None
2024-03-16 11:18:02.567 DEBUG (MainThread) [pymodbus.logging] Processing: 0x0 0x13 0x0 0x0 0x0 0x58 0x1 0x41 0x25 0x54 0x0 0x20 0x44 0xa8 0x3b 0xcc 0x75 0x31 0xed 0x8a 0xb3 0x20 0xe5 0x4f 0x1e 0x84 0x18 0x58 0xdd 0x1 0x7b 0xc9 0xaf 0xa6 0xa3 0xe6 0x79 0xc7 0x1 0x1f 0xb5 0x30 0x0 0xda 0x0 0x0 0x51 0xa6 0xab 0x4 0xa5 0xbf 0xc7 0xf7 0xcf 0x31 0xc4 0x9c 0x14 0x13 0x75 0x67 0x10 0x7f 0xd3 0x65 0x12 0x19 0x8c 0x7f 0x4a 0x5a 0x8d 0x4 0xee 0xd6 0xe0 0x24 0xc5 0xa4 0x35 0x42 0xa3 0x6d 0x8f 0x16 0xc9 0xeb 0x66 0x20 0x72 0xe5 0x4a 0xd
2024-03-16 11:18:02.567 DEBUG (MainThread) [pymodbus.logging] Factory Response[PrivateHuaweiModbusResponse': 65]
2024-03-16 11:18:02.567 DEBUG (MainThread) [pymodbus.logging] Getting transaction 19
2024-03-16 11:18:02.568 INFO (MainThread) [custom_components.huawei_solar] Cannot create optimizer sensor entities as the integration has insufficient permissions. Consider enabling elevated permissions to get more optimizer data
Traceback (most recent call last):
  File "/config/custom_components/huawei_solar/__init__.py", line 141, in async_setup_entry
    await bridge.get_optimizer_system_information_data()
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/bridge.py", line 277, in get_optimizer_system_information_data
    file_data = await self._read_file(OptimizerSystemInformationDataFile.FILE_TYPE)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/bridge.py", line 229, in _read_file
    logged_in = await self.login(self.__username, self.__password)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/bridge.py", line 334, in login
    self.start_heartbeat()
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/bridge.py", line 341, in start_heartbeat
    raise HuaweiSolarException("Cannot start heartbeat as it's still running!")
huawei_solar.exceptions.HuaweiSolarException: Cannot start heartbeat as it's still running!
2024-03-16 11:18:02.589 DEBUG (MainThread) [custom_components.huawei_solar.number] No battery detected on slave 0. Skipping energy storage number entities
```

### Please confirm the following:

- [X] I'm running the latest release of Home Assistant.
- [X] I'm running the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

wlcrs, 2024-03-17:
This looks like a sneaky race condition bug. I've added an additional locking mechanism for preventing this from happening again.

Can you please test if the issue persists when using the latest release of the underlying `huawei-solar` library? You can do this by changing the [requirements in manifest.json](https://github.com/wlcrs/huawei_solar/blob/main/manifest.json#L11) to read: `huawei-solar==2.3.0b2`. Then restart your HA to load the new code.

techwolfpl, 2024-03-18:
Hi! Thanks for the quick response.
Now error has gone but still no optimizer data, stil 34 entities

'''
2024-03-18 15:58:30.381 DEBUG (MainThread) [huawei_solar.huawei_solar]
Reading register 32000 with length 20 from slave 1
2024-03-18 15:58:30.381 DEBUG (MainThread) [pymodbus.logging] send: 0x0
0x16 0x0 0x0 0x0 0x6 0x1 0x3 0x7d 0x0 0x0 0x14
2024-03-18 15:58:30.381 DEBUG (MainThread) [pymodbus.logging] Adding
transaction 22
2024-03-18 15:58:30.473 DEBUG (MainThread) [pymodbus.logging] recv: 0x0
0x16 0x0 0x0 0x0 0x2b 0x1 0x3 0x28 0x0 0x6 0x0 0x0 0x0 0x7 0x0 0x0 0x0 0x0
0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0
0x0 0xcd 0xfc 0x5 0x57 0x0 0x6d 0x0 0x0 0x0 0x0 old_data: addr=None
2024-03-18 15:58:30.473 DEBUG (MainThread) [pymodbus.logging] Processing:
0x0 0x16 0x0 0x0 0x0 0x2b 0x1 0x3 0x28 0x0 0x6 0x0 0x0 0x0 0x7 0x0 0x0 0x0
0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0
0x0 0x0 0xcd 0xfc 0x5 0x57 0x0 0x6d 0x0 0x0 0x0 0x0
2024-03-18 15:58:30.473 DEBUG (MainThread) [pymodbus.logging] Factory
Response[ReadHoldingRegistersResponse': 3]
2024-03-18 15:58:30.474 DEBUG (MainThread) [pymodbus.logging] Getting
transaction 22
2024-03-18 15:58:30.474 DEBUG (MainThread) [pymodbus.logging] [6, 0, 7, 0,
0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 52732, 1367, 109, 0, 0]
2024-03-18 15:58:30.474 DEBUG (MainThread) [pymodbus.logging] handle:
[b'\x00\x00', b'\x00\x00']
2024-03-18 15:58:30.525 DEBUG (MainThread) [huawei_solar.huawei_solar]
Reading register 37201 with length 1 from slave 1
2024-03-18 15:58:30.525 DEBUG (MainThread) [pymodbus.logging] send: 0x0
0x17 0x0 0x0 0x0 0x6 0x1 0x3 0x91 0x51 0x0 0x1
2024-03-18 15:58:30.525 DEBUG (MainThread) [pymodbus.logging] Adding
transaction 23
2024-03-18 15:58:30.894 DEBUG (MainThread) [pymodbus.logging] recv: 0x0
0x17 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0xc old_data: addr=None
2024-03-18 15:58:30.894 DEBUG (MainThread) [pymodbus.logging] Processing:
0x0 0x17 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0xc
2024-03-18 15:58:30.894 DEBUG (MainThread) [pymodbus.logging] Factory
Response[ReadHoldingRegistersResponse': 3]
2024-03-18 15:58:30.894 DEBUG (MainThread) [pymodbus.logging] Getting
transaction 23
2024-03-18 15:58:30.894 DEBUG (MainThread) [pymodbus.logging] [12]
2024-03-18 15:58:30.894 DEBUG (MainThread) [custom_components.huawei_solar]
Finished fetching HV2140107737_data_update_coordinator data in 0.727
seconds (success: True)
'''

techwolfpl, 2024-03-18:
Wait, I just get it again.

'
2024-03-18 15:57:07.915 DEBUG (MainThread) [pymodbus.logging] Adding
transaction 18
2024-03-18 15:57:09.243 DEBUG (MainThread) [pymodbus.logging] recv: 0x0
0x12 0x0 0x0 0x0 0x58 0x1 0x41 0x25 0x54 0x0 0x20 0xa7 0x84 0x3e 0xa8 0x4b
0x43 0xa5 0xd 0x47 0xf0 0x17 0x13 0x2d 0x5f 0xd9 0xc2 0x42 0xd0 0x99 0x6e
0x6a 0xed 0x26 0x26 0xfa 0x1b 0x61 0x9d 0x3 0xa 0xd7 0x48 0x0 0x0 0x5a 0xce
0x7f 0x95 0x9e 0x16 0x14 0x9f 0x3f 0xa3 0x7c 0xef 0x37 0xee 0x39 0x19 0x82
0x30 0x74 0xed 0xfb 0x80 0xda 0x2c 0xe7 0x71 0xf7 0xe4 0xba 0x6c 0x3 0xb9
0x44 0x88 0xd6 0x69 0x5e 0x21 0xdd 0x47 0x1d 0x96 0xc6 0x98 0xad 0x66 0xe0
0xbc old_data: addr=None
2024-03-18 15:57:09.243 DEBUG (MainThread) [pymodbus.logging] Processing:
0x0 0x12 0x0 0x0 0x0 0x58 0x1 0x41 0x25 0x54 0x0 0x20 0xa7 0x84 0x3e 0xa8
0x4b 0x43 0xa5 0xd 0x47 0xf0 0x17 0x13 0x2d 0x5f 0xd9 0xc2 0x42 0xd0 0x99
0x6e 0x6a 0xed 0x26 0x26 0xfa 0x1b 0x61 0x9d 0x3 0xa 0xd7 0x48 0x0 0x0 0x5a
0xce 0x7f 0x95 0x9e 0x16 0x14 0x9f 0x3f 0xa3 0x7c 0xef 0x37 0xee 0x39 0x19
0x82 0x30 0x74 0xed 0xfb 0x80 0xda 0x2c 0xe7 0x71 0xf7 0xe4 0xba 0x6c 0x3
0xb9 0x44 0x88 0xd6 0x69 0x5e 0x21 0xdd 0x47 0x1d 0x96 0xc6 0x98 0xad 0x66
0xe0 0xbc
2024-03-18 15:57:09.244 DEBUG (MainThread) [pymodbus.logging] Factory
Response[PrivateHuaweiModbusResponse': 65]
2024-03-18 15:57:09.244 DEBUG (MainThread) [pymodbus.logging] Getting
transaction 18
2024-03-18 15:57:09.244 INFO (MainThread) [custom_components.huawei_solar]
Cannot create optimizer sensor entities as the integration has insufficient
permissions. Consider enabling elevated permissions to get more optimizer
data
Traceback (most recent call last):
File "/config/custom_components/huawei_solar/__init__.py", line 141, in
async_setup_entry
await bridge.get_optimizer_system_information_data()
File "/usr/local/lib/python3.12/site-packages/huawei_solar/bridge.py", line
278, in get_optimizer_system_information_data
file_data = await
self._read_file(OptimizerSystemInformationDataFile.FILE_TYPE)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.12/site-packages/huawei_solar/bridge.py", line
230, in _read_file
logged_in = await self.ensure_logged_in()
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.12/site-packages/huawei_solar/bridge.py", line
345, in ensure_logged_in
self.start_heartbeat()
File "/usr/local/lib/python3.12/site-packages/huawei_solar/bridge.py", line
367, in start_heartbeat
raise HuaweiSolarException("Cannot start heartbeat as it's still running!")
huawei_solar.exceptions.HuaweiSolarException: Cannot start heartbeat as
it's still running!
2024-03-18 15:57:09.251 DEBUG (MainThread)
[custom_components.huawei_solar.number] No battery detected on slave 0.
Skipping energy storage number entities
'

wlcrs, 2024-03-19:
Can you please try upgrading to v1.3.4 beta 1 ?

techwolfpl, 2024-03-19:
Sure! I just did! Seems to be much better. Now I have all the opts as devices and over 150 entities. I will let you know tomorrow how about the readings because it's after sunset now and optimizers are disonnected.

Thank you so much!

techwolfpl, 2024-03-20:
I works perfectly! Thanks again!

[image]

closed 2024-03-21 by wlcrs