wlcrs/huawei_solar #657: [Bug]: AsyncModbusTcpClient.close() got an unexpected keyword argument 'intern'

barto64, 2024-03-18:
### Describe the issue

My HA instance is losing connection to my Huawei Solar installation after a few hours working. I have bounced a few times the whole solar pant and the HA with no results.

### Bescribe your Huawei Solar Setup

Inverter Type: SUN2000-5KTL & SUN2000-3.68K
Inverter Firmware version: V200R001C00SPC136 for both
SDongle present: yes (Firmaware version V200R022C10SPC108)
Power meter present: single phase
Battery: LUNA2000 10kWh
Battery Firmware version: V100R002C00SPC119

### How do you connect to the inverter?

Via the SDongle, wired connection

### Upload your Diagnostics File

[config_entry-huawei_solar-5775d102eb06432ba9dd94f28e4b5f7a.json](https://github.com/wlcrs/huawei_solar/files/14639463/config_entry-huawei_solar-5775d102eb06432ba9dd94f28e4b5f7a.json)

### Upload your relevant debug logs

```text
2024-03-18 17:13:48.015 ERROR (MainThread) [custom_components.huawei_solar] Unexpected error fetching TA2190014966_configuration_update_coordinator data: AsyncModbusTcpClient.close() got an unexpected keyword argument 'intern'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 318, in _async_refresh
    self.data = await self._async_update_data()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/huawei_solar/__init__.py", line 414, in _async_update_data
    return await self.bridge.update_configuration_registers()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/bridge.py", line 213, in update_configuration_registers
    result.update(await self._get_multiple_to_dict(ENERGY_STORAGE_CONFIGURATION_PARAMETERS_1))
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/bridge.py", line 162, in _get_multiple_to_dict
    return dict(zip(names, await self.client.get_multiple(names, self.slave_id)))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 276, in get_multiple
    response = await self._read_registers(registers[0].register, total_length, slave)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 398, in _read_registers
    return await _do_read()
           ^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/backoff/_async.py", line 151, in retry
    ret = await target(*args, **kwargs)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/backoff/_async.py", line 151, in retry
    ret = await target(*args, **kwargs)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 351, in _do_read
    response = await self._client.read_holding_registers(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pymodbus/client/base.py", line 179, in async_execute
    self.close(reconnect=True)
  File "/usr/local/lib/python3.12/site-packages/pymodbus/client/tcp.py", line 92, in close
    super().close(reconnect=reconnect)
  File "/usr/local/lib/python3.12/site-packages/pymodbus/client/base.py", line 130, in close
    self.connection_lost(asyncio.TimeoutError("Server not responding"))
  File "/usr/local/lib/python3.12/site-packages/huawei_solar/modbus.py", line 40, in connection_lost
    super().connection_lost(exc)
  File "/usr/local/lib/python3.12/site-packages/pymodbus/transport/transport.py", line 288, in connection_lost
    self.close(intern=True)
TypeError: AsyncModbusTcpClient.close() got an unexpected keyword argument 'intern'
```

### Please confirm the following:

- [X] I'm running the latest release of Home Assistant.
- [X] I'm running the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

vwtuner, 2024-03-19:
[image]
[image]

I have the same problem.

HA restarted and Huawei worked perfectly again for a few hours

wlcrs, 2024-03-19:
Related to #658 . I'll need to upgrade the supported pymodbus version .

wlcrs, 2024-03-19:
Can you please try upgrading to v1.3.4 beta 1 ?

barto64, 2024-03-19:
I am sorry I am on a work related travel and will not have access to my HA until Friday.

LotharWoman, 2024-03-19:
@all Please test sDongle firmeware V200R022C10SPC113. Since i have update to this version i haven't lost the connection in weeks and this with an interval of just 10 seconds.

barto64, 2024-03-19:
For my dongle (SDongleA-05) the one I have (108) is the most recent than the eu5 portal offers me for update and is the one I have installed. I cannot see the one you say (113).

Thanks any way for the advice.

vwtuner, 2024-03-20:
I installed the beta and didn't have any disconnections last night

wlcrs, 2024-03-20:
Ok, I'm leaving this open until I release a stable version containing this fix. Otherwise somebody will make a duplicate issue.

barto64, 2024-03-20:
Great thanks, I will also install the beta when I am back at home on Friday and will provide feedback

barto64, 2024-03-21:
I finally managed to install it last night. Almost 24h later all is good.

vwtuner, 2024-03-22:
[image]

today the same problem

wlcrs, 2024-03-22:
Please upgrade to HA 2024.3.3 . If the problem persists: please provide a diagnostics file and proper debug logs. Your screenshot does not help me solve your problem.

vwtuner, 2024-03-22:
Wo finde ich den Debug file?

wlcrs, 2024-03-22:
[image]

barto64, 2024-03-23:
FYI, mine remains working well

fboundy, 2024-03-25:
Hi @wlcrs - can you give any info on the fix as the same error is occurring with Solis inverters: https://github.com/wills106/homeassistant-solax-modbus/issues/818

barto64, 2024-03-25:
> Hi @wlcrs - can you give any info on the fix as the same error is occurring with Solis inverters: https://github.com/wills106/homeassistant-solax-modbus/issues/818

As per the above on this issue:

Related to #658 . I'll need to upgrade the supported pymodbus version .

vwtuner, 2024-03-25:
upgrade to HA 2024.3.3 And Huawei beta 
I have no problems today

fboundy, 2024-03-25:
Thanks - I just figured that it requires pymodbus 3.6.6 and hence HA Core 2024.3.2+

closed 2024-05-20 by github-actions[bot]