wlcrs/huawei_solar #661: [Bug]: pymodbus latest update included in HA today preventing Huawei Solar starting

ruteclrp, 2024-03-22:
### Describe the issue

Unable to start integration

### Bescribe your Huawei Solar Setup

Inverter Type: SUN2000-8KTL-M1
Inverter Firmware version:
SDongle present: yes
Power meter present: three phase
Battery: LUNA2000 15kWh
Battery Firmware version:

### How do you connect to the inverter?

Via the SDongle, wired connection

### Upload your Diagnostics File

Drag & Drop your Diagnostics File here.

### Upload your relevant debug logs

```text
2024-03-22 15:45:12.384 ERROR (MainThread) [homeassistant.setup] Setup failed for custom integration 'huawei_solar': Unable to import component: cannot import name 'checkCRC' from 'pymodbus.utilities' (/usr/local/lib/python3.12/site-packages/pymodbus/utilities.py)
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/setup.py", line 296, in _async_setup_component
component = await integration.async_get_component()
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/src/homeassistant/homeassistant/loader.py", line 894, in async_get_component
comp = self.get_component()
^^^^^^^^^^^^^^^^^^^^
File "/usr/src/homeassistant/homeassistant/loader.py", line 921, in get_component
ComponentProtocol, importlib.import_module(self.pkg_path)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.12/importlib/__init__.py", line 90, in import_module
return _bootstrap._gcd_import(name[level:], package, level)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
File "<frozen importlib._bootstrap_external>", line 995, in exec_module
File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
File "/config/custom_components/huawei_solar/__init__.py", line 17, in <module>
from huawei_solar import HuaweiSolarBridge, HuaweiSolarException, InvalidCredentials, register_values as rv
File "/usr/local/lib/python3.12/site-packages/huawei_solar/__init__.py", line 6, in <module>
from .bridge import HuaweiSolarBridge # noqa
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.12/site-packages/huawei_solar/bridge.py", line 16, in <module>
from .huawei_solar import DEFAULT_BAUDRATE, DEFAULT_SLAVE, DEFAULT_TCP_PORT, AsyncHuaweiSolar, Result
File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 19, in <module>
from pymodbus.utilities import checkCRC, computeCRC
ImportError: cannot import name 'checkCRC' from 'pymodbus.utilities' (/usr/local/lib/python3.12/site-packages/pymodbus/utilities.py)
```

### Please confirm the following:

- [ ] I'm running the latest release of Home Assistant.
- [ ] I'm running the latest release of this integration.
- [ ] I did not find an existing issue describing this problem.
- [ ] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [ ] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

xm4rcell0x, 2024-03-22:
I second this
Latest 2024.3.2 break huawei_solar

BIGede, 2024-03-22:
The same here....

wlcrs, 2024-03-22:
Your installation has the wrong version of pyModbus (a depenency that this integration uses to make the connection) installed. This is due to another integration being installed/upgraded after installing this integration, which overrides the pyModbus version and causes a conflict.

I've already released a beta version which has upgraded pyModbus to the latest version.

You can expect this issue to arise every so often when the pyModbus versions get out of sync again and incompatibilities surface. Also known as: the Python dependency hell

ruteclrp, 2024-03-22:
Updated to HA Core 2024.3.3 and that had no effect on- as expected. 
Earlier today updated HA Core to 2024.3.2 which Bumped pymodbus to v3.6.6
Upon restart after this pymodbus update, Huawei Solar cannot anymore be loaded. It also means that it is not possible to download a diagnostics file.

wlcrs, 2024-03-23:
Have you installed the latest beta version?

Eeleco, 2024-03-23:
I've had the same problem, but after instelling the beta everything looks okay again.

madsdyd, 2024-03-23:
I have the same issue, I believe. On 24.3.3 and everything seems to be latest version.

How do I update to the beta version? Is it something I need to do in HACS or? No updates pop up in HA. I have never tried a beta version before.

Thanks

Eeleco, 2024-03-23:
> I have the same issue, I believe. On 24.3.3 and everything seems to be latest version.
> 
> How do I update to the beta version? Is it something I need to do in HACS or? No updates pop up in HA. I have never tried a beta version before.
> 
> Thanks

Okay so open HACS, go to Huawei Solar, go here;
[image]

My installation is in Dutch but I guess it will translate to "download again" or something.

After that just enable seeing beta versions and select the beta version (spoiler; it has a 'b' in it)

[image]

patrickbohren, 2024-03-23:
I had the same issue but got it running again  by updating and checking out in custom_components/huawei_solar directory the git repo to the latest  main state. Most probably I installed it in the manual way long time ago.

duayfabi, 2024-03-23:
Same issue for me. Updated to the beta version and all fine now. Thx for the responsiveness.

madsdyd, 2024-03-23:
Thank you veru much @Eeleco 

It seems my HACS is UK, so I replicated the process for others that may be as HACS challenged as I, and has a harder time understanding Dutch (I am Danish, so Dutch is almost readable).

Open HACS, and find the Huawei Solar package:

[image]

Click, it, note the version (just for verification) and click the dot menu:

[image]

Click "Redownload":

[image]

Enable beta versions, select the new betaversion from the list, and download it:

[image]

Now restart Home Assistant.

I can confirm that this fixed my problem.

Thanks again, @Eeleco! And thank you, @wlcrs for creating and maintaing this amazing integration, and doing a fix for this issue so quickly! Amazing work!

ablyes, 2024-03-23:
I can confirm. The 1.3.4b1 versions is working fine. 
Thank you for your high reactivity regarding this problem.

ruteclrp, 2024-03-23:
Can confirm the new beta version makes the issue go away. Thanks for prompt attention. 

Closed.

tfriberg, 2024-03-27:
Thank you! 1.3.4b1 did the trick for me. I no longer have to restart the service every morning.

Misiu, 2024-03-29:
Thank you! Works great.

closed 2024-04-29 by ruteclrp