wlcrs/huawei_solar #662: [Feature Request]: Periodic restart of modbus TCP connection

redmars27, 2024-03-25:
### Describe your feature request

I am experiencing a estrange behavior with my SUN2000-5KTL-L1. From time to time (it can be weeks) it gets disconnected from FusionSolar and stop sending data to the portal and I think it is indirectly cause by the connection with home assistant.

Restarting home assistant solves the problem instantly, and the inverter gets connected again with FusionSolar portal.

Is anyone experiencing a similar problem?

My installation is:
Inverter: SUN2000-5KTL-L1
Inverter Firmware: SUN2000L V200R001C00SPC133
I had the issue originally with SUN2000L V200R001C00SPC130 but after speaking with huawei I upgraded to 133 with same results.
SDongle: No
Inverter connected to normal WIFI to access the internet
HomeAssistant connected to inverter internal WIFI to stablish the mod bus tcp connection.

The inverter working alone (without having a connection with home assistant) works without issues.

The integration works flawless and it is not having any issue at all.

I have been able to reproduce the error (but not always) shutting down the internet connection, so I think that if the Inverter gets disconnected from the internet and have an active modbus tcp connection, it doesn't reconnect again. To me the problem is on the inverter, as the modbus tcp connection is working flawlessly and collecting data all the time. So it is not a bug of this integration.

Speaking with Huawei, as this is not a supported integration, after updating the firmware, they don't offer any other solution as this is not a supported setup and if we dont have home assistant it is not failing.

What I see is that restarting the modbus tcp connection solves the problem, on the moment the modbus tcp connection is down, the inverter reconnect with fusion solar and resend any missing data, (at least the last few days of data).

So my feature request is to enable a setting on this integration to restart the modbus tcp once per day, for example by night at 2am or at any given time, shutdown the connection, wait 1 second and reconnect again. With that, the inverter will reconnect if it connection was down, and resend the missing data of the day.

### Proper usage

- [X] I confirm that this is not a bug report or support request
- [X] I confirm that this feature request is within the stated scope of the integration

wlcrs, 2024-03-25:
This can already be easily achieved by creating an automation that calls the `homeassistant.reload_config_entry` service at the moment of your choice.

As there is an easy workaround and I have received very little other reports of this problem, I don't think that the added complexity of this feature is warranted to be added into the integration. Only if I get extensive community feedback that this is a widespread problem, then I will reconsider this.

gialme, 2024-03-28:
I experience the exact same behavior with the SUN2000-6KTL-L1 inverter. I'll try reloading the integration config entry as suggested by @wlcrs to see if it fixes the problem. Thanks

redmars27, 2024-04-08:
Confirmed, reloading the integration solves the problems. Thanks

closed 2024-04-08 by redmars27