wlcrs/huawei_solar #665: [Bug]: huawei_solar v 1.3.4b1 integration is not loaded

Almetu, 2024-03-29:
### Describe the issue

Just after update/installation  the integration loaded and works normally, but after some time or HA restart it’s not loaded. 

Home Assistant
Core 2024.3.3
Supervisor 2024.03.1
Operating System 12.1
Frontend 20240307.0

### Bescribe your Huawei Solar Setup

Inverter Type: SUN2000-6KTL-M1
Inverter Firmware version: the latest 
SDongle present: yes 
Power meter present: three phase / yes
Battery: LG RESU xxkWh / LUNA2000 xxkWh / None
Battery Firmware version:

### How do you connect to the inverter?

Via the `SUN2000-<serial_no>` WiFi

### Upload your Diagnostics File

2024-03-29 13:07:21.566 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration huawei_solar which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-03-29 13:07:51.739 ERROR (MainThread) [homeassistant.setup] Setup failed for custom integration 'huawei_solar': Unable to import component: No module named 'pymodbus.message'
File "/config/custom_components/huawei_solar/__init__.py", line 17, in <module>
from huawei_solar import HuaweiSolarBridge, HuaweiSolarException, InvalidCredentials, register_values as rv
File "/usr/local/lib/python3.12/site-packages/huawei_solar/__init__.py", line 7, in <module>
File "/usr/local/lib/python3.12/site-packages/huawei_solar/bridge.py", line 17, in <module>
from .huawei_solar import DEFAULT_BAUDRATE, DEFAULT_SLAVE, DEFAULT_TCP_PORT, AsyncHuaweiSolar, Result
File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 18, in <module>

### Upload your relevant debug logs

```text
2024-03-29 13:07:21.566 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration huawei_solar which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-03-29 13:07:51.739 ERROR (MainThread) [homeassistant.setup] Setup failed for custom integration 'huawei_solar': Unable to import component: No module named 'pymodbus.message'
File "/config/custom_components/huawei_solar/__init__.py", line 17, in <module>
from huawei_solar import HuaweiSolarBridge, HuaweiSolarException, InvalidCredentials, register_values as rv
File "/usr/local/lib/python3.12/site-packages/huawei_solar/__init__.py", line 7, in <module>
File "/usr/local/lib/python3.12/site-packages/huawei_solar/bridge.py", line 17, in <module>
from .huawei_solar import DEFAULT_BAUDRATE, DEFAULT_SLAVE, DEFAULT_TCP_PORT, AsyncHuaweiSolar, Result
File "/usr/local/lib/python3.12/site-packages/huawei_solar/huawei_solar.py", line 18, in <module>
```

### Please confirm the following:

- [X] I'm running the latest release of Home Assistant.
- [X] I'm running the latest release of this integration.
- [X] I did not find an existing issue describing this problem.
- [X] I did upload the diagnostics-file that I could retrieve from  the 'Devices & Services Page'
- [X] I did upload the relevant debug logs (via 'Enable Debug Logging'-feature or by manually configuring HA logging)

bufferzone2, 2024-04-01:
Same problem with me. I updated and after the restart I lost all the sensors. I deleted the integration and added it again but nothing happens, it connects to the inverter but no longer displays any sensors.

wlcrs, 2024-04-01:
Another pymodbus version is installed than what the integration supports. This happens when two or more integrations use that same dependency, but require different versions. v1.3.4b1 requires pyModbus v3.6.6

You will need to make sure that all modbus-dependent integrations support the same pymodbus version. Welcome in this special place of hell called the "Python dependency hell".

RonnyStraetling, 2024-04-03:
Any idea how to fix this? How can I check the versions in use? Is there a way making sure pymodbus is updated to 3.6.6?

Harald1387, 2024-04-18:
Hi, I think I have the same problem. This morning all worked well but than i lost all sensors. "Einrichtung fehlgeschlagen". I updated to the actual beta version but it didn't help. Is there a solution?

sLesage, 2024-04-22:
Same here, I only get errors since a few days again... Seems to be happening every time there was an update to something I believe. I'll try to do the remove / reinstall dance to see if that helps fix the problem again.

themaninspain, 2024-04-24:
I have the same problem.
Have tried to remove and reinstall, going all the way to removing it totally (via HACS) and doing a fresh download.
It now gets stuck on "Please wait, starting configuration wizard for Huawei Solar" which eventually times out with an error.
The version is the latest alpha.

faithless11, 2024-05-01:
+1 same here

dkrebs, 2024-05-01:
Same problem here …

themaninspain, 2024-05-01:
I finally solved it by deleting the integration, removing it from HACS and reinstalling (v1.3.3) from scratch.
The integration now works with no problems.

themaninspain, 2024-05-02:
Just updated to HA Core 2024.5.0 and the problem is back!
Now recovering to a backup of 2024.4.4 (which was working).

Harald1387, 2024-05-28:
Did someone find a solution for that problem? 
I have installed all HA updates since the problem appeared. The behavior is still the same. The integration works for some hours and than the error appears. After some hours without doing anything it works again and than the error comes back. But I really have no idea why. Some weeks ago all worked fine for month. Nothing changed. The fusion solar app is working.

Almetu, 2024-05-28:
Hi! 
I think that I somehow fixed that problem. I’d just simply connected my inverter via wired connection, using   Waveshare RS485 to ETH(B) POE Modbus Gateway.  
The integration has been working perfectly for almost two months. Thera are no errors or disconnections.

My current versions:

Home Assistant

Core 2024.5.5
Supervisor 2024.05.1
Operating System 12.3
Frontend 20240501.1

Huawei solar : 1.3.4b1

dkrebs, 2024-05-28:
I did not get the integration to work anymore. As I found it not very reliable and it stopped working several times over time, I switched to the native `modbus` integration via Modbus proxy. Modbus proxy is running on an old raspberry connected to the short ranged Wifi of the inverter and via cable to the home network.

[configuration.yml](https://gist.github.com/dkrebs/82b0ac30db6bd1042931ad8019a20889)
[huawei-sun-2000.yaml](https://gist.github.com/dkrebs/65e8387ddfd53300baf3a5f5871bcaf6)

wlcrs, 2024-06-15:
v1.4.0 is out

closed 2024-06-15 by wlcrs

themaninspain, 2024-06-24:
And I still have the same problem. Every time there is an update to HA Core or Supervisor the integration does not load.
No error messages appear and the only solution is to delete the integration and it's config and remove it from HACS. Reboot and and redownload and install from scratch - the whole process takes about an hour as I if I don't restart HACS after every step it fails to install.

brusch, 2024-07-02:
@wlcrs  Having the same issue with v1.4.1 and Core 2024.6.4

```
homeassistant  | 2024-07-02 21:39:37.290 ERROR (MainThread) [homeassistant.config_entries] Error occurred loading flow for integration huawei_solar: No module named 'huawei_solar'
```

I'm using HA in a docker container, also had to add the mount `./local:/.local` because it tries to write some stuff there, maybe it's a good idea to mention this somewhere, or to write everything into `/config` as the docs currently indicates.

brusch, 2024-07-02:
Uninstall / installed fixed it. Works now. Sorry 😊

themaninspain, 2024-07-11:
I just wanted to say that I've  now update to HA Core 2204.7.2 with no problems!